rockowitz/ddcutil #148: DDC communication failed over HDMI

inoton, 2020-10-19:
I have Fedora32 installed on my desktop computer with an RX 480 PCI-E video card, connected to a GreatWall 32cL36 32" curved monitor via HDMI.

Installed ddcuil from [this repository](https://copr.fedorainfracloud.org/coprs/rockowitz/ddcutil/):
dnf copr enable rockowitz/ddcutil

When I run `$ ddcutil detect`, I get an error message: `DDC communication failed`.
```
$ ddcutil detect

Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           CGC
      Model:            32CL36
      Serial number:    32CL36R2W0001
      Manufacture year: 2016
      EDID version:     1.3
   DDC communication failed
```
```
# ddcutil interrogate
```
[interrogate-inoton-GreatWall-32cL36.txt](https://github.com/rockowitz/ddcutil/files/5402676/interrogate-inoton-GreatWall-32cL36.txt)

```
$ i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a 3b -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

```
$ ddcutil detect --verbose

Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Invalid display
   I2C bus:             /dev/i2c-3
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true 
      /sys/bus/i2c/devices/i2c-3/name:   AMDGPU DM i2c hw bus 3
   EDID synopsis:
      Mfg id:           CGC
      Model:            32CL36
      Serial number:    32CL36R2W0001
      Manufacture year: 2016
      EDID version:     1.3
      Product code:     12854
      Binary sn:        1 (0x00000001)
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 0c e3 36 32 01 00 00 00   ..........62....
      +0010   32 1a 01 03 80 46 27 78 fe ee 95 a3 54 4c 99 26   2....F'x....TL.&
      +0020   0f 50 54 ad cf 00 81 40 81 c0 81 80 81 00 95 00   .PT....@........
      +0030   a9 c0 b3 00 d1 c0 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 55 50 21 00 00 1a 66 21 56 aa 51 00 1e 30   5.UP!...f!V.Q..0
      +0050   46 8f 33 00 c4 fe 10 00 00 1e 00 00 00 ff 00 33   F.3............3
      +0060   32 43 4c 33 36 52 32 57 30 30 30 31 00 00 00 fc   2CL36R2W0001....
      +0070   00 33 32 43 4c 33 36 0a 20 20 20 20 20 20 01 d5   .32CL36.      ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```
other
[image]

Regards

rockowitz, 2020-10-19:
> display-OSD

It appears that your monitor has a marginal I2C implementation.  At 
lines 508-515 of the interrogate output ddcutil makes three attempt to 
read feature x10 (brightness).  The first read() returns bytes that are 
almost valid, the second and third are pure garbage.  You will see this 
more clearly if you run ddcutil with the "--ddc" option, which reports 
DDC/CI data errors, and the  "-- stats" option which reports among other 
things error statistics.

The one suggestion I have is to increase the waits between when ddcutil 
sends a request to the monitor and when it reads the reply.   These 
waits, 40 - 200 ms, are mandated by the DDC/CI protocol.  The 
"--sleep-multiplier" multiplier option increases these wait times.  For 
example, "--sleep-multiplier 4" causes 40 ms waits to become 160 ms.  In 
fact, I'd try using a multiplier value of 4.  It's huge, and if that 
doesn't solve the problem I doubt anything larger would succeed. If it 
does solve the problem, you can test to see how small you can make the 
multiplier value.

If you do succeed, please send the output of "ddcutil probe". (with of 
course whatever other options you are using).  I will be interested to 
see display model and version.

Good luck!

--Sanford

inoton, 2020-10-20:
Thank you for developing such a useful tool and for taking the time to answer for me.
Frankly, there is some confusion with these specialized terms. As I understand it, I do the following.
<details>
<summary> Set the multiplication factor for DDC sleeps as: 4, 16, 32, 64, 96. </summary>

```
# ddcutil detect --sleep-multiplier 4
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           CGC
      Model:            32CL36
      Serial number:    32CL36R2W0001
      Manufacture year: 2016
      EDID version:     1.3
   DDC communication failed

```

```
# ddcutil detect --sleep-multiplier 16
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      ...
      EDID version:     1.3
   DDC communication failed
```

```
# ddcutil detect --sleep-multiplier 32
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      ...
      EDID version:     1.3
   DDC communication failed
```

```
# ddcutil detect --sleep-multiplier 64
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      ...
      EDID version:     1.3
   DDC communication failed
```

```
# ddcutil detect --sleep-multiplier 96
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      ...
      EDID version:     1.3
   DDC communication failed
```

```
# ddcutil getvcp 10 --sleep-multiplier 64
Display not found

# ddcutil getvcp 10 --sleep-multiplier 96
Display not found
```
</details>

It could be that the monitor doesn't support ddc/ci and it didn't work.

rockowitz, 2020-10-20:
There is no point in setting a sleep-multiplier greater than 4.  If 
giving the monitor 4 times the amount of time required by the DDC/CI 
specification to formulate a response doesn't solve the problem, it's 
extraordinarily unlikely that using a higher value will.  All it will do 
is increase elapsed execution time painfully.  Note that the --sleep 
multiplier value is a floating point number.  I've seen well behaved 
monitors that work with a --sleep-multiplier value of .1, which greatly 
speeds up ddcutil execution.  See Option: --sleep-multplier 
<http://www.ddcutil.com/performance_options/#option-sleep-multiplier>for 
details.

The problem is that the monitor doesn't support DDC/CI correctly. It 
responds to DDC requests,  I2C slave address x37 is active, the monitor 
responds to DDC requests, but  the bytes in a response sometimes look 
like they were valid before transmission errors.

closed 2020-11-09 by rockowitz