rockowitz/ddcutil #149: HDMI to VGA adapter - DDC communication failed

endyb66, 2020-10-30:
Device: Raspberry Pi 3 Model B
Adapter: [https://sandberg.at/de/product/HDMI-to-VGA-Converter](Sandberg HDMI to VGA Converter)

`ddcutil detect`

```
Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           RTK
      Model:            VGA
      Serial number:    Unspecified
      Manufacture year: 2017
      EDID version:     1.3
   DDC communication failed
```

Are there compatible HDMI to VGA adapters available, or do any other solution exist, to control my VGA display from my Pi 3?

rockowitz, 2020-10-30:
I am unfamiliar with specific HDMI - VGA adapters.  However the 
following information may be of use to you.

Unlike DVI-I and DVI-A, HDMI carries no analog signal.  The adapter 
reads the digital HDMI input and generates VGA analog output.

In both VGA and HDMI, DDC information is carried on 2 wires known as SCL 
and SDA (and associated grounds).  In principle an adapter can simply 
connect the wires.  As to whether a particular adapter does so you'll 
have to check with the manufacturer.

endyb66, 2020-10-31:
Thank you @rockowitz. So then would it be possible, to interface only 2 pins (SDA & SCL) from a VGA input to directly control the monitor right?

rockowitz, 2020-11-01:
What are you trying to do?

endyb66, 2020-11-01:
I have connected a device over HDMI to my monitor. This device is not able to send DDC/CI commands. So my idea is to use the secondary input on the monitor, which is VGA, and connect only SDA and SCL on it to control it over I2C from e.g. a Raspberry

rockowitz, 2020-11-01:
The most basic answer is .. it depends.  Some monitors accept DDC/CI 
commands only on the currently selected input, which would make what you 
want to do impossible.  Other monitors  accept DDC/CI commands from all 
inputs.

Then there the question of how you're going to use the control device. 
Let's say it's a Raspberry Pi.  Here are some things to consider.

Will you try to use the HDMI connector for DDC communication?  If you 
want to drive a normal monitor as well, you'll be looking at a Pi 4 
which has 2 HDMI connectors.  Last I was aware, the video driver for the 
Pi 4 had not been updated to support DDC.  If a single HDMI output is 
adequate, will /dev/i2-2 (the I2C bus for the HDMI port) be active if no 
monitor is detected?  ddcutil may or may not work in this situation, 
depending on whether it can read the EDID over the I2C bus.

You'll more likely want to use the I2C connections on the GPIO header.  
In this case, you'll be programming at the DDC protocol level.

Good luck!

endyb66, 2020-11-02:
Thank you @rockowitz for your detailed answer! Now i've tested every possible solution and like you said, the montor which i need to control does only support DDC control on the current selected input. 
Now i modified a HDMI cable and soldered 2 wires to the I2C bus (SDA & SCL) and connected them to the I2C bus on my Raspberry Pi 3. This way i can interface DDC but using any videosource over HDMI to my monitor. I'm not very familiar with the I2C protocol, but what i can tell is, that i can read some data over it. What currently does not work (maybe i'm making a mistake), is writing to the brigtness register. I can read the device (0x50) and also can read the register for brightness (0x10) with the command `i2cget -y 1 0x50 0x10`. But what would the command look like, to write a new value to that register? I think it should be something like `i2cset -y 1 0x50 0x10` or am i completely wrong?

rockowitz, 2020-11-02:
Slave address x50 on the I2C bus provides the monitor's EDID.  Your 
i2cget command is reading byte x10 of the EDID.

DDC communication is on I2C slave address x37.  This is not a simple 
read and write.  The host writes a request packet to the display, then 
(typically) reads a reply packet.  (Some operations, such as setting a 
feature value, entail only writing a packet.) The primary document for 
the DDC/CI protocol is "VESA Display Data Channel Command Interface 
Standard, Version 1.1, October 29, 2004."  Document "VESA Enhanced 
Display Data Channel (EDDC) Standard, Version 1.2, December 26, 2007" 
can be helpful in clarifying the protocol.  Googling either of these 
documents will take you to the VESA web site, where they can be 
downloaded. An extensive set of references can be found here 
<https://www.ddcutil.com/bibliography>. See also this discussion of DDC 
vs I2C <http://www.ddcutil.com/ddc_vs_i2c/>.

Having said all that, it may be that ddcutil can communicate with your 
monitor, since at least the EDID appears to be readable at slave address 
x50 on /dev/i2c-1. Give it a try.

endyb66, 2020-11-06:
Now i've done some testing, it does work with ddcutil + the given i2c interface. But does give an error sometimes. 
The whole purpose is, that i have to read and write (set) the brightness (and only that value), from within C# code on a WinIotCore device. I took a look at the source code and read also the docs about the specification. I understand it only half way, it seems that i make a mistake somewhere. In gereral, is this way the correct one?

- byte[] writeBuf2 = new byte[] { 0x6E, 0x51, 0x02, 0x01, 0x10, 0x00 };
- Calculation checksum from the first 5 bytes from writeBuf -> Checksum is 0x42
- Set the cehckusm as byte #6 in the writeBuf array (overwrite the last one - 0x00)
- Write the writeBuf-array to the i2c address 0x37
- Read buffer as a byte-array with size of 12

As an answer i get:
0x00 0x00 0x64 0x00 0x64 0xa4 0x00 0x00 0x00 0x00 0x00 0x00

Do i make a misstake here?

rockowitz, 2020-11-08:
First, it seems you're bouncing back and forth between Linux (likely 
Raspian) and WinIotCore.  So I'm not quite sure how to respond, but here 
goes.

>
> Now i've done some testing, it does work with ddcutil + the given i2c 
> interface. But does give an error sometimes.

If ddcutil works sometimes but fails at others, the problem is likely a 
marginal monitor DDC/CI implementation.  You're not going to improve 
things by writing the low level code yourself.  There's a lot of 
functionality in ddcutil to work around/recover from DDC/CI problems.

The following options ddcutil may be of use:
   --ddc    Show ddc data errors
   --stats   Report error statistics
   --sleep-multiplier  Sometimes helps with marginal monitors

For details on the options, see the man page or better www.ddcutil.com.

> The whole purpose is, that i have to read and write (set) the 
> brightness (and only that value), from within C# code on a WinIotCore 
> device.
If you're programming a WinIotCore device, you're out of my area of 
expertise.   Windows has a very nice set of DDC/CI functions 
(https://docs.microsoft.com/en-us/windows/win32/monitor/monitor-configuration). 
Are these not available on WinIotCore?

> I took a look at the source code and read also the docs about the
> specification. I understand it only half way, it seems that i make a
> mistake somewhere. In gereral, is this way the correct [...]
If you're trying to write to and read from a Linux /dev/i2c device as a 
file, the following comments apply.

1) The high order bit of the length byte is always set.   The length is 
the low order 7 bits.  So the length byte in your example is x82, not x02.

2) The first byte of the array (destination address) is not written to 
/dev/i2c.  In your example the 5 bytes starting with x51 are written. 
The destination address (x37) is set before the write() call using the 
I2C_SLAVE or I2C_SLAVE_FORCE ioctl on the /dev/i2c device.

3) Per the DDC/CI spec, you must wait at leas 40 ms between writing the 
request packet and reading the response packet. Increasing the delay 
sometimes helps with marginal monitors.

endyb66, 2020-11-08:
Thank you very much, with the help of your tips i was able to get it to work! 
The integrated set of DDC/CI functions in windows, are not available in WinIoTCore, sadly. That's why i had to build my own low level class. I used Linux (raspbian) to first test the connection of the i2c bus directly to my monitors controller.
Now most things regarding the DDC/CI specification is clear to me, but i have to say, that the documents about the sepc is not 100% clearly written. But i'm very happy now, that it works, i can read and change values.

So again, thank you very much @rockowitz, spending you a virtual beer on this sunny sunday😊

rockowitz, 2020-11-08:
Congratulations on getting things working.  I agree that the DDC/CI spec 
is opaque, to say the least.  As I've written elsewhere, it's as if 
DDC/CI is a subclass of superclass I2C, with all the I2C details, e.g. 
start bits, stop bits, ACKs, included. Which makes sense if you're 
writing a bit-banging driver, but is irrelevant if you're using the file 
system abstraction provided by /dev/i2c.  Layered on top of that are the 
quirks of the /dev/i2c interface.

Sanford

closed 2020-11-09 by rockowitz