rockowitz/ddcutil #151: Dell U2518D minimal contrast level

eddyz87, 2020-11-13:
Hello,

Thanks for the decent utility, super helpful when I want to switch between "movie" & "reading" mode. However, I have a strange issue with my Dell U2518D. Using monitor physical controls it is possible to set contrast in the range between 0 - 100, using `ddcutil` it is only possible to set contrast in the range between 25 - 100. An attempt to set contrast below 25 leads to contrast being set to 25. For example:

```bash
$ ddcutil setvcp 0x12 30
$ ddcutil getvcp 0x12 
VCP code 0x12 (Contrast                      ): current value =    30, max value =   100
$ ddcutil setvcp 0x12 10
Verification failed for feature 12
$ ddcutil getvcp 0x12 
VCP code 0x12 (Contrast                      ): current value =    25, max value =   100
```
Is there anything that could be done here, or this is something hard-coded in the monitor firmware?

The interrogation report is attached: [ddcutil-interrogate-report.txt](https://github.com/rockowitz/ddcutil/files/5539489/ddcutil-interrogate-report.txt).

Best regards,
eddyz87

rockowitz, 2020-11-14:
Looks like the same problem (issue #121) that @epskampie analysed in detail for a Dell U2407wfp.   I'm afraid it's just a quirk of the monitor firmware. Out curiosity, does feature x10 show similar behaviour?

eddyz87, 2020-11-14:
The x10 could be set in the range between 0 to 100. Might be some kind of "foolproof" decision indeed, x10 at 0 renders screen very dim but readable, x12 at 0 (using monitor OSD) renders screen completely black.

Well, it is what it is then, thank you for comment :+1:

closed 2020-11-15 by rockowitz

szotsaki, 2023-10-03:
I have the same problem on a Dell C3422WE: contrast goes only until 25. What is more interesting, that the opensource [Monitorian](https://github.com/emoacht/Monitorian) handles my monitor well, I can go below 25 with it on Windows.

eddyz87, 2023-10-03:
> I have the same problem on a Dell C3422WE: contrast goes only until 25. What is more interesting, that the opensource [Monitorian](https://github.com/emoacht/Monitorian) handles my monitor well, I can go below 25 with it on Windows.

It looks like this tool uses directx to actually set the values, e.g. for brightness it calls one of two functions: [SetVCPFeature](https://github.com/emoacht/Monitorian/blob/master/Source/Monitorian.Core/Models/Monitor/MonitorConfiguration.cs#L101) or [SetMonitorBrightness](https://github.com/emoacht/Monitorian/blob/master/Source/Monitorian.Core/Models/Monitor/MonitorConfiguration.cs#L63C29-L63C49), both are imported from "Dxva2.dll".

It might be interesting to sniff what commands are sent via MCCS, but when I google for it I get this: https://www.ddcutil.com/sniffing/ , which suggests using a modified cable. Oh, my...

rockowitz, 2023-10-03:
I suspect that Monitorian only seems to be able to correctly set contrast below 25.  When a DDC/CI Set VCP Feature request packet is sent to a monitor, the only status returned is that the monitor has successfully received the packet.  There is no guarantee that the monitor has processed the request as expected.  Which is why, by default, the ddcutil **setvcp** command automatically then reads the current value and compares it to the requested value.  This check can be disabled using the ***--noverify*** option. 

As user @epskampie   reported in  issue #121, at least for his monitor requesting a value below below a certain level changed the contrast, but not to the expected number.  Below that level, the effect was non-linear.   Now, Monitorian may have some code to handle this special case, but I doubt it.

If you want see the I<sup>2</sup>C packets being processed by driver i2c-dev, you can instrument the driver, as described [here](https://michlstechblog.info/blog/linux-enable-i2c-kernel-tracing/l).  I would expect that Windows has a similar capability, but this is a question for someone familiar with Windows driver development.

szotsaki, 2023-10-04:
> There is no guarantee that the monitor has processed the request as expected.

You're right. I've just double checked it. The Monitorian contrast slider can be adjusted below 25, but the screen contrast won't change. Sorry for the noise.

I'll ask Dell support, hopefully they know what prevents it exactly.

rockowitz, 2023-10-04:
See the discussion on issue #121.  Our conjecture was it was a way to prevent the user from setting a value so low that they couldn't see the screen.  

Unfortunately, the DDC/CI spec reports the max value for a feature, but no way to report a minimum value.  It's always 0.