rockowitz/ddcutil #152: DDC Communication failed with a HP LA2006

bjonnh, 2020-11-15:
# ddcutil detect
```
Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           HWP
      Model:            HP LA2006
      Serial number:    CNC236Q04G
      Manufacture year: 2012
      EDID version:     1.3
   DDC communication failed

```
# ddcutil detect --verbose --stats --trace i2c --trace ddcio
[PasteBin](https://pastebin.com/PR4gsscD)

I also tried with --maxtries "0,15,0" and --sleep-multiplier 4

DDC is enabled on the screen.

This is with a direct DVI-D connection, it failed with a HDMI->DVI-D connection as well.

bjonnh, 2020-11-15:
(ddcutil works perfectly on my other screen DELL P2418D through DP)

rockowitz, 2020-11-16:
The problem is clearly with the monitor, though the problem may appear only with a particular video card.  Every I2C read response packet starts with a double x6e byte. .   The data errors will become clearer if you replace the --trace option with --ddc, which simply reports DDC data errors.

You've already tried the --sleep-multiplier option, which is the first thing I would have suggested. 

In terms of signalling, HDMI and DVI are identical.  If you can, try a DisplayPort<->DisplayPort connection.  (As I read the manual for the LA2006, it has a DP connector.)  Instead of using dedicated wires in the cable, DisplayPort "emulates" I2C communication over its AUX channel, so the communication mechanism is a bit different.  I doubt the DisplayPort connection will succeed when the DVI/HDMI connections do not, but it's worth a shot.  

Given that DDC communication works on the P2418D, it would be interesting to see if the LA2006 fails when connected to a different computer.

bjonnh, 2020-11-16:
I connected the monitor on the same machine through both DVI and DisplayPort, I can talk to it with DisplayPort:

Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           HWP
      Model:            HP LA2006
      Serial number:    CNC236Q04G
      Manufacture year: 2012
      EDID version:     1.3
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           HWP
      Model:            HP LA2006
      Serial number:    CNC236Q04G
      Manufacture year: 2012
      EDID version:     1.4
   VCP version:         2.2

bjonnh, 2020-11-16:
I tried on another machine (laptop) through VGA and it is detected

rockowitz, 2020-11-17:
One thing to keep in mind in your tests if your have multiple connections to the monitor.  Some monitors respond to DDC communication only  on the currently selected input.  Others respond to DDC communication on any input.

bjonnh, 2020-11-17:
Yes I was aware of that and was careful to talk to the right one at the right moment.

closed 2025-02-09 by rockowitz