rockowitz/ddcutil #153: Xiaomi Curved Screen OSD stopped working

norpol, 2020-11-19:
Hey, after playing with power modes and brightness via ddcutil I can't access the OSD of my monitor anymore.

Previously I would push the "menu" button and I would be presented with an OSD where I can do the monitor settings from hand.

Setting the "Restore Factory Defaults" value didn't help. I couldn't find any information on possible resetting the monitor preferences via button-key combination.

Do you have any idea or hints how I can recover from such a situation?

rockowitz, 2020-11-19:
Does your monitor implement feature xCA - OSB button control?

If your VCP version is 2.2 (quite possible since curved monitors are recent) it's a bit more complicated than this, but basically the values are:
 - x01 enabled
 - x02 disabled
 - xff  display cannot supply this information

If this doesn't solve your problem, please run "ddcutil probe" for the monitor and send the output as an attachment.

norpol, 2020-11-19:
Thank you for your response.

If I'm running, I receive this error below.

```bash
$ ddcutil setvcp 0xCA x01
Verification failed for feature ca
```

Probe output lists

> `VCP code 0xca (OSD                           ): OSD Enabled (sl=0x02)`

Attached are both [ `ddcutil probe`](https://github.com/rockowitz/ddcutil/files/5569742/probe.txt) and [`ddccontrol -p -c -d`](https://github.com/rockowitz/ddcutil/files/5569744/inital.txt) outputs.

I'm using `ddcutil setvcp 0xD6 "01"``, `ddcutil setvcp 0xD6 "04"`` and `ddcutil setvcp 0x10 "10"` in my scripts.

rockowitz, 2020-11-19:
First, an apology, which I'm afraid confused things.  The values for feature xCA are
- x01 - OSD   disabled
- x02 - OSD - enabled
- xFF - display cannot provide this information

In the probe output, the capabilities string does not include feature  xCA.  Yet probe of xCA (i.e. getvcp) reports the sensible value of x02 (enabled).  It does not set the unsupported feature bit in the reply. Interestingly, trying to setvcp feature xCA to x01 is rejected, so maybe feature xCA is half-supported, i.e. it can be read but not changed.

Bottom line: the monitor is reporting that OSD is enabled.

I have no idea why changing the value of feature xD6 between x01 and x04 (twiddling DPM on/off) should affect whether the OSD is enabled.  All I can say as a vague generality is that changing the value of one feature can affect others.  Such interactions are not documented.  I've only been interested in the interactions of colour related features.

I'm assuming that when you say you've tried resetting factory defaults, you set feature x04 to a non-zero value. 

I'm afraid that all I can suggest at this point is unplugging the monitor and let it sit for a bit.  This is the first instance I've ever heard of/experienced where  changing VCP values puts the monitor into an unrecoverable state.

norpol, 2020-11-20:
Hey, thank you for your quick reply again. I did not experience these issues immediately but just after the next day, given your reply I can only assume that the sudden OSD functionality being gone does not relate to VCP control values but perhaps some faulty unit.

I'll try your suggestion over night and hope for the best.

I'm closing this now, perhaps if someone will encounter a similar issue they will find this issue anyway.

If I can contribute with any other helpful values for development please let me know.

closed 2020-11-20 by norpol

rockowitz, 2020-11-20:
What precisely is your monitor model? It's unclear from the EDID.  I'd 
like to take a look at the manual.

norpol, 2020-11-24:
Oh, sorry I missed your response. Xiaomi only call the screen ["Mi Curved Gaming Monitor 34"](https://www.mi.com/global/monitor34/), I guess from the manual PDF filename it's `XMMNTWQ34`.

The manual [mi_xmmntwq34.pdf](https://github.com/rockowitz/ddcutil/files/5592965/mi_xmmntwq34.pdf) is just about build up and some basic firmware features.

m49n, 2022-06-04:
didn't find a solution?

eggsampler, 2022-06-04:
I have had a similar issue after attempting to use the power mode to turn the monitor off - no access to osd, buttons only turn the monitor black for a second.

No vcp setting brings the osd back or restores button functionality. I have left it off for many days, attempted turning on while holding all of the button combos, no luck.

For anyone finding this issue, I would strongly recommend not using any ddc/ci commands with this monitor. If you have, the monitor still works but you will not have access to the so won't be able to change the input or any settings. If it's in warranty period, return it.

norpol, 2022-06-04:
I've successfully returned it back then - I assumed it was an unrelated issue, I've been using ddcutil for many years on many screens and never encountered this bug before. Sorry about your problem. Maybe we can update ddcutil to block certain monitors/actions?

rockowitz, 2022-06-06:
@norpol  I'm not sure about blocking, but as a first step I've implemented a monitor quirk table that causes **detect** to append a warning message for problematic monitors.   Its initial member is the Xiaomi monitor. The implementation is in the current ddcutil development branch, 1.2.3-dev.

Note that detection entails reading feature values x00 (which provides useful information even though the feature should not exist) and feature xDF (VCP version).  If there are monitors for which reading any value can be fatal the check would have to be implemented differently.

I don't have a copy of the EDID for this monitor, so I may not have the model name exactly correct.  Can someone with this monitor send me the output of **ddcutil detect --verbose** so I can be sure of the Xiaomi monitor identifiers to use in the table?  Or at least confirm that **ddcutil detect --verbose** warns about the monitor.

ZarredFelicite, 2022-08-05:
Display 2
   I2C bus:  /dev/i2c-8
      Driver:                             nvidia
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-8/name     NVIDIA i2c adapter 8 at 2d:00.0
      PCI device path:                    /sys/devices/pci0000:00/0000:00:03.1/0000:2d:00.0/i2c-8
   EDID synopsis:
      Mfg id:               XMI - UNK
      Model:                Mi Monitor
      Product code:         13380  (0x3444)
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2020,  Week: 16
      EDID version:         1.4
      Extra descriptor:
      Video input definition:    0xb5 - Digital Input (DisplayPort), Bit depth: 10
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.677, 0.314
      Green x,y:        0.286, 0.637
      Blue  x,y:        0.136, 0.077
      Extension blocks: 2
   EDID source: I2C
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 61 a9 44 34 00 00 00 00   ........a.D4....
      +0010   10 1e 01 04 b5 50 21 78 3b 64 f5 ad 50 49 a3 22   .....P!x;d..PI."
      +0020   13 50 54 ad cf 00 71 4f 81 c0 81 40 81 80 95 00   .PT...qO...@....
      +0030   a9 c0 b3 00 d1 c0 22 68 70 a0 d0 a0 29 50 30 20   ......"hp...)P0
      +0040   35 00 1d 4e 31 00 00 1a 20 fd 70 a0 d0 a0 3c 50   5..N1... .p...<P
      +0050   30 20 35 00 1d 4e 31 00 00 1e 00 00 00 fd 00 30   0 5..N1........0
      +0060   90 a0 a0 3c 01 0a 20 20 20 20 20 20 00 00 00 fc   ...<..      ....
      +0070   00 4d 69 20 4d 6f 6e 69 74 6f 72 0a 20 20 02 ac   .Mi Monitor.  ..
   VCP version:         2.1
   Controller mfg:      Mstar
   Firmware version:    0.0
   Monitor returns DDC Null Response for unsupported features: false
   WARNING: Setting feature values has been reported to permanently cripple this monitor!

Diegohead, 2022-09-08:
> Hey, after playing with power modes and brightness via ddcutil I can't access the OSD of my monitor anymore.
>
> Previously I would push the "menu" button and I would be presented with an OSD where I [...]

I'm having the same problem but with the xiaomi 2k 165hz monitor (xmmnt27hq), even though I'm not using ddcutil or anything else, I'm on windows, it just died. 
I was searching and found this posts so I thought I could get some help or give my info to the community.

rockowitz, 2022-09-08:
@norpol Can you describe for @Diegohead what you did to recover your monitor?

norpol, 2022-09-08:
@rockowitz and @Diegohead I'm very sorry but I simply returned the screen as it showed up in the first few days.

HeckfyCU, 2023-01-14:
Hello. I'm having same issues, may be someone have some kind of remedy for this issue?

rockowitz, 2023-01-14:
@HeckfyCU First, let's make sure this is the same monitor as previously discussed on this thread. Please execute **ddcutil detect --verbose** and **ddcutil environment --verbose** and submit the output as attachments.  There will undoubtedly be lots of errors, but the output will still be useful.

HeckfyCU, 2023-01-14:
@rockowitz I would like to, but I don't have linux installed, only windows and Mac OS. But I used similar tools to control my display through ddc ci like https://www.nirsoft.net/utils/control_my_monitor.html and https://github.com/chrismah/ClickMonitorDDC7.2 and now i also can't turn osd. The only thing I managed to do is reset language options, so now display enters energy saving using Chinese language.

supermar1010, 2023-02-28:
I also came here after reading the other pieces of information available on the internet. Someone wrote that it might be an issue with the inputs:

```
What I noticed is that the VCP code for the input is currently 7 (listed as "S-Video1" , which is obviously wrong), but the possible value are 15, 16, 17 and 18. So I tried to connect my PC to other ports and the value changed to 5, 6 and 8. So there is a sort of offset of 10 from the actual values that it should take. 
```

Maybe that's a point from where one could start investigating. Maybe there's some way to reset those values to something sane?

supermar1010, 2023-03-05:
I just tried setting all the values from 0x0 to 0xff, nothing changes the monitor always stays at 0x07.

leag, 2023-08-25:
XMI3446 monitor has the same issue

RichardFevrier, 2023-10-18:
Same monitor, same problems, to fix them I have to unplug it sometimes and do an hard reset when the OSD is willing to appear.

supermar1010, 2023-10-18:
@RichardFevrier can you describe your procedure? I'd really like to have a fully working monitor again :) 

How do you do the hard reset?

RichardFevrier, 2023-10-18:
Probably not what you want to hear @supermar1010 but I ended up by uninstalling ddcutil. 
Then unpluged the power cord and after that the OSD was finally back.

supermar1010, 2023-10-18:
Well if it brings my osd back, that's absolutely fine, will tra that soon, j hope I have success, having no osd really sucks. I kinda accepted the monitor as half bricked tbh

supermar1010, 2023-10-29:
Hmm that did not work unfortunately, still got the same problem.
Uninstalled all ddc stuff, rebooted computer and monitor, when pressing any of the buttons the monitor just blacks out for a few seconds, but no osd

supermar1010, 2023-10-29:
Hmm that did not work unfortunately, still got the same problem.
Uninstalled all ddc stuff, rebooted computer and monitor, when pressing any of the buttons the monitor just blacks out for a few seconds, but no osd :/

RichardFevrier, 2023-10-30:
When you say rebooted monitor, have you unplugged the power cord? This was the trick for me

supermar1010, 2023-11-25:
Yes I unplugged for some time. But it did not help unfortunately

RichardFevrier, 2023-11-27:
Sorry to hear that.

brlin-tw, 2024-08-17:
Was trying to play with `ddcutil` until noticing the `ddcutil detect` warning text, thanks to everyone who has helped on this issue.

madjoe1976, 2024-09-20:
any update on this. i still have no OSD

brlin-tw, 2024-09-20:
> any update on this. i still have no OSD

I believe the one should have any update would be XiaoMi, which shipped the buggy display firmwarem

RichardFevrier, 2024-09-20:
Unfortunately that will probably never happen.

brlin-tw, 2024-09-20:
@RichardFevrier > Unfortunately that will probably never happen.

That's also true, let's vote with our wallet when the next purchase comes.

brlin-tw, 2024-09-20:
Also, you may want to file a complaint via [XiaoMi's official support channels](https://www.mi.com/global/support/contact/) and spread the issue so that everyone will know they are selling crappy displays.

ozp3, 2024-10-06:
Monitor with no osd lol. It seems that the only way we can reach the people who took part in the creation of this monitor. I wont buy any xiaomi monitor again.
https://github.com/waydabber/BetterDisplay/discussions/2190#discussioncomment-10855567

thewh1teagle, 2024-12-17:
Did you found a way to switch the input through DDC/CI? I experience the same issue in https://github.com/mjkoo/monitor-switch/issues/2

rm-killer, 2025-11-23:
> Hey, after playing with power modes and brightness via ddcutil I can't access the OSD of my monitor anymore.
>
> Previously I would push the "menu" button and I would be presented with an OSD where I [...]

Bro. i have xiaomi 34 curve and i have same problem. plz can you help me with this. i lost my osd and i cant change my input for my monitor bc of stupid app i used to turn off my monitor via ccd functionality. for the love of god if you know how can i fix it plz help me.

rm-killer, 2025-11-27:
> [@rm-killer](https://github.com/rm-killer) I'm afraid I don't know how to reenable the OSD on your monitor. (If you figure it out, please post your solution here for others to see.) However, I do want to help others avoid getting into this situation.
> 
> Since release 1.3.0, **ddcutil detect** has issued a warning against trying to set feature values for xiaomi monitors with name "Mi Monitor". What is the model name of your monitor (reported by **ddcutil detect**)? If it's different from "Mi Monitor", perhaps this warning should be issued for all xiaomi monitors.
> 
> More strictly, **ddutil** could refuse to set feature values for these monitors. Thoughts?

@rockowitz 
Good idea. 
I will tell you guys if i could solve it but i don't think so. all i did was to write an script to put my monitor in the sleep mode so the monitor detect the second signal and switched to it. with this method i can atleast switch inputs.
i never wanted to use ddcutil software. all of this happened bc of Anydesk problem with turned off monitor. i wanted to being able use anydesk even when my monitor was off. so i found out in google search that we can use ddcutil app and then this happened. but the better solution was just using virtual monitor. i wish i  saw that methos first. 
i dont have the exact model number of my monitor. cause i didnt keep the box and the manual. i bought this monitor 3 years ago. its curved 34 inch monitor with Ultra wide resolution and 144hrtz. it was a good deal those days for me.

rm-killer, 2025-11-27:
> Command **ddcutil detect** will report the model name from the EDID. Or you can use one of the many other EDID reporting utilities that exist.

im at windows. this commands dont work for me. and even if it works. i dont wanna run any ddc command again. i had enought runing those command.

supermar1010, 2026-10-02:
I managed to fix my monitor, well it was definitely not me but claude doing it. So do use this at your own risk but it worked on mine.

For diagnosis I also used claude, it ran some tests on the monitor including inspecting the firmware of it. This required the monitor to be connected via hdmi (apparently via DP the I2C goes via a the DP controller instead of directly so some stuff is blocked).

This is the gist where claude wrote down all the findings and also a script to fix your monitor use this at your own discretion. i'm not responsible for any broken monitors, it worked on mine though!

https://gist.github.com/supermar1010/3e81e3f23ce16c826e383edbf9442351

Batmaaaaan, 2026-10-02:
> I managed to fix my monitor, well it was definitely not me but claude doing it. So do use this at your own risk but it worked on mine.
>
> For diagnosis I also used claude, it ran some tests on the [...]

THANK YOU SO MUCH!!! I was looking for something like this!!! I'll try ASAP once I'm back home!!

Batmaaaaan, 2026-10-02:
> I managed to fix my monitor, well it was definitely not me but claude doing it. So do use this at your own risk but it worked on mine.
>
> For diagnosis I also used claude, it ran some tests on the [...]

You think I could do this on windows somehow?? I couldn't try anything on Windows, either my knowledge on python scripts is absolute ass or windows was blocking my scripts secretly or something..

supermar1010, 2026-10-02:
I don't know how the whole I2C and ddc works on windows, I'm sorry. I thought I'd share what I found out though, maybe you can get claude to make it run on windows, or use a live linux system from a usb stick to fix it?

Batmaaaaan, 2026-10-02:
> I don't know how the whole I2C and ddc works on windows, I'm sorry. I thought I'd share what I found out though, maybe you can get claude to make it run on windows, or use a live linux system from a usb stick to fix it?

Yeah those were the exact two things I was gonna do!! Thanks!!

Also congratulations!!! You might be the first one to figure out how to fix it fr!!!!

chid, 2026-10-02:
That's a pretty incredible solution. My device from 6 years ago finally reset to factory (I've not yet run it) but just thinking of that in anticipation.

chid, 2026-10-03:
As per in the gist - we now have Linux, Windows and Mac support, confirming that it works well.

https://github.com/chid/xiaomi-curved-34-osd-fix-macos

Batmaaaaan, 2026-10-13:
For 2k 27 Gaming Monitor XMI2701. Here's the fix for Windows: https://github.com/Batmaaaaan/Xiaomi-2K-27-Gaming-Monitor-XMI2701-OSD-Screen-Goes-Black-Fix-for-Windows