rockowitz/ddcutil #154: patch inline: watch_displays_using_udev when it finds zero connected screens crashes

rastermann, 2020-11-20:
Asan found this baby:

```
=================================================================
==62810==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000007091 at pc 0x7f167dd1795e bp 0x7f1678331cc0 sp 0x7f1678331468
READ of size 2 at 0x602000007091 thread T3 (watch_displays)
    #0 0x7f167dd1795d in __interceptor_strlen /build/gcc/src/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:370
    #1 0x7f16787a091a in join_string_g_ptr_array_t /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/util/glib_string_util.c:86
    #2 0x7f167876735c in watch_displays_using_udev /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/ddc/ddc_watch_displays.c:347
    #3 0x7f16784acce0  (/usr/lib/libglib-2.0.so.0+0x7fce0)
    #4 0x7f167d8363e8 in start_thread (/usr/lib/libpthread.so.0+0x93e8)
    #5 0x7f167d764292 in __GI___clone (/usr/lib/libc.so.6+0x100292)

0x602000007091 is located 0 bytes to the right of 1-byte region [0x602000007090,0x602000007091)
allocated by thread T3 (watch_displays) here:
    #0 0x7f167dd8c459 in __interceptor_malloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f16787a54c8 in strjoin /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/util/string_util.c:351
    #2 0x7f16787a08cc in join_string_g_ptr_array /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/util/glib_string_util.c:44
    #3 0x7f16787a090f in join_string_g_ptr_array_t /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/util/glib_string_util.c:85
    #4 0x7f167876735c in watch_displays_using_udev /home/raster/C/other/AUR/ddcutil-git/src/ddcutil/src/ddc/ddc_watch_displays.c:347
    #5 0x7f16784acce0  (/usr/lib/libglib-2.0.so.0+0x7fce0)
```

This turns out to be the debug print:

   GPtrArray * prev_displays = get_sysfs_drm_displays();
   DBGTRC(debug, TRACE_GROUP,
          "Initial connected displays: %s", join_string_g_ptr_array_t(prev_displays, ", ") );

that string join tries to join and array of zero strings.. this fails in strjoin() which happens to malloc a single byte then and never init it at all (so its junk) and so any string stuff can walk off the end of this string array. simple fix - always put a nul byte in. later strcpy's that add in string items will overwrite this. the array is empty because no screens report being connected in /sys/class/drm/cardX/status ... which is itself bizarre, but it is something ddcutil shouldn't crash on.

```
diff --git a/src/util/string_util.c b/src/util/string_util.c
index 0e13aaa0..3f81d4db 100644
--- a/src/util/string_util.c
+++ b/src/util/string_util.c
@@ -349,6 +349,7 @@ char * strjoin( const char ** pieces, const int ct0, const char * sepstr) {

    // printf("(%s) ct=%d, total_length=%d\n", __func__, ct, total_length);
    char * result = malloc(total_length);
+   result[0] = '\0';
    char * end = result;
    for (ndx=0; ndx<ct; ndx++) {
       if (ndx > 0 && seplen > 0) {
```

I don't need attribution - just want to fix this for a user who is finding endless segfaults, so consider this 1 liner patch public domain.

Background - we're using libdductil in enlightenment (we dlopen libddcutil.so2 or libddcutil.so.3 and dlsym it for symbols and then use it from a setuid root slave util to control external monitor brightness just like we control internal laptop panel brightness via sysfs  - so it's automagically enabled as a feature if ddcutil just gets installed at all - this slave util we launch was continually crashing and being restarted for a user when they enabled asan as it found this issue).

enjoy!

rockowitz, 2020-11-20:
Thanks for the precise bug report. I've pushed out a fix to the 1.0.0 
branch.

I also appreciate hearing about how libddcutil is being used. Comments 
on its usability are welcome.

/sys is the proverbial maze of twisty passageways, all alike. I've been 
mucking around in it lately, trying to address the following problem:   
Occasionally the same monitor will appear at 2 different /dev/i2c-N 
devices.  Sometimes both are writable, sometimes only one.  Sometimes 
it's associated with DisplayPort MST and possibly docking stations.

The  version of "ddcutil environment --v" in branch 1.0.0-dev dumps out 
a lot of information about relevant contents.  You may find it helpful 
trying to understand what is going on.  Different drivers seem to use 
/sys differently.

You mention you're looking at /sys/class/drm/cardX/status.  This may 
just be shorthand describing what you're doing.  But note that 
/sys/class/drm is a directory with subdirectories card0, card1, etc.  It 
also contains symbolic links to the members of card0, card1, etc., e.g.  
card0-DP-1 -> card0/card0-DP-1. So drm/card0-DP-1/status exists, but 
drm/card0/status does not.

Regards,
Sanford

rastermann, 2020-11-21:
Oh I was describing what ddcutil was doing when it as walking through /sys/class/drm/cardX devices (card0, card1, card2, etc.). It then walked through the subdirs (outputs) and checked status - it just so happens on this system (not mine - a user's system so I can't reproduce. I worked with them to identify the issue as it was having major adverse effects elsewhere for us). It just so happened this system had no connected outputs there. My machine does - this user' smachine doesn't. It seems it's probably unusual as otherwise you'd have hit this bug earlier. :)

Indeed sysfs is an adhoc maze. I've spent decades dealing with it and its changes too over time... I remember long ago when /proc/apm was simple and all you needed to report battery info... :) That has all changed dramatically over the years. I totally understand your pain. I'm happy to piggy-back off your pain to save some of my own. I actually plan in the future to make a complete monitor configuration Ui based on everything ddcutil can swizzle (basically a kbd+mouse driven OSD config menu on your monitor - not only brightness, but the rest too). I plan to also poll the rotation property and use that to auto-rotate screens (randr). I'm happy ddcutil makes this all possible, so thanks so much for the work you put in. :) I know how hard this kind of stuff is and I hope I just helped make your life a little better by fixing something for you. My way of saying thanks.

Just some comments/asides you may find of interest. It seems ddcutil has been a bit slow in doing things like changing properties. It also regularly even fails to do it - especially if you change backlight right after the monitor woke up from DPMS. I am sure this is a monitor problem, but the slowness I did notice was the usleeps in ddcutil. At the time i reduced the usleep time and found it to work, but a little less reliably. I dedicated a thread just to ddcutil to make sure it doesn't impact anything else. I actually literally fade the backlight in and out when monitors come on or are sent to sleep, or go idle or come out of idle by sending my backlight change requests over the space of a second (If possible it'd be nice it changes every 16ms as I send these request at refresh framerate). I've had to do some hackery to set backlight and then read it back to ensure it worked as it sometimes does not. It'd be nice if this were all solid and reliable, but I've done my workarounds to beat it into shape already.  So just some info in how ddcutili is being abused so you can account for it in future. :)

rockowitz, 2020-11-21:
I've found the location of the sloppy check for status.  It's in a loop 
that watches for displays to be added and removed.   The code to modify 
internal data structures and notify the library's caller of the change 
is not yet written, so at this point it has no effect.  Assuming you're 
building from source, you can disable the loop by commenting out the 
call to ddc_start_watch_displays() in function _ddca_init() at 
approximately line 165 of file src/libmain/api_base.c.

I hadn't considered that the library might be used in a headless 
environment.  Is this some sort of continuous integration context?  Even 
so, I'm surprised that a connector
"node" in sysfs (e.g. card0/card0-HDMI-A-1) exists without a status 
element.  Can I ask you to run "ddcutil environment --verbose" in the 
problematic environment (using the latest 1.0.0-dev branch) and send the 
output as an attachment?  Its probing of /sys should help me understand 
what's going on.

If you're looking to build a UI "based on everything that ddcutil can 
swizzle", that's an enormous number of VCP features whose existence 
varies by  monitor and whose interpretation varies by VCP/MCCS version.  
Most features are not of interest to general users.  I suspect that what 
you really want to do is present a unified interface to monitors that 
implement DDC/CI along with those that don't (i.e. laptop displays), and 
include properties such as xrandr rotation that are outside of DDC/CI.   
I assume you've looked at ddcui. The code is ugly (it's my first Qt 
project) but it does try to handle the full set of VCP features. It's 
designed as a "sys admin" type of tool, but it may give you some ideas.

I'm open to  extending the API if that would make your use easier.  The 
API design has been driven by the needs of ddcui, and that's a sample 
size of 1.  Can you point me at the code that's currently using ddcutil? 
I'd like to see how you're using the API and I may have suggestions.

If you're making brightness changes in rapid succession for 
fade-in/fade-out purposes, you probably want to verify the value only at 
the end of the changes.  Also, it probably doesn't matter if some of the 
changes fail so long as the final set succeeds.  It might make sense to 
allow for reducing the DDC/CI spec mandated delay after setting the 
feature value in this case.  Relatedly, though it doesn't address your 
use case, the way that ddcui avoids overloading the API with tiny 
changes to continuous values is by sending a feature change request to 
the API at most once a second when the slider or spin-box is being 
changed, and finally when the slider or spin-box is released. It's a UI 
cheat that seems to work.

Regards,
Sanford

rastermann, 2020-11-22:
That's the odd thing... it's not a headless system. It's got a GPU and monitor plugged in yet this is finding no connected displays in sysfs - it literally reports nothing there, but there is a screen. As it's not my system I'm limited in knowing the exact makeup of it, but I do know this. I asks the user to find any status files in that subtree for an output - there are none. the status files are missing entirely, and yet ddcutil works, so it's swizzling the monitor properties in some other way.

Well when I said "complete UI" for ddc ... I didn't mean literally probe every DDC property val and expose it - go through all the major known ones and provide a good GUI for it. Expand it over time - like modify color temperature, brightness, contrast (already do backlight), That kind of thing. a nice GUI for what you find in the monitor's own OSD controls, but you don't need to use those unusable monitor UI controls. Any UI that just dumbly lists property strings with some value field is invariably a poor UI and at that point - ddcutil on the cmdline will do and be good enough and about as friendly. We already fully cover randr in other screen setup stuff. This is about complementing that with more controls per screen, remembering them and applying them. As we already have all the resolution, layout, rotation configuring - it makes sense to listen to rotation info and apply accordingly as well as other monitor controls available as above.

Relevant libddcutil using code here:

https://git.enlightenment.org/core/enlightenment.git/tree/src/bin/system/e_system_ddc.c - it hooks into https://git.enlightenment.org/core/enlightenment.git/tree/src/bin/system/e_system_main.c with i/o to calling process stdio handled by https://git.enlightenment.org/core/enlightenment.git/tree/src/bin/system/e_system_inout.c and shared header https://git.enlightenment.org/core/enlightenment.git/tree/src/bin/system/e_system.h (there are other subsystems inside that dir that hook in to this slave setuid root util for other services too).

I'v pretty cleanly sliced it in with  thread hiding the latency and what not, but you can tell I've had to do some workarounds failure like with the probing. You can see the API and code is actually generic allowing getting/setting any property async (the animation is handled outside this slave fromt he calling process so it usses lots of property sets in a row). That code is https://git.enlightenment.org/core/enlightenment.git/tree/src/bin/e_backlight.c (which also handles swizzling backlight via sysfs and xrandr as well as ddc - ddc and sysfs talk to this setuid slave).

I know the UI cheat you do - common and I've done it many times, but the point here is to actually try smoothly fade/animate. I think I wrote this code before ddcutil had an api to change the usleep time - thus i experimented with manually modifying it. I've been using ddcutil's library api now for quite a while. :)

rockowitz, 2020-11-22:
> That's the odd thing... it's not a headless system. It's got a GPU and
> monitor plugged in yet this is finding no connected displays in sysfs
> - it literally reports nothing there, but there is a [...]
ddcutil finds DDC capable monitors by scanning the /dev/i2c devices.   
It ignores SMBus devices  and those I2C devices whose name is not of the 
form /dev/i2c-N.  It first tries to read the EDID, and if one is found 
it attempts DDC/CI communication, which is an expensive operation 
because of the sleeps.

The sysfs status error occurs in the loop that watches for monitors 
being connected or disconnected.  If ddcutil was built with udev, the 
loop waits on a monitor change event.  Otherwise, it simply polls the 
list of monitors the monitors reported in /sys every three seconds.  If 
not, it polls the list of monitors every 3 seconds. These are cheap 
operations.  Only in the rare case of a monitor being added or removed 
are the expensive /dev/i2c checks required.

Again, I'd really appreciate it if "ddcutil environment --verbose" can 
be run on the problematic system, built from the current 1.0.0-dev 
branch. I recently added code that reports relevant parts of the /sys 
tree, and will hopefully indicate how it is that there's no status element.
> Well when I said "complete UI" for ddc ... I didn't mean literally
> probe every DDC property val and expose it - go through all the major
> known ones and provide a good GUI for it. Expand it over [...]
A few comments on the code.

It looks like you're assuming that continuous features have values in 
the range 0..100.  This is common, but not fixed.  The maximum value has 
to be obtained from the mh/ml bytes of the getvcp response. 0..255 is 
frequently seen, and I even had a ddcui bug report from a user whose 
monitor actually used values greater than 255.

The code verifies that the value read for a feature value is the same as 
the value set, modulo  floating point conversion.  However, it's 
sometimes the case that the final value is off by 1, e.g. if a value of 
95 is set, the subsequent read of the feature might return 94 or 96. 
While you may want to hack for it in the UI, there's no point in 
treating the off by 1 state as a DDC error. (I suspect that this is 
because of floating point conversions in the monitor.)

The binary EDID  is not perfectly reliable as a unique monitor 
identifier.  I've seen one monitor, a LG 27MU67, which has neither an 
ASCII nor a binary serial number.  So if 2 of these monitors were 
manufactured in the same week, they would have identical EDIDs. It's 
also possible that the user has loaded identical custom EDIDs into 
multiple monitors. Finally, there's the situation that led to me to 
write the code to examine /sys: occasionally the same monitor will 
appear as 2 different /dev/i2c devices.  Hopefully I will figure out 
what's going on sufficiently to submit a bug report, or at least to work 
around it, but until then 2 /dev/i2c devices with the same EDID are 
possible.

Try calling:

     ddca_enable_sleep_suppression(true)

This experimental option improves performance by eliminating some 
sleeps, and does not appear to increase  the number of retries. If that 
holds true, I expect it will become the default.

Regards,
Sanford

rastermann, 2020-11-25:
Hey - back. I was waiting to get in touch with the user again. here is his output:

```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

The following tests probe the runtime environment using multiple overlapping methods.
   Current time (local): 2020-11-25 12:49:03 PST
   Current time (UTC):   2020-11-25 20:49:03
   Seconds since boot:   599117

*** Basic System Information ***

ddcutil version: 0.9.9

/proc/version:
   Linux version 5.9.0-2-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.0-16) 10.2.0, GNU ld (GNU Binutils for Debian) 2.35.1) #1 SMP Debian 5.9.6-1 (2020-11-08)

Architecture:     x86_64
Distributor id:   Devuan
Release:          testing/unstable
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/boot/vmlinuz-5.9.0-2-amd64 root=UUID=9f612bdc-2265-4916-bc17-a96756a424ff ro libata.force=4.00:noncq

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 10.2.0

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Byte Order:                      Little Endian
   Address sizes:                   48 bits physical, 48 bits virtual
   CPU(s):                          4
   On-line CPU(s) list:             0-3
   Thread(s) per core:              1
   Core(s) per socket:              4
   Socket(s):                       1
   NUMA node(s):                    1
   Vendor ID:                       AuthenticAMD
   CPU family:                      16
   Model:                           4
   Model name:                      AMD Phenom(tm) II X4 975 Processor
   Stepping:                        3
   CPU MHz:                         800.000
   CPU max MHz:                     3600.0000
   CPU min MHz:                     800.0000
   BogoMIPS:                        7224.21
   Virtualization:                  AMD-V
   L1d cache:                       256 KiB
   L1i cache:                       256 KiB
   L2 cache:                        2 MiB
   L3 cache:                        6 MiB
   NUMA node0 CPU(s):               0-3
   Vulnerability Itlb multihit:     Not affected
   Vulnerability L1tf:              Not affected
   Vulnerability Mds:               Not affected
[... 583 lines cut ...]
      Module dev-i2c is required.
      Module dev-i2c is not loaded
   Suggestion:
      Manually load module i2c-dev using the command:
         sudo modprobe i2c-dev
      If this solves the problem, put an entry in directory /etc/modules-load.d
      that will cause i2c-dev to be loaded.  Type "man modules-load.d" for details
```

He's using devuan - so trying to avoid systemd... so that would make it a lot different to my machines. All systemd (for my sanity). They tell me the outputs actually are DP -> DVI cables. It still worked so I assumed it found it some other way. I noticed I had to modrpboe i2c-dev to have it work... thus why this is done from our end in the setuid root too. :) Anyway...

> It looks like you're assuming that continuous features have values in the range 0..100.  This is common, but not fixed.  The maximum value has to be obtained from the mh/ml bytes of the getvcp [...]

Oh... thanks for that. I was going by what my monitors were saying... I didn't have any obvious information tell me otherwise. I'll fix this. I've made a note in the code right now.

> The code verifies that the value read for a feature value is the same as
the value set

Ahhh yes - and this was NOT because of rounding. I actually found sometimes my monitor entirely ignores a request - not rounding. Like I request to set to 30 (current level is 0) and nothing happens. If I just send another request then it works. It's probably a monitor bug. This tends to happen much more if waking up from DPMS sleep just before... this also happens to handle rounding issues like when going to 0 or 100 brightness when I really want it to be there (so end of a fade in or out it ends at the right value). But the issue here is that sometimes it just doesn't like being sent lots of changes rapidly and decides to drop some.

> The binary EDID  is not perfectly reliable as a unique monitor identifier.  I've seen one monitor, a LG 27MU67, which has neither an ASCII nor a binary serial number.  So if 2 of these monitors were manufactured in the same week, they would have identical EDIDs.

we base all our monitor detection on EDID binary hex ... PLUS output name. so the entire screen handling is EDIDHEX/OUTPUT ... this accounts for the exact same EDID but attached to a different output... but for ddc .... we don't know the output name to match this up. I couldnt see any other way to do this. FYI both my monitors are Samsung monitors I bought at the same time... they are 2 serial numbers apart.... their EDID's have this embedded and so don't match. They ... at least, are well behaved. :) Wel the monitors at home. I haven't been to the office since February ... but they monitors are different models so also don't match. But... what other solutions do we have right now? If we could somehow know the output name too ... the [... cut ...] ame like DP0 or HDMI-1 etc. ... I at least am just now crossing my fingers and hoping people rarely have this identical EDID issue. In this case I guess I'd just say "sorry - this feature won't work for you properly". I guess I should actually put in some identical EDID detection code and at least warn them of pending doom... :)

I'll try turn on ddca_enable_sleep_suppression() as I have the above "check if value set succeeded and retry" anyway that seems necessary that would handle error cases.

rastermann, 2020-11-26:
oh ... sorry wrong dump from ddcutil. here:

```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Open failed for /dev/i2c-3: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-4: errno=EACCES(13): Permission denied
Open failed for /dev/i2c-5: errno=EACCES(13): Permission denied
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

The following tests probe the runtime environment using multiple overlapping methods.
   Current time (local): 2020-11-26 12:15:07 AEST
   Current time (UTC):   2020-11-26 02:15:07
   Seconds since boot:   8748

*** Basic System Information ***

ddcutil version: 0.9.9

/proc/version:
   Linux version 5.9.9-arch1-1 (linux@archlinux) (gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.1) #1 SMP PREEMPT Wed, 18 Nov 2020 19:52:04 +0000

Architecture:     x86_64
Distributor id:   Arch
Release:          rolling
Found a known architecture

/proc/cmdline:
   initrd=\amd-ucode.img initrd=\initramfs-linux.img root="UUID=0e426a15-59bd-431a-b3a1-ee157ac0428c" rw

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 10.2.0

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Byte Order:                      Little Endian
   Address sizes:                   43 bits physical, 48 bits virtual
   CPU(s):                          4
   On-line CPU(s) list:             0-3
   Thread(s) per core:              1
   Core(s) per socket:              4
   Socket(s):                       1
   NUMA node(s):                    1
   Vendor ID:                       AuthenticAMD
   CPU family:                      23
   Model:                           1
   Model name:                      AMD Ryzen 3 1300X Quad-Core Processor
   Stepping:                        1
   Frequency boost:                 enabled
   CPU MHz:                         3396.633
   CPU max MHz:                     3500.0000
   CPU min MHz:                     1550.0000
   BogoMIPS:                        7002.84
   Virtualization:                  AMD-V
   L1d cache:                       128 KiB
   L1i cache:                       256 KiB
   L2 cache:                        2 MiB
   L3 cache:                        8 MiB
[... 709 lines cut ...]
   group_i2c_checked:             true
   group_i2c_exists:              false
   dev_i2c_common_group_name:     root
   all_dev_i2c_has_group_i2c:     false
   any_dev_i2c_has_group_i2c:     false
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       false
   cur_uname:                     matheros
   cur_uid:                       1000
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        false
   cur_user_all_devi2c_rw:        false

Configuration suggestions:
   Issue:
      Group i2c does not exist.
   Suggestion:
      Create group i2c. To create group i2c, use command:
         sudo groupadd --system i2c
      Assign /dev/i2c-N devices to group i2c by adding a rule to /etc/udev/rules.d
      Add the current user to group i2c:
         sudo usermod -G i2c -a <username>
      After this, you will have to logout and login again.
      The changes to the user's group list are not read until a new login.
```

rockowitz, 2020-11-30:
Is it possible for user gnarface to rerun "ddcutil environment --verbose" built from the 1.0.0-dev branch?  That is the version that contains the detailed analysis of /sys.  Also, the command needs to be run as root, as that is needed for the extended checks for the "--verbose" option. 

It's unclear to me why you sent the second listing for user matheros.  Does this system also exhibit the missing status attribute?  Again, I would need the detailed analysis of /sys provided by "ddcutil environment --verbose" in branch 1.1.0-dev. And again, the command should be run as root. 

Please send voluminous output as attachments rather than inline.  They're hard to make sense of without copying the output to separate files, and more importantly they make it hard to follow the thread of an issue. 

Re your additional comments: 

For our purposes the major difference between VGA/DVI/HDMI and DisplayPort is that the former use dedicated pins for I2C communication, while the latter carries the logical I2C signal over its AUX channel.  So a DP-DVI adapter will have to perform a conversion.  I believe this is one of the differences between active vs passive adapters, but I have not seen this documented and have not tested to confirm. 

I agree that it makes sense to punt on the very unlikely case of multiple monitors with the same EDID.  However, at least for drm devices it is possible to associate a connector/edid pair with an I2C device by examining /sys/class/drm/cardN-connector. Attribute edid is, well, the EDID.  For DisplayPort connectors the i2c bus is attribute i2c-N. For non-DisplayPort connectors, the i2c bus is found at ddc/i2c-dev/i2c-N.

rockowitz, 2020-11-30:
A further follow up re lost setvcp requests.  For getvcp, the host sends a getvcp request packet to the display.   As part of the protocol for sending an I2C packet,  a sequence of ACKs and NACKs ensure that the packet was delivered.  The display subsequently sends a response packet to the host, which includes a flag indicating whether the feature is supported, along with its value.  Or it may return a DDC Null Message, indicating that it can't respond, e.g. if it doesn't know what to do with the request.

For setvcp, there is only the request packet sent from the host to the display.  There is no response packet.  So all the host knows is that the request was delivered, not that it was processed.  If the requests arrive too quickly, unless the display has an internal request queue of sufficient size, some requests will not be processed properly, i.e. some setvcp requests will fail.

rastermann, 2020-12-17:
matheros was the right user - I mixed up 2 users who are using ddcutil. gnarface didn't have the issues - matheros did. I asked them today (they've been away for a bit) - they are using the ddcutil-git AUR pkg - so this should be latest and greatest for you, right?

[out.txt](https://github.com/rockowitz/ddcutil/files/5708653/out.txt)

As for DDC signals over DP - for me at least its not DVI -> DP. It's direct DP to DP - no DVI conversion cable involved. Well it was until yesterday. One of my screens flat-out died and doesn't power on - the other is on the blink (can't use DPMS...). I've now had to buy 2 replacements and they are now HDMI only (well in theory I could use USB-C alt-mode with DP ... if my GPU had that output... it doesn't)

I found that in some cases like when the monitor just powers up (eg. from dpms sleep) it may drop setvcp's and ignore them for some period of time. Thus why I started doing gt's and re-tries to see if it worked. :)

rockowitz, 2020-12-17:
@rastermann File out.txt contains output for the 0.9.9 production release,  This is the master branch in this git repository.  The additional code exploring /sys is only in the 1.0.0-dev development branch.   Also, for some reason the out.txt file ends in what should be the middle of the output.

rastermann, 2020-12-17:
you you do your git upside down to us - master for us is dev, and stable releases in branches... hmm i have to convince them to swtich branches in their aur build..