rockowitz/ddcutil #156: Acer ED320QR weird input mapping

parkerlreed, 2020-11-29:
Lastest ddcutil from master

The input labels vs physical vs what is reported are all screwed up. 

```
   Feature: 60 (Input Source)
      Values:
         01: VGA-1
         0f: DisplayPort-1
         11: HDMI-1
----
[parker@wolfcola ~]$ sudo ddcutil setvcp 60 0x01                            <-- Physically DisplayPort
[parker@wolfcola ~]$ sudo ddcutil getvcp 60     
VCP code 0x60 (Input Source): DisplayPort-1 (sl=0x0f)   
[parker@wolfcola ~]$ sudo ddcutil setvcp 60 0x0f                            <-- Physically HDMI 2
[parker@wolfcola ~]$ sudo ddcutil getvcp 60     
VCP code 0x60 (Input Source): HDMI-1 (sl=0x11)
[parker@wolfcola ~]$ sudo ddcutil setvcp 60 0x11                            <-- Physically HDMI 1
[parker@wolfcola ~]$ sudo ddcutil getvcp 60     
VCP code 0x60 (Input Source): HDMI-2 (sl=0x12)
```

Detect verbose
```
[parker@wolfcola ~]$ sudo ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Display 1
   I2C bus:             /dev/i2c-6
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true 
      /sys/bus/i2c/devices/i2c-6/name:   AMDGPU DM aux hw bus 0
   EDID synopsis:
      Mfg id:           ACR
      Model:            ED320QR S
      Serial number:    TJNAA0023W01
      Manufacture year: 2020
      EDID version:     1.4
      Product code:     0
      Binary sn:        42999500 (0x02901ecc)
      Extra descriptor: 
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.312, 0.329
      Red   x,y:        0.648, 0.330
      Green x,y:        0.319, 0.617
      Blue  x,y:        0.150, 0.071
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 04 72 00 00 cc 1e 90 02   .........r......
      +0010   1d 1e 01 04 a5 46 27 78 3b 2c 91 a6 54 51 9e 26   .....F'x;,..TQ.&
      +0020   12 50 54 af cf 00 d1 c0 81 80 31 68 31 7c 45 68   .PT.......1h1|Eh
      +0030   45 7c 61 68 61 7c 02 3a 80 18 71 38 2d 40 58 2c   E|aha|.:..q8-@X,
      +0040   45 00 ba 89 21 00 00 1e 00 00 00 fd 00 30 a5 c1   E...!........0..
      +0050   c1 29 01 0a 20 20 20 20 20 20 00 00 00 fc 00 45   .)..      .....E
      +0060   44 33 32 30 51 52 20 53 0a 20 20 20 00 00 00 ff   D320QR S.   ....
      +0070   00 54 4a 4e 41 41 30 30 32 33 57 30 31 0a 01 17   .TJNAA0023W01...
   VCP version:         2.2
   Controller mfg:      RealTek
   Firmware version:    0.1
   Monitor returns DDC Null Response for unsupported features: false
```

rockowitz, 2020-11-29:
Let me be sure I'm reading this right:

The display is a 32" Acer curved monitor.  Per the  listing at Walmart, 
it has 1 DisplayPort and 2 HDMI  connections.

The capabilities string reports that feature x60 supports 3 values, x01, 
x0f, and x11.  Per the MCCS spec, these values represent VGA, 
DisplayPort-1 and  HDMI-1 and respectively.

"setvcp 60 x01" sets the actual value, as reported by the subsequent 
"getvcp 60" as x0f.
"setvcp 60 x0f" sets the actual value, as reported by the subsequent 
"getvcp 60", ax x11.
"setvcp 60 x11" sets the actual value, as reported by the subsequent 
"getvcp 60" , as x12.

That seems very screwed up.

But you only have 1 monitor, in which case I would expect "setvcp 60" to 
either have no effect,
or to set the input source to a non-existent source in 2 out of 3 
cases.  How was this test done?

Finally, I'm unsure how to interpret your "Physically DisplayPort", etc. 
tags.  If they were on the getvcp output lines I would take them to 
indicate the observed state of the monitor for that getvcp output.  But 
what do they mean on the setvcp lines?

parkerlreed, 2020-11-29:
It sets to a valid source in all three cases. 

The "<-- Physically" is what the actual input is when it switches.

So 

And setting those three values the getvcp return value is different in every case

         01: VGA-1 returns DisplayPort-1 (sl=0x0f)
         0f: DisplayPort-1 returns HDMI-1 (sl=0x11)
         11: HDMI-1 returns HDMI-2 (sl=0x12)

My end goal is using the getvcp and setvcp to setup a script to toggle through the inputs (The monitor itself makes you go through a menu and 4 button presses just to change input)

parkerlreed, 2020-11-29:
My comment got screwed up on the first save. Not sure if it sends edits to emails.

parkerlreed, 2020-11-29:
Well I got the script working. I was going to work with the return values from getvcp but since those don't match I'm just keeping track of what input I'm on in a file

So is the weirdness just Acer not following the spec? (I wouldn't be surprised)

```
#!/bin/bash
current=$(cat /home/parker/bin/current)
if [[ $current == "01" ]]; then
        sudo ddcutil setvcp 60 0x11 
        echo -n "11" > /home/parker/bin/current
        exit
elif [[ $current == "11" ]]; then
        sudo ddcutil setvcp 60 0x0f
        echo -n "0f" > /home/parker/bin/current
        exit
elif [[ $current == "0f" ]]; then
        sudo ddcutil setvcp 60 0x01
        echo -n "01" > /home/parker/bin/current
        exit
fi
```

rockowitz, 2020-11-30:
The --brief option causes getvcp to emit its response in machine readable form.  For example: 
~~~
$ ddcutil getvcp 60 --brief
VCP 60 SNC x03
~~~

You can thus get the feature value into a script variable without having to maintain the value in a file. 
~~~
current=$(ddcutil getvcp 60 --brief |  cut -f 4 -d " " )
~~~

To make it clearer for me what is going on, please run the following variant of your script: 
~~~
#!/bin/bash
current=$(ddcutil getvcp 60 --brief |  cut -f 4 -d " " )
echo "Initial feature 60 value: $current"
if [[ $current == "x01" ]]; then
        echo "setting x11"
        sudo ddcutil setvcp 60 0x11 
elif [[ $current == "x11" ]]; then
        echo "setting x0f"
        sudo ddcutil setvcp 60 0x0f
elif [[ $current == "x0f" ]]; then
        echo "setting x01"
        sudo ddcutil setvcp 60 0x01
 else 
        echo "Not setting feature 60"
fi
current=$(ddcutil getvcp 60 --brief |  cut -f 4 -d " " )
echo "Final feature 60 value: $current"
~~~

parkerlreed, 2020-11-30:
I had to add a sleep after the if/else statement because it tried to run ddcutil too soon and the monitor doesn't respond to commands mid input switch.

It switched but went from DisplayPort to DisplayPort (Since 01 is supposed to be VGA but is actually DisplayPort)

It seems the getvcp return values are just fubar. 

```
[parker@wolfcola bin]$ ./input-test 
Initial feature 60 value: x0f
setting x01
Final feature 60 value: x0f
```

Attaching probe log for posterity.

[probe.txt](https://github.com/rockowitz/ddcutil/files/5616221/probe.txt)

rockowitz, 2020-11-30:
I should consider increasing the sleep after "setvcp 60" as a special case. 

Keep in mind that the monitor's capabilities string is purely (mis)informational - it is often wrong. ddcutil does not rely on it.  

Is it ever the case that "getvcp 60" returns x01?  

The way I interpret the script and the probe output is that there is one monitor, which is connected using its DisplayPort connector.   There's no other monitor to switch to, so "setvcp 60 x01" can have no effect,  The initial value of "getvcp 60" is x0f (DisplayPort) and the final value is x0f (DisplayPort).

parkerlreed, 2020-12-02:
The monitor responds to DDC commands on DisplayPort no matter what input is on. 

I ended up returning the monitor anyways due to other circumstances so I guess this can be closed out if you want.

closed 2020-12-02 by rockowitz