rockowitz/ddcutil #157: Monoprice Dark Matter 40776 input labels incorrect

ethack, 2020-11-30:
This is intended for http://www.ddcutil.com/monitor_notes/ if you think it is appropriate.

I have a Monoprice Dark Matter 40776. Product page [here](https://www.monoprice.com/product?p_id=40776). The inputs are mislabeled from which physical port they actually correspond to. Unlike https://github.com/rockowitz/ddcutil/issues/156 the hex values in `setvcp` and `getvcp` do match each other. They just have the wrong human-readable label as well as not matching up with the reported capabilities.

This monitor has 2x HDMI ports and 2x DisplayPorts. Which is in line with what `capabilities` reports.

```
   Feature: 60 (Input Source)
      Values (unparsed):  11 12 0F 10
      Values (  parsed):
         11: HDMI-1
         12: HDMI-2
         0f: DisplayPort-1
         10: DisplayPort-2
```

However, using `setvcp` with the above values has no effect and and `getvcp` reports different values for the active inputs. 

Instead, the following are the hex values that I can use with `setvcp` to switch inputs and what `getvcp` reports when that input is active.
- HDMI_1=0x05
- HDMI_2=0x06
- DISPLAYPORT_1=0x07
- DISPLAYPORT_2=0x08 

```
# Currently on HDMI 1 input
$ ddcutil getvcp 0x60     
VCP code 0x60 (Input Source                  ): Composite video 1 (sl=0x05)

# Currently on HDMI 2 input
$ ddcutil getvcp 0x60         
VCP code 0x60 (Input Source                  ): Composite video 2 (sl=0x06)

# Currently on DisplayPort 1 input
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): S-Video-1 (sl=0x07)

# Currently on DisplayPort 2 input
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): S-Video-2 (sl=0x08)
```

Additionally, I can only switch inputs by using `ddcutil setvcp 0x60` from the active input. E.g.
I have inputs connected to HDMI 2 (0x06) and DisplayPort 1 (0x07). If HDMI 2 (0x06) is active then I can switch to DisplayPort 1 (0x07) by running this command on the system connected to HDMI 2 (0x06).

```
# Switch to DisplayPort 1
$ ddcutil setvcp 0x60 0x07
```

But in the same scenario if I run the same command from the system connected to DisplayPort 1 (0x07) (which is inactive) then it has no effect and the monitor does not switch inputs from  HDMI 2 (0x06) to DisplayPort 1 (0x07).

Likewise, I cannot query the active input from an inactive system. Only the active input gets a valid response. This is what I get running the command from an inactive input.

```
$ ddcutil getvcp 0x60
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
VCP code 0x60 (Input Source                  ): Invalid response. status code=-3011, Display_Handle[i2c: fd=3, busno=5]
```

---

ddcutil 0.9.9 installed using Launchpad PPA.
Pop!_OS 20.04 (Ubuntu derivative)

Here's some raw output. Let me know if you'd like any others. (Note: I had two other monitors when I ran these.) 
```
$ ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

<<<2 other monitors clipped>>>

Display 3
   I2C bus:             /dev/i2c-5
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true 
      /sys/bus/i2c/devices/i2c-5/name:   NVIDIA i2c adapter 6 at 1:00.0
   EDID synopsis:
      Mfg id:           LHC
      Model:            34CHR
      Serial number:    
      Manufacture year: 2020
      EDID version:     1.3
      Product code:     52
      Binary sn:        0 (0x00000000)
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.670, 0.330
      Green x,y:        0.210, 0.710
      Blue  x,y:        0.140, 0.080
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 31 03 34 00 00 00 00 00   ........1.4.....
      +0010   1f 1e 01 03 80 50 21 78 aa af e5 ab 54 35 b5 23   .....P!x....T5.#
      +0020   14 50 54 a5 4b 00 71 4f 81 c0 81 40 81 80 95 00   .PT.K.qO...@....
      +0030   a9 c0 b3 00 d1 c0 f5 7c 70 a0 d0 a0 29 50 30 20   .......|p...)P0 
      +0040   35 00 1d 4e 31 00 00 1a 21 68 70 a0 d0 a0 29 50   5..N1...!hp...)P
      +0050   30 20 35 00 1d 4e 31 00 00 1a 00 00 00 fd 00 30   0 5..N1........0
      +0060   65 a0 a0 3c 01 0a 20 20 20 20 20 20 00 00 00 fc   e..<..      ....
      +0070   00 33 34 43 48 52 0a 20 20 20 20 20 20 20 01 97   .34CHR.       ..
   VCP version:         2.1
   Controller mfg:      Mstar
   Firmware version:    0.0
   Monitor returns DDC Null Response for unsupported features: false
```

```
$ ddcutil -d 3 capabilities --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Unparsed capabilities string: (prot(monitor)type(lcd)MStarcmds(01 02 03 07 0C E3 F3)vcp(02 04 05 08 10 12 14(05 08 0B 0C) 16 18 1A 52 60( 11 12 0F 10) AA(01 02) AC AE B2 B6 C6 C8 C9 D6(01 04 05) DC(00 02 03 05 ) DF FD)mccs_ver(2.1)mswhql(1))
Model: Not specified
MCCS version: 2.1
VCP Features:
   Feature: 02 (New control value)
   Feature: 04 (Restore factory defaults)
   Feature: 05 (Restore factory brightness/contrast defaults)
   Feature: 08 (Restore color defaults)
   Feature: 10 (Brightness)
   Feature: 12 (Contrast)
   Feature: 14 (Select color preset)
      Values (unparsed): 05 08 0B 0C
      Values (  parsed):
         05: 6500 K
         08: 9300 K
         0b: User 1
         0c: User 2
   Feature: 16 (Video gain: Red)
   Feature: 18 (Video gain: Green)
   Feature: 1A (Video gain: Blue)
   Feature: 52 (Active control)
   Feature: 60 (Input Source)
      Values (unparsed):  11 12 0F 10
      Values (  parsed):
         11: HDMI-1
         12: HDMI-2
         0f: DisplayPort-1
         10: DisplayPort-2
   Feature: AA (Screen Orientation)
      Values (unparsed): 01 02
      Values (  parsed):
         01: 0 degrees
         02: 90 degrees
   Feature: AC (Horizontal frequency)
   Feature: AE (Vertical frequency)
   Feature: B2 (Flat panel sub-pixel layout)
   Feature: B6 (Display technology type)
   Feature: C6 (Application enable key)
   Feature: C8 (Display controller type)
   Feature: C9 (Display firmware level)
   Feature: D6 (Power mode)
      Values (unparsed): 01 04 05
      Values (  parsed):
         01: DPM: On,  DPMS: Off
         04: DPM: Off, DPMS: Off
         05: Write only value to turn off display
   Feature: DC (Display Mode)
      Values (unparsed): 00 02 03 05 
      Values (  parsed):
         00: Standard/Default mode
         02: Mixed
         03: Movie
         05: Games
   Feature: DF (VCP Version)
   Feature: FD (manufacturer specific feature)
```

rockowitz, 2020-11-30:
Thank you for the clear report. This monitor definitely belongs in the rogues' gallery of problematic monitors. 

Some monitors respond to setvcp commands on an input other than the currently active input, others do not.  What's  more unusual here is that the monitor does not respond to a getvcp request unless it comes on the currently active input.  Is this the case for features in general, or just for feature x60? 

I'd appreciate it if you could run "ddcutil detect --verbose" and "ddcutil probe" and send the output as attachments.  Attachments are easier to work with, and their use makes it easier to follow the comments on an issue. 

-- Sanford

ethack, 2020-11-30:
I tried a couple more things just now and they didn't work from the inactive input. I've only really used the input switching feature yet but I may try to experiment with color profiles and Picture by Picture later on. Btw, thank you very much for an awesome and well-documented tool! My biggest concern about this monitor was the 5+ button pushes to switch inputs and thanks to ddcutil I have a keyboard hotkey that toggles inputs. My biggest concern turned into one of my favorite features!

```
# From active input
$ ddcutil getvcp 0xc0
VCP code 0xc0 (Display usage time            ): Usage time (hours) = 85 (0x000055) mh=0xff, ml=0xff, sh=0x00, sl=0x55

# From inactive input
$ ddcutil getvcp 0xc0
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
VCP code 0xc0 (Display usage time            ): Invalid response. status code=-3011, Display_Handle[i2c: fd=3, busno=5]
```

```
# From active input
$ ddcutil getvcp 0x10
VCP code 0x10 (Brightness                    ): current value =    90, max value =   100
$ ddcutil setvcp 0x10 50
# Screen brightness successfully changed the it dimmed.

# From inactive input
$ ddcutil getvcp 0x10   
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
VCP code 0x10 (Brightness                    ): Invalid response. status code=-3011, Display_Handle[i2c: fd=3, busno=5]
$ ddcutil setvcp 0x10 50
Setting value failed for feature 10. rc=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message
# No result. Screen brightness not changed.
```

Here is the output your requested:
[ddcutil-detect-verbose.txt](https://github.com/rockowitz/ddcutil/files/5619302/ddcutil-detect-verbose.txt)
[ddcutil-probe.txt](https://github.com/rockowitz/ddcutil/files/5619303/ddcutil-probe.txt)

And for good measure, here is the output from an inactive input:
[ddcutil-detect-verbose-inactive.txt](https://github.com/rockowitz/ddcutil/files/5619343/ddcutil-detect-verbose-inactive.txt)
[ddcutil-probe-inactive.txt](https://github.com/rockowitz/ddcutil/files/5619344/ddcutil-probe-inactive.txt)

rockowitz, 2020-12-01:
Based on your reports, I've created an entry for your display on page [Notes on Specific Monitors](https://www.ddcutil.com/monitor_notes#monoprice-dark-matter-40776).  Let me know if it looks right to you. 

If you want to use ddcutil for color adjustment or access to features like Picture By Picture, bear in mind that you're likely to see baffling interactions among VCP features.  Monitor settings have outstripped what was envisioned in the MCCS spec.  For example, I have a Samsung U32H750 that has complicated interactions between feature xdc (display mode), feature x14 (select color preset) and various specific settings such as x16 (red gain).  And some features such as Picture By Picture appear to  be managed only thorough the OSD. I've not found any VCP feature code, including manufacturer reserved codes, that affect them.

ethack, 2020-12-01:
Great, thanks for your advice and the new entry!

One small typo I noticed:
> x07 - DisplayPort-2

Should be:
> x08 - DisplayPort-2

The only other suggestion I might make would be to add the word "only" to this sentence. It might be implied based on the previous sentence but doesn't hurt to stress it even more.
> It is also the case that getvcp and other queries **only** work when issued on the current input source.

rockowitz, 2020-12-01:
Fixed, thanks.

closed 2020-12-02 by ethack