rockowitz/ddcutil #158: Setvcp command not parsing minus shortcut

yuannan, 2020-12-13:
ddcutil setvcp 10 +10 works fine
ddcutil setvcp 10 -10 does not

proper syntax should have a space between the operator and the number but + does not require it while - does.

rockowitz, 2020-12-16:
Fixed in branch 1.0.0-dev. 

Fixing this bug has caused me to reconsider the syntax for specifying a relative value.  If spaces were not required between the "+" or "-" and the value, the rule would be that e.g. "setvcp 10 +5" indicates a relative value, while "setvcp 10 5" indicates an absolute value. The existing form with a space between the "+" or "-" and the value would continue to work. Thoughts?

yuannan, 2020-12-17:
> Fixed in branch 1.0.0-dev.
>
> Fixing this bug has caused me to reconsider the syntax for specifying a relative value. If spaces were not required between the "+" or "-" and the value, the rule would [...]

I think it should have both. With space and without. This keeps the largest amount of compatibility and user friendliness just in case other have had scripts and other none dynamic services running that may have this as a dependency. The extra lines in the code is not gonna make a difference at runtime but this will.

closed 2021-01-02 by rockowitz