rockowitz/ddcutil #162: getvcp/setvcp 0x60 failed with Philips 328P6VUBREB/11

sattoke, 2020-12-20:
# Environment 

## Hosts
I'm using ddcutil on a Raspbery Pi 4 and a Raspbery Pi Zero WH.

Both of them did not work properly due to the following problems.
https://github.com/rockowitz/ddcutil/issues/97
However, by using bangom's workarround below, I am able to use it without any problems.
https://github.com/raspberrypi/linux/issues/3152#issuecomment-569510398

According to the latest comments on the above issue, it looks like this Raspbery Pi issue has just been resolved. Excellent.

## Monitor
My monitor is a Philips 328P6VUBREB/11.
https://www.philips.co.jp/c-p/328P6VUBREB_11/brilliance-lcd-monitor-with-usb-c-dock

The monitor has the following four input sources

- HDMI-1
- HDMI-2
- DisplayPort-1
- USB-C

## ddcutil
Using the latest commit from the 1.0.0-dev branch as of today.

https://github.com/rockowitz/ddcutil/commits/1.0.0-dev 
https://github.com/rockowitz/ddcutil/tree/c2c9d0a3472926c1fc2762d3cdb3d031a6eea079

# Problem
I have been using the above environment with almost no problem, but I noticed that each VCP code may or may not work properly.

## Example of both getvcp and setvcp working properly 0x62 (Audio speaker volume)

```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x62
VCP code 0x62 (Audio speaker volume          ): Fixed (default) level (0x00)
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x62 0x01
(app_set_vcp_value             ) Starting. feature=0x62, new_value=0x01, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x62
VCP code 0x62 (Audio speaker volume          ): Volume level: 1 (00x01)
```

## Example of neither getvcp nor setvcp working properly 0x60 (Input Source)

```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): Unsupported feature code (DDC NULL Message)
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 0x12
(app_set_vcp_value             ) Starting. feature=0x60, new_value=0x12, value_type=VALUE_TYPE_ABSOLUTE, force=false
Invalid hex value
(app_set_vcp_value             ) Done.     Returning: Error_Info[DDCRC_ARG in app_set_vcp_value]
Invalid hex value
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 12
(app_set_vcp_value             ) Starting. feature=0x60, new_value=12, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
```

# Investigation

## VCP version 3.0
I started investigating from the source code about the `0xnn` notation not being available when using `setvcp`. Then I noticed that when using that monitor, it seems to be processed for VCP version 3.0, which is not used very often. In fact, the VCP version obtained by `ddcutil detect` was 3.0.

```
[sat@rp4:~/work/ddcutil]
$ ddcutil detect
Display 1
   I2C bus:  /dev/i2c-1
   EDID synopsis:
      Mfg id:               PHL
      Model:                PHL 328P6VU
      Product code:         2343
      Serial number:
      Binary serial number: 417 (0x000001a1)
      Manufacture year:     2019
      EDID version:         1.3
   VCP version:         3.0

```

So I assumed that the monitor was actually running on 2.2 or something like that, and investigated the rest on that assumption.

(supplementary) After I finished the whole investigation, I noticed that the output of `ddcutil interrogate` seems to be 2.2 in the `MCCS version` obtained by `ddcutil capabilities`.

## `--mccs` option
I tried to specify 2.2 with the `--mccs` option, and the option itself was accepted, but from the source code, it seems to work only with `vcpinfo`.

```
[sat@rp4:~/work/ddcutil]
$ ddcutil --mccs 2.2 getvcp 0x60
VCP code 0x60 (Input Source                  ): Unsupported feature code (DDC NULL Message)
```

## Force change to VCP version 2.2 by hard coding

```diff
[sat@rp4:~/work/ddcutil]
$ git diff
diff --git a/src/ddc/ddc_vcp_version.c b/src/ddc/ddc_vcp_version.c
index b9108745..23fba4b5 100644
--- a/src/ddc/ddc_vcp_version.c
+++ b/src/ddc/ddc_vcp_version.c
@@ -49,6 +49,9 @@
  *    could not be retrieved (pre MCCS v2)
  */
 DDCA_MCCS_Version_Spec get_vcp_version_by_display_handle(Display_Handle * dh) {
+   // for debug
+   return DDCA_VSPEC_V22;
+
    assert(dh);
    bool debug = false;
    // TMI
```

This made getvcp/setvcp 0x60 work as expected.

```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): HDMI-1 (sl=0x11)
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): HDMI-2 (sl=0x12)
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): DisplayPort-1 (sl=0x0f)
[sat@rp4:~/work/ddcutil]
$ ddcutil getvcp 0x60
VCP code 0x60 (Input Source                  ): Invalid value (sl=0x00)
```

```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 0x11
(app_set_vcp_value             ) Starting. feature=0x60, new_value=0x11, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 0x12
(app_set_vcp_value             ) Starting. feature=0x60, new_value=0x12, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 0x0f
(app_set_vcp_value             ) Starting. feature=0x60, new_value=0x0f, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
[sat@rp4:~/work/ddcutil]
$ ddcutil setvcp 0x60 0x13
(app_set_vcp_value             ) Starting. feature=0x60, new_value=0x13, value_type=VALUE_TYPE_ABSOLUTE, force=false
(app_set_vcp_value             ) Done.     Returning: NULL
```

### (Supplemental) `Invalid Value` in `getvcp 0x60`
When USB-C is the input source, `getvcp 0x60` returns `Invalid value`. I haven't been able to track it properly, but it looks like the monitor is actually returning `0x00`, so there's nothing I can do about it.

```
Breakpoint 1, ddc_write_read (dh=0x4cb9d0, request_packet_ptr=0x4cb7e0, read_bytewise=false, max_read_bytes=20, expected_response_type=2 '\002', expected_subtype=96 '`', response_packet_ptr_loc=0xbefff20c) at ddc_packet_io.c:570
570        if (psc >= 0) {
(gdb) x/20bx readbuf
0x4cbea8:       0x6e    0x88    0x02    0x00    0x60    0x00    0x20    0x0e
0x4cbeb0:       0x00    0x00    0xfa    0x28    0x31    0x29    0x61    0x73
0x4cbeb8:       0x73    0x65    0x74    0x5f
```

Address `0x4cbeb1` in `readbuf` after `ddc_write_read_raw()` is set to `0x00`.

https://github.com/rockowitz/ddcutil/blob/c2c9d0a3472926c1fc2762d3cdb3d031a6eea079/src/ddc/ddc_packet_io.c#L562-L569

# Request
It would be nice to be able to force the VCP version to be used for getvcp and setvcp using options such as `--mccs`.

However, since this seems to be a problem specific to the monitor, and not many users may be affected by it, and since passing `--mccs` may affect many functions, I don't intend to make such a strong request.

# Attachment
- [ddcutil_1.0.0_c2c9d0a_interrogate_before.txt](https://github.com/rockowitz/ddcutil/files/5720944/ddcutil_1.0.0_c2c9d0a_interrogate_before.txt)
  - Result of `ddcutil interrogate` with vanilla ddcutil
- [ddcutil_1.0.0_c2c9d0a_interrogate_after.txt](https://github.com/rockowitz/ddcutil/files/5720946/ddcutil_1.0.0_c2c9d0a_interrogate_after.txt)
  - The result of `ddcutil interrogate` with the patch to force VCP version 2.2 mentioned above.

rockowitz, 2020-12-20:
Thank you for the detailed report. There's a lot to digest.

This is the first monitor I've ever encountered that claims to support MCCS 3.0.  But as you note the MCCS value obtained from the capabilities string differers from that returned by feature xDF. 

More to follow.

rockowitz, 2020-12-23:
Your problem report was the impetus to address enhancements that were "on the back burner". 

Commands **vcpinfo**, **capabiilites**, **getvcp**, **setvcp**, **dumpvcp**,  and **watch** now recognize the ***--mccs** option. The effect of the option, of course, depends on what the command does. 

**ddcutil**, for some time now, has had undocumented supported for user supplied feature definitions.  I've hesitated to announce the facility because the semantics, and particularly the format of feature definition files, is subject to change.  A feature definition file for a monitor model specifies feature definitions that add to or override those in the MCCS spec.  It also can optionally specify the MCCS version of the monitor.   The facility is enabled by option ***--udf***.  The default is currently ***--noudf***.  Preliminary documentation is at page [User Defined Features](http://www/ddcutil.com/udf).

Command line option ***--mccs*** takes precedence over a MCCS version in a feature definition file, which in turn overrides the value obtained from feature xDF (VCP Version).  The value in the capabilities string is not part of the algorithm. 

Let me know of any aspects of the design you regard as problematic, as well as any bugs.  Extending and integrating these two features required significant refactoring, and there are undoubtedly as yet unaddressed "corner cases".

Finally, let me say again how much I appreciated the clarity and thoroughness of your problem report.  It encouraged me to address these long standing enhancements. 

Regards, 
- Sanford

sattoke, 2020-12-26:
Thanks for your quick response!
I've tried it.

# `--mccs` option

```sh
[sat@rp4:~/work/ddcutil]
$ git rev-parse HEAD
943f8e7a13f97bfc01544abb1e4f85fb278bba82
[sat@rp4:~/work/ddcutil]
$ ddcutil --mccs 2.2 getvcp 0x60
VCP code 0x60 (Input Source                  ): DisplayPort-1 (sl=0x0f)
[sat@rp4:~/work/ddcutil]
$ ddcutil --mccs 2.2 setvcp 0x60 0x12
[sat@rp4:~/work/ddcutil]
$ ddcutil --mccs 2.2 getvcp 0x60
VCP code 0x60 (Input Source                  ): HDMI-2 (sl=0x12)
```

Great! Both getvcp and setvcp gave me the expected results.

# `--udf` option

Next, I created a user defined feature file as shown below and tried the `--udf` option.

```sh
[sat@rp4:~/work/ddcutil]
$ cat ~/.local/share/ddcutil/PHL-PHL_328P6VU-2343.mccs
MFG_ID       PHL
MODEL        PHL 328P6VU
PRODUCT_CODE 2343
MCCS_VERSION 2.2
FEATURE_CODE 60 Input Source
    ATTRS NC RW
    VALUE 0x11 HDMI-1
    VALUE 0x12 HDMI-2
    VALUE 0x0f DisplayPort-1
    VALUE 0x13 USB-C
```

## Switching input sources
```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf getvcp 0x60
VCP code 0x60 (Input Source                  ): DisplayPort-1 (sl=0x0f)
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf setvcp 0x60 0x12
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf getvcp 0x60
VCP code 0x60 (Input Source                  ): HDMI-2 (sl=0x12)
```

Excellent! This is also exactly what I expected.

## `getvcp` to an input source whose name is undefined in the MCCS specification
```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf setvcp 0x60 0x13
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf getvcp 0x60
VCP code 0x60 (Input Source                  ): Unrecognized value (sl=0x00)
```

Oops, I should have written the following.

```sh
[sat@rp4:~/work/ddcutil]
$ cat ~/.local/share/ddcutil/PHL-PHL_328P6VU-2343.mccs
MFG_ID       PHL
MODEL        PHL 328P6VU
PRODUCT_CODE 2343
MCCS_VERSION 2.2
FEATURE_CODE 60 Input Source
    ATTRS NC RW
    VALUE 0x11 HDMI-1
    VALUE 0x12 HDMI-2
    VALUE 0x0f DisplayPort-1
    VALUE 0x00 USB-C
```

I tried again with this file.

```sh
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf setvcp 0x60 0x13
[sat@rp4:~/work/ddcutil]
$ ddcutil --udf getvcp 0x60
VCP code 0x60 (Input Source                  ): USB-C (sl=0x00)
```

Wonderful!

---

I don't have any problems with these new features as far as I am currently trying to use it.

Thanks a lot!

rockowitz, 2020-12-27:
I have created an [entry](http://www.ddcutil.com/monitor_notes/#phillips_328p6vu) for your monitor on the ddcutil web site.   Let me know if anything should be changed or added.

sattoke, 2020-12-27:
LGTM!

closed 2020-12-27 by rockowitz