rockowitz/ddcutil #164: Acer KA222Q

allmyhardwareinfogobrr, 2020-12-29:
[in.txt](https://github.com/rockowitz/ddcutil/files/5751183/in.txt)
The output of `sudo ddcutil interrogate >in.txt 2>&1` is above. I'm a dual booter and ClickMonitorDDC_7_2 just works on Windows, so no blaming my hardware or monitor settings.

rockowitz, 2020-12-29:
Your DDC communication is corrupt.  The first 4 bytes of the **getvcp** response are repeated.  You will see the DDC data errors more clearly if you execute **ddcutil** with the **--ddc** option.  There are Nvidia driver errors in your logs.  See the journalctl output and /var/log/syslog.  The problem is likely in Nvidia's proprietary driver.  What happens if you drive the monitor with the Vega gpu on the Ryzen chip?  

That ClickMonitorDDC works is not dispositive.  Windows drivers appear to be more tolerant of marginal monitors.  Bear in mind that while monitor manufacturers test with Windows, I doubt they test various Linux configurations.

Finally, I see that driver ddccui is loaded.  I doubt that is causing your problems but it should be unloaded for testing.

allmyhardwareinfogobrr, 2020-12-30:
> What happens if you drive the monitor with the Vega gpu on the Ryzen chip?

Can I? How? My motherboard doesn't have video output, I'm relying on Nvidia's card for that.

(a few commands later)
All is irrelevant. I guess I should have rebooted after doing the `90-nvidia_i2c.conf` magic properly (I initially made a `/etc/X11/xorg.conf.d` file), or you should have written just that in the doc, because I don't see any other reason it wasn't working yesterday but is today. BTW, "ddccui" seems to be a typo of yours. BTW, the `http://www.ddcutil.com/support` link in your README is a 404, and also non-https - disgusting. BTW, does ddcui automate changing brightness based on the time of day? I won't bother with it otherwise. BTW, why is DDC so obscure? BTW, great work, thanks! @rockowitz

closed 2020-12-30 by allmyhardwareinfogobrr

rockowitz, 2020-12-31:
Very minimal motherboard you have there.  Supports the Ryzen chip but I see that it doesn't expose Vega gpu. 

I have added a comment re the need to reboot after updating xorg configuration settings. 

The link to www.ddcutil.com/support has been corrected in README.md in branch 1.0.0-dev.  Unfortunately, GitHub uses the README file in the master branch as the project description, and I have not found a way to change that.  So the project description will remain unchanged until the next release.

Yes, "ddccui" is a typo.  I was referring to driver ddcci from ddccontrol.

Changing brightness based on time of day is not on the enhancement list. **ddcutil** is "simply" a utility for access the monitor's Virtual Control Panel.  It can be scripted, and the shared library can be called from C and C++ programs.

allmyhardwareinfogobrr, 2021-01-01:
Bigger projects, [like this one](https://github.com/ytdl-org/youtube-dl), solve this problem you're describing by just not freezing `master`. MkDocs seems to be barebones enough not to support redirects, but why would you not just keep the legacy page name until next release? Or, you know, forever? Very unusual. Happy New Year.