rockowitz/ddcutil #165: Null deref when parsing PCI

baloo, 2020-12-31:
Just got a null deref when trying to run `ddcutil environement`

```
(gdb) run environment
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /nix/store/0in47i7q3caywbg2kjh94ys1qcxk2xfj-ddcutil-0.9.9/bin/ddcutil environment
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/nix/store/1yvpgm763b3hvg8q4fzpzmflr5674x4j-glibc-2.32-10/lib/libthread_db.so.1".
Open failed for /dev/usb/hiddev0: errno=EACCES(13): Permission denied
The following tests probe the runtime environment using multiple overlapping methods.

*** Basic System Information ***

ddcutil version: 0.9.9

/proc/version:
   Linux version 5.10.4 (nixbld@localhost) (gcc (GCC) 9.3.0, GNU ld (GNU Binutils) 2.31.1) #1-NixOS SMP Wed Dec 30 10:54:29 UTC 2020
[Detaching after vfork from child process 8234]
[Detaching after vfork from child process 8236]
[Detaching after vfork from child process 8238]

Architecture:     x86_64
Distributor id:   (null)
Release:          (null)
Found a known architecture

/proc/cmdline:
   initrd=\efi\nixos\q56fd750dghanv1wab2dqwm8yp3ngn50-initrd-linux-5.10.4-initrd.efi systemConfig=/nix/store/cyc5sch9glggxkwszg93gw6jnmch542r-nixos-system-vic-21.03.git.70a061bd6c5 init=/nix/store/cyc5sch9glggxkwszg93gw6jnmch542r-nixos-system-vic-21.03.git.70a061bd6c5/init systemd.unified_cgroup_hierarchy=0 loglevel=4

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:0a:00.0 (boot_vga flag is set)
   Device class:        x030000    VGA compatible controller

Program received signal SIGSEGV, Segmentation fault.
mlm_get_names2 (mlm=0x0, levelct=levelct@entry=3, ids=ids@entry=0x7fffffff9184) at multi_level_map.c:236
236	in multi_level_map.c
(gdb) bt
#0  mlm_get_names2 (mlm=0x0, levelct=levelct@entry=3, ids=ids@entry=0x7fffffff9184) at multi_level_map.c:236
#1  0x00005555555a100c in devid_get_pci_names (vendor_id=vendor_id@entry=4098, device_id=device_id@entry=26591, subvendor_id=subvendor_id@entry=7586, subdevice_id=subdevice_id@entry=58195, argct=argct@entry=4)
    at device_id_util.c:811
#2  0x000055555557cfb0 in report_device_identification (sysfs_device_dir=sysfs_device_dir@entry=0x7fffffff9290 "/sys/bus/pci/devices/0000:0a:00.0", depth=depth@entry=0) at query_sysenv_sysfs.c:305
#3  0x000055555557d321 in each_video_pci_device (dirname=<optimized out>, fn=0x55555561ef53 "0000:0a:00.0", accumulator=0x5555555f2f50, depth=0) at query_sysenv_sysfs.c:422
#4  0x0000555555579844 in dir_foreach (dirname=dirname@entry=0x5555555b5bbb "/sys/bus/pci/devices", fn_filter=fn_filter@entry=0x0, func=func@entry=0x55555557d141 <each_video_pci_device>,
    accumulator=accumulator@entry=0x5555555f2f50, depth=depth@entry=0) at query_sysenv_base.c:470
#5  0x000055555557d592 in query_card_and_driver_using_sysfs (accum=accum@entry=0x5555555f2f50) at query_sysenv_sysfs.c:532
#6  0x0000555555578006 in query_sysenv () at query_sysenv.c:606
#7  0x0000555555573d42 in main (argc=<optimized out>, argv=<optimized out>) at app_ddcutil/main.c:524
(gdb) print mlm
$2 = (Multi_Level_Map *) 0x0
```

I'm really not used to the code base, and I'm not too sure I can provide a fix for that.

rockowitz, 2020-12-31:
ddcutil did not handle the case where no pci.ids file or usb.ids file was found.  That should be fixed now in branch 1.0.0-dev.  Let me know if the changes work for you.

Do these files really not exist on your system?  If they are, where are they located?

baloo, 2021-01-01:
Ha!
Well, currently it's `/nix/store/24hqvmf93nq74i65zlga3pb31zbh0skz-hwdata-0.335/share/hwdata/pci.ids`, but (in nixos's world) that path is expected to be injected at build-time and hwdata is expected to be declared as a dependency.

I can try to provide a patch for that.

baloo, 2021-01-01:
> ddcutil did not handle the case where no pci.ids file or usb.ids file was found. That should be fixed now in branch 1.0.0-dev. Let me know if the changes work for you.
Just tested `1.0.0-dev` and it's fixed indeed

rockowitz, 2021-01-01:
FYI in considering a patch, the lack of pci.ids and usb.ids has minimal impact on ddcutil.  The environment command just does not report the names associated with the vendor/product/subsystem ids. 

Some time ago I received a change request from Pino Toscano, packager of pci.ids etc., which you may find of interest.  Send an email to rockowitz@minsoft.com and I'll forward it to you.

closed 2021-01-02 by rockowitz