rockowitz/ddcutil #167: asserts in v1.0.0

pjaitken, 2021-01-01:
Asserts seen in v1.0.0 :

$ sudo src/ddcutil getvcp -b 1 0x0b
VCP code 0x0b (Color temperature increment   ): 100 degree(s) Kelvin

(ddcutil:21673): GLib-CRITICAL **: 17:33:18.626: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed

$ sudo src/ddcutil getvcp SCAN > scan.1

(ddcutil:21160): GLib-CRITICAL **: 17:29:16.921: g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed

rockowitz, 2021-01-02:
(gist of reply that went only to @pjaitken) 

I cannot reproduce these failures locally.  Please run using valgrind or gdb to show backtraces for the failures.

Also, ddcutil's "--trace option" should indicate the execution paths that let to the assert() failures.

$ src/ddcutil --trace all 2>&1

Thank you.

pjaitken, 2021-01-02:
The problem is that both main and _ddca_terminate call release_base_services, so destroy per_thead_data_hash is invoked twice with the same hash pointer:

```
$ src/ddcutil --foobar
Option parsing failed: Unknown option --foobar
No command specified

(main                          ) PJ: main -> release_base_services
(release_base_services         ) PJ: release_base_services
(release_thread_data_module    ) destroy per_thead_data_hash = 0x5564bb8552a0

(_ddca_terminate               ) PJ: _ddca_terminate -> release_base_services
(release_base_services         ) PJ: release_base_services
(release_thread_data_module    ) destroy per_thead_data_hash = 0x5564bb8552a0

free(): double free detected in tcache 2
Aborted
```

rockowitz, 2021-01-03:
It looks like **ddcutil** was built with **configure** option -***-enable-use-api***.  While the name is tempting, it is a vestige of earlier development work and was not meant for public use. It caused what are now library-only files to be linked into the standalone **ddcutil** executable. The option has been deleted.

**configure** options not meant for public use are now (mostly) marked as "Developer-only" in the help mesage.  (Some options, particularly those related to Doxygen, are generated by included macros and not easily controlled.)

The following **configure** options are publicly supported: 
***--enable-lib***
***--enable-envcmds***
***--enable-usb***
***--enable-udev***
***--enable-drm***
***--enable-x11***
***--enable-asan***  (as of release 1.0.0) 

With the exception of ***--enable-asan***, the options allow for building a leaner executable.

pjaitken, 2021-01-03:
Thanks - reconfiguring without ``--enable-use-api`` removed the issue.

closed 2021-01-04 by rockowitz