rockowitz/ddcutil #168: DDC communication failed with ASUSVG248QE and DVI-D Cable

mslcarlson, 2021-01-06:
I have a 1080p 144hz ASUS VG248QE monitor connected to my desktop's graphics card (AMD RX 580). I was able to get ddcutil working with a DisplayPort cable but not DVI-D (using dual link to take advantage of 144hz).

Things I've verified:
i2c_dev module loaded (OS Artix Linux)
Running ddcutil as root
DDC enabled in monitor's OSD

Here is the output of `ddcutil interrogate 2>&1`:
[output.txt](https://github.com/rockowitz/ddcutil/files/5777467/output.txt)

rockowitz, 2021-01-10:
Not sure what to suggest. If only one type of connection fails, it's typically DisplayPort.  The response on the I2C bus to a DDC request is just a string of repeated x6E (which is normally the first byte of a response).  You should be able to see this using **ddcutil** option **--ddc**, which reports DDC data errors. You might try **--sleep-multiplier 2.0** to increase the the time that **ddcutil** waits between writing a request to the monitor and reading the response.

main--, 2021-01-22:
I have the same setup and the same issue. I can confirm that the response is mostly x6E, but with some other data mixed in:

```
$ ddcutil --ddc --sleep-multiplier 1 -b 6 capabilities
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 88 02 02 01 00 01 ff ff ff ff 00 00 00 b4 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 88 02 01 00 01 01 ff ff ff ff 00 00 b4 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC communication failed for monitor on I2C bus /dev/i2c-6
```

To me this looks almost like the monitor is constantly spamming x6E and any actual response is buried somewhere in between that. Playing with `--sleep-multiplier` changes nothing, the output is always a soup of x6E with some randomly distributed `88 02 01 01 00 01 ff ff ff ff 00 00 b4` (or similar) in there.

closed 2025-02-11 by rockowitz