rockowitz/ddcutil #169: "Display not found" every second time

suuuehgi, 2021-01-08:
Hi,

I have a peculiar behavior: I get a "Display not found" error every second time.

```fish
> sudo ddcutil -v setvcp 10 15
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Display not found
> sudo ddcutil -v setvcp 10 15
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Verifying that value of feature 0x10 successfully set...
Verification succeeded
```

Attached are `ddcutil interrogate --verbose` and `ddcutil environment --verbose`.

Current Arch repo-version ddcutil 0.9.9.
[ddcutil.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5788661/ddcutil.log.tar.gz)

rockowitz, 2021-01-10:
@suuuehgi @matmo  I can see the i2c communication failure at the lowest level in the environment command output.   What happens if you repeatedly issue the command **i2cdetect -y N** (where i2c bus number).  What about **get-edid -b N | parse-edid**?  Is the monitor connected using the dock?  If not, what connector on the laptop is being used. (@matmo - what is your hardware configuration?) To each of you, if you can build ddcutil from the 1.0-dev branch and run **ddcutil environment --verbose** that would be helpful.  The latest **environment** command contains extended probing of /sys.

suuuehgi, 2021-01-10:
Thank you very much for the fast and productive answer!

```fish
> ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-4
   EDID synopsis:
      Mfg id:           AUO
      Model:            
      Serial number:    
      Manufacture year: 2017
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           AOC
      Model:            U3277WB
      Serial number:    
      Manufacture year: 2019
      EDID version:     1.4
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-7
   EDID synopsis:
      Mfg id:           AOC
      Model:            U3277WB
      Serial number:    
      Manufacture year: 2019
      EDID version:     1.4
   VCP version:         2.2
```

So it's bus number 7.

```fish
> i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77
```

This is consistent over 5 iterations.
[ddcutil_environment_verbose.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5792484/ddcutil_environment_verbose.log.tar.gz)

```fish
> get-edid -b 7 | parse-edid
7
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
Only trying 7 as per your request.
256-byte EDID successfully retrieved from i2c bus 7
Looks like i2c was successful. Have a good day.
Checksum Correct

Section "Monitor"
	Identifier "U3277WB"
	ModelName "U3277WB"
	VendorName "AOC"
	# Monitor Manufactured week 33 of 2019
	# EDID version 1.4
	# Digital Display
	DisplaySize 700 390
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-160
	VertRefresh 23-80
	# Maximum pixel clock is 600MHz
	#Not giving standard mode: 1920x1080, 60Hz
	#Not giving standard mode: 1280x1024, 60Hz
	#Not giving standard mode: 1280x960, 60Hz
	#Not giving standard mode: 1440x900, 75Hz
	#Not giving standard mode: 1440x900, 60Hz
	#Not giving standard mode: 1680x1050, 60Hz
	#Not giving standard mode: 1280x720, 60Hz

	#Extension block found. Parsing...
	Modeline 	"Mode 13" -hsync -vsync 
	Modeline 	"Mode 0" +hsync -vsync 
	Modeline 	"Mode 1" +hsync -vsync 
	Modeline 	"Mode 2" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 3" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 4" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 5" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 6" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 7" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 8" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 9" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 10" 25.200 640 656 752 800 480 490 492 525 -hsync -vsync
	Modeline 	"Mode 11" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 12" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 14" +hsync +vsync 
	Modeline 	"Mode 15" +hsync +vsync 
	Modeline 	"Mode 16" +hsync -vsync 
	Option "PreferredMode" "Mode 13"
EndSection
```

No change on repetition as well.

The monitor is connected using the dock: Thunderbolt --> Dock --> DP

The output of `ddcutil environment --verbose` from the 1.0.0-dev branch is attached below.

**P.s.** I just noticed that it works every time when I also connect the screen via HDMI to the dock. Whatever that means.

rockowitz, 2021-01-11:
To confirm my understanding: 
- Thunderbolt->Dock->DPI: ddcutil fails every other time
- Thunderbolt->Dock->HDMI: ddcutil fails every other time 
- with both Thunderbolt->Dock->HDMI and Thunderbolt->Dock->DP: no failures

What happens without the docking station?
- laptop->USB-C
- laptop->HDMI

Does the display appear on 2 /dev/i2c buses in all configurations or just some?  Which?

That the EDIDs are different on different connectors is expected, if for no other reason than there's a bit indicating DisplayPort.  As different EDIDs have different versions is surprising.  It could reflect what's in the secondary blocks. (ddcutil uses only the base block (i.e. the first 128 bytes) so the EDID version does not matter to it. 

Finally, the tar file I received has 0 length.  Please resend it.  I need it to reflect a docking station configuration where the monitor appears on 2 /dev/i2c buses.  

Thanks.

matmo, 2021-01-12:
My hardware is Dell XPS15 with a Dell WD19TB (thunderbolt) dock, here are different configurations I have tried:

* Laptop --- (USB-C) ---> Display: **works everytime**
* Laptiop --- (Thunderbolt) ---> Dock --- (USB-C) ---> Display: **fails every 2nd time**
* Laptiop --- (Thunderbolt) ---> Dock --- (DP) ---> Display: **fails every 2nd time**

So this seems to be an issue with the dock, and the display also appears on two busids (5 and 8 in my case, where 8 is ddc capable, see [interrogate](https://github.com/rockowitz/ddcutil/files/5801401/ddcutil_interrogate.txt)).

---

1. `i2cdetect -y `: Works consistently as above.

2. `get-edid -b 8 | parse-edid` (works repeatedly):

```
8
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
Only trying 8 as per your request.
256-byte EDID successfully retrieved from i2c bus 8
Looks like i2c was successful. Have a good day.
Checksum Correct

Section "Monitor"
	Identifier "DELL U2720Q"
	ModelName "DELL U2720Q"
	VendorName "DEL"
	# Monitor Manufactured week 1 of 2020
	# EDID version 1.4
	# Digital Display
	DisplaySize 600 340
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-140
	VertRefresh 24-75
	# Maximum pixel clock is 540MHz
	#Not giving standard mode: 2048x1152, 60Hz
	#Not giving standard mode: 1920x1200, 60Hz
	#Not giving standard mode: 1920x1080, 60Hz
	#Not giving standard mode: 1680x1050, 60Hz
	#Not giving standard mode: 1600x1200, 60Hz
	#Not giving standard mode: 1280x1024, 60Hz
	#Not giving standard mode: 1280x800, 60Hz
	#Not giving standard mode: 1152x864, 75Hz

	#Extension block found. Parsing...
	Modeline 	"Mode 13" +hsync -vsync 
	Modeline 	"Mode 0" +hsync -vsync 
	Modeline 	"Mode 1" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 2" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 3" 74.250 1920 2558 2602 2750 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 4" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 5" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 6" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 7" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 8" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 9" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 10" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 11" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 12" 25.200 640 656 752 800 480 490 492 525 -hsync -vsync
	Modeline 	"Mode 14" +hsync -vsync 
	Modeline 	"Mode 15" +hsync -vsync 
	Option "PreferredMode" "Mode 13"
EndSection
```

3. `ddcutil environment --verbose` with the dev-1.0.0 branch: [ddcutil_environment_broken](https://github.com/rockowitz/ddcutil/files/5801630/ddcutil_environment_dev1_0_0_2.txt), [ddcutil_environment_detected](https://github.com/rockowitz/ddcutil/files/5801631/ddcutil_environment_dev1_0_0.txt)

suuuehgi, 2021-01-13:
>     * Thunderbolt->Dock->DPI: ddcutil fails every other time 
>     * Thunderbolt->Dock->HDMI: ddcutil fails every other time

Both fail every _second_ time.

>     * with both Thunderbolt->Dock->HDMI and Thunderbolt->Dock->DP: no failures

Correct 
 
> Does the display appear on 2 /dev/i2c buses in all configurations or just some? Which?

`ddcutil -v detect` alternated between bus 5 and 7 ..

```fish
> ddcutil -v detect

Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  true
   Dynamic sleep adjustment enabled:            false

(i2c_set_addr) Error in ioctl(I2C_SLAVE), errno=EBUSY(16): Device or resource busy
(i2c_set_addr                  ) addr = 0x50. Returning EBUSY(-16): Device or resource busy
Invalid display
   I2C bus:  /dev/i2c-4
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      true 
      /sys/bus/i2c/devices/i2c-4/name:   AUX A/port A
   EDID synopsis:
      Mfg id:               AUO
      Model:                
      Product code:         9014
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2017
      EDID version:         1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:  /dev/i2c-5
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  false
      /sys/bus/i2c/devices/i2c-5/name:   AUX B/port B
   EDID synopsis:
      Mfg id:               AOC
      Model:                U3277WB
      Product code:         12919
      Serial number:        
      Binary serial number: 349 (0x0000015d)
      Manufacture year:     2019
      EDID version:         1.4
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
   ```
   ```fish
   > ddcutil -v detect
   
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
[... 89 lines cut ...]
70: -- -- -- -- -- -- -- --
```

```fish
> i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77     
```

Now, `ddcutil` fails every time. :confused: 

Pulling and re-plugging the monitor resets the situation.

> Finally, the tar file I received has 0 length. Please resend it. I need it to reflect a docking station configuration where the monitor appears on 2 /dev/i2c buses.

Oh my, what ever happened there ...

In the attached [ddcutil.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5810716/ddcutil.log.tar.gz), you find two files, _normal_ and _fail_. They contain the output in the "normal" alternating situation and the situation described above, failing every time.

suuuehgi, 2021-01-13:
> What happens without the docking station?
> 
>     * laptop->USB-C

Oh my, for the sake of this nice program (the ports are inaccessible and the screen fixed on the wall).

Yes, works every time on USB-C -> DP.

```fish
> i2cdetect -y 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77
```
>     * laptop->HDMI

Works every time as well.

```fish
> i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- 54 -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

In case you need them, you find the corresponding logs in the attached 
[ddcutil2.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5810802/ddcutil2.log.tar.gz)
 as _USBC_ and _HDMI_.

rockowitz, 2021-01-16:
@suuuehgi  & @matmo: 

Unless I've misunderstood something, the "display not found every other time" problem only occurs when the docking station is being used, not when a monitor is connected directly to the host. So this appears to be a problem in the new Multi-Stream Transport driver code.  
What I observed in the logs is that sometimes the second block of the EDID (bytes 128-255) is read instead of the first block, i.e. 
the read position does get reset to byte 0.  I've made a few changes that (hunch, hunch) may work around this problem. First, even though we only need the first 128 bytes of the edid, the code now tries to read 256 bytes. Second if an invalid EDID block is read, it is reread several times in hopes that something will sync up.  The changes have been pushed to branch 1.0.0-dev

The --trcfunc option is probably the best way keep the voluminous tracing manageable.  Let me know if these changes indeed solve the problem

$ ddcutil detect --verbose --trcfunc i2c_get_parsed_edid_by_fd --trcfunc i2c_get_raw_edid_by_fd --trcfunc i2c_get_edid_bytes_directly

If these changes indeed fix the "display not found every other time" problem, hopefully we can distil a sufficiently small test case for a driver bug report.

And @suuuehgi, thank you for being nice to this nice program.

suuuehgi, 2021-01-16:
> the "display not found every other time" problem only occurs when the docking station is being used, not when a monitor is connected directly to the host.

Yes, correct.

> The changes have been pushed to branch 1.0.0-dev
> [...]
> Let me know if these changes indeed solve the problem

Unfortunately, now it is consistent in not detecting the monitor (`/dev/i2c-7`). :/

I attached the log: 
[ddcutil_detect_--verbose_--trcfunc_i2c_get_parsed_edid_by_fd_--trcfunc_i2c_get_raw_edid_by_fd_--trcfunc_i2c_get_edid_bytes_directly.log](https://github.com/rockowitz/ddcutil/files/5825070/ddcutil_detect_--verbose_--trcfunc_i2c_get_parsed_edid_by_fd_--trcfunc_i2c_get_raw_edid_by_fd_--trcfunc_i2c_get_edid_bytes_directly.log)
There are `Input/output error`s for `/dev/i2c-7`.

> And @suuuehgi, thank you for being nice to this nice program.

I'll do my very best. :} Thank you!

rockowitz, 2021-01-18:
The latest 1.0.0-dev branch contains extensive changes which further allow for variations in display detection.  They should revert ddcutil to at least it's prior state where failure occurs every other time when the display is connected to the dock.

I dusted off an old test framework for display detection to vary the new parameterization and better report results. To access it, 
use command **ddcutil detect --f4**.  All the variants work for my 3 monitor setup, 

I've also written a  [simple program](https://github.com/rockowitz/ddcutil/files/5832672/buscheck.c.gz)  that reads the EDID and reads a single VCP feature value.  If this program shows the same failure on every other call with the docking station.  If it fails it can be used for a bug report to the i915 driver folks.  If not, well, hopefully the **--f4** option will reveal what **ddcutil** is doing wrong. **gcc buscheck.c -o buscheck** compiles it.

suuuehgi, 2021-01-19:
Still, ddcutil from the latest 1.0.0-dev branch completely fails to detect the screen.

Attached is the output of `ddcutil detect --f4`: 
[ddcutil_detect_--f4.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5835701/ddcutil_detect_--f4.log.tar.gz)

`buscheck` fails every time:

```fish
> ./buscheck 7
Checking /dev/i2c-7...
Reading EDID...
read EDID: Input/output error
```

rockowitz, 2021-01-19:
Well, this is bloody befuddling.  On the --f4 tests, the EDID on /dev/i2c-7 is read successfully IFF 128 bytes are read, not 256.  That's the opposite of what I see elsewhere.  If we only read 128 bytes and the EDID is 256 bytes, the first time we read the first 128 bytes, the second time the second 128 bytes, which is irrelevant to ddcuti.   On the other hand, reading 256 bytes from a 128 byte EDID just reads 2 copies of the EDID.  So 256 bytes seemed to be the safe choice.  I need to think about this some more.

rockowitz, 2021-01-22:
The --f4 option in  current 1.0.0-dev branch tests an additional EDID size called "dynamic", which twice tries to read a 128 byte EDID, and twice tries to read a 256 byte EDID.  

I've also tweaked [buscheck](https://github.com/rockowitz/ddcutil/files/5852842/buscheck.c.gz) to perform the DDC test even if the EDID test fails.   You can change the EDIDSIZE define from 256 to 128 to see if that makes a difference.

Let me know how it goes, and send output as appropriate.  Thanks.

suuuehgi, 2021-01-22:
## --f4

Wonderful! Attached you find the next log iteration. :slightly_smiling_face: 
[ddcutil_detect_--f4.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5854888/ddcutil_detect_--f4.log.tar.gz)

## buscheck

### EDIDSIZE 128

It seems* to alternate between the following two cases:

```log
Checking /dev/i2c-7...
Reading EDID...
EDID:
        +0          +4          +8          +c            0   4   8   c   
+0000   02 03 26 f1 4b 10 1f 04 13 03 12 02 11 01 05 14   ..&.K...........
+0010   23 09 07 07 83 01 00 00 6d 03 0c 00 10 00 18 78   #.......m......x
+0020   20 00 60 01 02 03 8c 0a d0 8a 20 e0 2d 10 10 3e    .`....... .-..>
+0030   96 00 ba 89 21 00 00 18 01 1d 00 72 51 d0 1e 20   ....!......rQ.. 
+0040   6e 28 55 00 ba 89 21 00 00 1e 02 3a 80 d0 72 38   n(U...!....:..r8
+0050   2d 40 10 2c 45 80 ba 89 21 00 00 1e 4d 6c 80 a0   -@.,E...!...Ml..
+0060   70 70 3e 80 30 20 3a 00 ba 89 21 00 00 1a 00 00   pp>.0 :...!.....
+0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 23   ...............#

EDID is valid: false

Trying simple VCP read of feature 0x10...
Writing request packet...
Sleeping 50 ms..
Reading response...
Response bytes:
        +0          +4          +8          +c            0   4   8   c   
+0000   6e 88 02 00 10 00 00 64 00 2a ea                  n......d.*.     
Valid response: mh=0x00, ml=0x64, sh=0x00, sl=0x2a
DDC communication succeeded
```
```log
Checking /dev/i2c-7...
Reading EDID...
EDID:
        +0          +4          +8          +c            0   4   8   c   
+0000   00 ff ff ff ff ff ff 00 05 e3 77 32 5d 01 00 00   ..........w2]...
+0010   21 1d 01 04 b5 46 27 78 3a 59 05 af 4f 42 af 27   !....F'x:Y..OB.'
+0020   0e 50 54 bd 4b 00 d1 c0 81 80 81 40 95 0f 95 00   .PT.K......@....
+0030   b3 00 81 c0 01 01 22 cc 00 50 f0 70 3e 80 18 10   ......"..P.p>...
+0040   35 00 ba 89 21 00 00 1a a3 66 00 a0 f0 70 1f 80   5...!....f...p..
+0050   30 20 35 00 ba 89 21 00 00 1a 00 00 00 fc 00 55   0 5...!........U
+0060   33 32 37 37 57 42 0a 20 20 20 20 20 00 00 00 fd   3277WB.     ....
+0070   00 17 50 1e a0 3c 01 0a 20 20 20 20 20 20 01 dd   ..P..<..      ..

EDID is valid: true

Trying simple VCP read of feature 0x10...
Writing request packet...
Sleeping 50 ms..
Reading response...
Response bytes:
        +0          +4          +8          +c            0   4   8   c   
+0000   6e 88 02 00 10 00 00 64 00 2a ea                  n......d.*.     
Valid response: mh=0x00, ml=0x64, sh=0x00, sl=0x2a
DDC communication succeeded
```

### EDIDSIZE 256

The output is _mostly_** consistent:

```log
Checking /dev/i2c-7...
[... 179 lines cut ...]
> Response bytes:
>         +0          +4          +8          +c            0   4   8   c

> cat bus-2.log 
Checking /dev/i2c-7...
Reading EDID...

Trying simple VCP read of feature 0x10...
Writing request packet...
Sleeping 50 ms..
Reading response...
Response bytes:
        +0          +4          +8          +c            0   4   8   c   
+0000   ea ea ea ea ea ea ea ea ea ea ea                  ...........     
Invalid address byte in response, expected 0x6e, actual 0xea
Attempt 1 to read feature 0x10 failed.
  Retrying...Writing request packet...
Sleeping 50 ms..
Reading response...
Response bytes:
        +0          +4          +8          +c            0   4   8   c   
+0000   6e 88 02 00 10 00 00 64 00 2a ea                  n......d.*.     
Valid response: mh=0x00, ml=0x64, sh=0x00, sl=0x2a
DDC communication succeeded
```

rockowitz, 2021-01-23:
Please run **ddcutil detect --f4** with options **--trcfunc i2c_get_edid_bytes_directly --trcfunc i2c_get_edid_bytes_using_i2c_layer --stats errors**.  I had loosened the retry criteria.  I'd like to see if any of the usually fatal status codes result in successful retry in the tests.  If not, they can again become fatal.

Re the DDC test in **buscheck**, the occasional errors are normally caught by retry, so all is well.  Though it is interesting that we see both I2C bus data errors and the well formed DDC Null Message, which is monitor indicating it cannot process the request.

Thanks again.

suuuehgi, 2021-01-24:
> ddcutil detect --f4 --trcfunc i2c_get_edid_bytes_directly --trcfunc i2c_get_edid_bytes_using_i2c_layer --stats errors

There you go: 
[ddcutil_detect_--f4_--trcfunc_i2c_get_edid_bytes_directly_--trcfunc_i2c_get_edid_bytes_using_i2c_layer_--stats_errors.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5862096/ddcutil_detect_--f4_--trcfunc_i2c_get_edid_bytes_directly_--trcfunc_i2c_get_edid_bytes_using_i2c_layer_--stats_errors.log.tar.gz)

rockowitz, 2021-01-29:
I've tightened the checks for fatal, as opposed to retryable, errors in EDID detection to improve performance.   Please run **ddcutil detect --f4 --stats errors** to confirm that 128 byte and dynamic cases still read the EDID from /dev/i2c-7.

As a workaround for your situation, I've added yet another command line option: **--edid-read-size N**, where N is 128 or 256.  The default is 256, you will want to use **--edid-read-size 128**.  

This really is just masking the problem in the i915 driver.  Significantly though DDC communication on /dev/i2c-5 always fails, EDID reading always succeeds, no matter what the block size used.  

Relatedly, I've poured over the /sys probing on the **environment** command, and I can't find a way to associate /dev/i2c-5 with /dev/i2c-7.  (At first glance, the EDID would seem to be a way to make the association, but it is possible for 2 monitors to have the same EDID.  There exist monitor models that lack both an ASCII and a binary serial number.)

Lastly, the binary serial number on your monitor is extremely low (349), suggesting it's a very new model.  Perhaps there's something in the EDID implementation that's contributing to the situation.

suuuehgi, 2021-01-30:
> Please run ddcutil detect --f4 --stats errors [...]

There you go: 
[ddcutil_detect_--f4_--stats_errors.log.tar.gz](https://github.com/rockowitz/ddcutil/files/5898741/ddcutil_detect_--f4_--stats_errors.log.tar.gz)

> As a workaround for your situation, I've added yet another command line option: --edid-read-size N, where N is 128 or 256. The default is 256, you will want to use --edid-read-size 128.

That works nicely! :slightly_smiling_face: 

> Lastly, the binary serial number on your monitor is extremely low (349), suggesting it's a very new model.

Hm ... this very monitor is indeed quite new (the metadata says `Monitor Manufactured week 33 of 2019`) but it's a replacement monitor. The model was released 4-5 years ago and at least the UI is as shitty as the old one.

closed 2021-02-13 by rockowitz