rockowitz/ddcutil #170: Dell U2720Q only detected every second run of ddcutil

matmo, 2021-01-09:
When running ddcutil my monitor is only correctly recognized on every second run of ddcutil ([four_runs_of_ddcutil.txt](https://github.com/rockowitz/ddcutil/files/5790858/four_runs_of_ddcutil.txt)).

I suspect this is related to #58, here two some snippets using `--trace i2c` for the relevant busid:

First run:

```
(i2c_check_bus                 ) Starting. busno=8, buf_info=0x55cc28c4d320
(i2c_open_bus                  ) Starting. busno=8, callopts=0x80
(i2c_open_bus                  ) Done.     busno=8, Returning file descriptor: 3
(i2c_get_parsed_edid_by_fd     ) Starting. fd=3
(i2c_get_raw_edid_by_fd        ) Getting EDID. File descriptor = 3, filename=/dev/i2c-8
(i2c_set_addr                  ) fd=3, addr=0x50, i2c_force_slave_addr_flag=false, callopts=CALLOPT_ERR_MSG
(i2c_set_addr                  ) addr = 0x50. Returning OK(0): success
(i2c_get_edid_bytes_directly   ) Getting EDID. File descriptor = 3, filename=/dev/i2c-8, read_bytewise=false
(i2c_get_edid_bytes_directly   ) write() succeeded
(i2c_get_edid_bytes_directly   ) Returning: OK(0): success
(i2c_get_raw_edid_by_fd        ) get bytes returned:
   Buffer at 0x55cc28c4d890,  bytes addr=0x55cc28c4dbd0, len=128, max_size=128
           +0          +4          +8          +c            0   4   8   c   
   +0000   00 ff ff ff ff ff ff 00 10 ac b7 41 4c 36 4e 43   ...........AL6NC
   +0010   01 1e 01 04 b5 3c 22 78 3e ee 95 a3 54 4c 99 26   .....<"x>...TL.&
   +0020   0f 50 54 a5 4b 00 e1 c0 d1 00 d1 c0 b3 00 a9 40   .PT.K..........@
   +0030   81 80 81 00 71 4f 22 cc 00 50 f0 70 3e 80 18 10   ....qO"..P.p>...
   +0040   35 00 55 50 21 00 00 1a 00 00 00 ff 00 48 43 50   5.UP!........HCP
   +0050   58 54 53 32 0a 20 20 20 20 20 00 00 00 fd 00 18   XTS2.     ......
   +0060   4b 1e 8c 36 01 0a 20 20 20 20 20 20 00 00 00 fc   K..6..      ....
   +0070   00 44 45 4c 4c 20 55 32 37 32 30 51 0a 20 01 17   .DELL U2720Q. ..
(i2c_get_raw_edid_by_fd        ) edid checksum = 0
(i2c_get_raw_edid_by_fd        ) Returning OK(0): success
(i2c_get_parsed_edid_by_fd     ) Returning 0x55cc28c4ddc0
(is_edp_device                 ) busno=8, returning: false
(i2c_set_addr                  ) fd=3, addr=0x37, i2c_force_slave_addr_flag=false, callopts=CALLOPT_ERR_MSG
(i2c_set_addr                  ) addr = 0x37. Returning OK(0): success
(i2c_detect_x37                ) write() for slave address x37 returned 1
(i2c_detect_x37                ) read() for slave address x37 returned 4
(i2c_close_bus                 ) Starting. fd=3 - /dev/i2c-8, callopts=CALLOPT_ERR_MSG
(i2c_close_bus                 ) Done.     fd=3, filename=(null), Returning: OK(0): success
(i2c_check_bus                 ) Done. flags=0x0cf1, bus info:
      Bus /dev/i2c-8 found:   true
      Bus /dev/i2c-8 probed:  true
      Bus accessible:          true
      Bus is eDP:              false
      Valid bus name checked:  true
      I2C bus has valid name:  true
      Address 0x37 present:    true
      Address 0x50 present:    true
      EDID synopsis:
         Mfg id:               DEL
         Model:                DELL U2720Q
         Product code:         16823
         Serial number:        HCPXTS2
         Binary serial number: 1129199180 (0x434e364c)
         Manufacture year:     2020
         EDID version:         1.4
         Extra descriptor:        
         Video input definition:    0xb5 - Digital Input (DisplayPort)
         Supported features:
            DPMS active-off
            Digital display type: RGB 4:4:4 + YCrCb 4:2:2
            Standard sRGB color space: True
         White x,y:        0.313, 0.329
         Red   x,y:        0.640, 0.330
         Green x,y:        0.300, 0.600
         Blue  x,y:        0.150, 0.060
         Extension blocks: 1
      EDID source: 
      EDID hex dump:
                 +0          +4          +8          +c            0   4   8   c   
         +0000   00 ff ff ff ff ff ff 00 10 ac b7 41 4c 36 4e 43   ...........AL6NC
         +0010   01 1e 01 04 b5 3c 22 78 3e ee 95 a3 54 4c 99 26   .....<"x>...TL.&
         +0020   0f 50 54 a5 4b 00 e1 c0 d1 00 d1 c0 b3 00 a9 40   .PT.K..........@
         +0030   81 80 81 00 71 4f 22 cc 00 50 f0 70 3e 80 18 10   ....qO"..P.p>...
         +0040   35 00 55 50 21 00 00 1a 00 00 00 ff 00 48 43 50   5.UP!........HCP
         +0050   58 54 53 32 0a 20 20 20 20 20 00 00 00 fd 00 18   XTS2.     ......
         +0060   4b 1e 8c 36 01 0a 20 20 20 20 20 20 00 00 00 fc   K..6..      ....
         +0070   00 44 45 4c 4c 20 55 32 37 32 30 51 0a 20 01 17   .DELL U2720Q. ..
```

Second run:

```
(i2c_check_bus                 ) Starting. busno=8, buf_info=0x559c5bbc1320
(i2c_open_bus                  ) Starting. busno=8, callopts=0x80
(i2c_open_bus                  ) Done.     busno=8, Returning file descriptor: 3
(i2c_get_parsed_edid_by_fd     ) Starting. fd=3
(i2c_get_raw_edid_by_fd        ) Getting EDID. File descriptor = 3, filename=/dev/i2c-8
(i2c_set_addr                  ) fd=3, addr=0x50, i2c_force_slave_addr_flag=false, callopts=CALLOPT_ERR_MSG
(i2c_set_addr                  ) addr = 0x50. Returning OK(0): success
(i2c_get_edid_bytes_directly   ) Getting EDID. File descriptor = 3, filename=/dev/i2c-8, read_bytewise=false
(i2c_get_edid_bytes_directly   ) write() succeeded
(i2c_get_edid_bytes_directly   ) Returning: OK(0): success
(i2c_get_raw_edid_by_fd        ) get bytes returned:
   Buffer at 0x559c5bbc1890,  bytes addr=0x559c5bbc1bd0, len=128, max_size=128
           +0          +4          +8          +c            0   4   8   c   
   +0000   02 03 24 f1 4c 10 1f 20 05 14 04 13 12 11 03 02   ..$.L.. ........
   +0010   01 23 09 7f 07 83 01 00 00 e3 05 ff 01 e6 06 07   .#..............
   +0020   01 60 56 28 a3 66 00 a0 f0 70 3e 80 30 20 35 00   .`V(.f...p>.0 5.
   +0030   55 50 21 00 00 1a 56 5e 00 a0 a0 a0 29 50 30 20   UP!...V^....)P0 
   +0040   35 00 55 50 21 00 00 1a 11 44 00 a0 80 00 25 50   5.UP!....D....%P
   +0050   30 20 36 00 55 50 21 00 00 1a 00 00 00 00 00 00   0 6.UP!.........
   +0060   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
   +0070   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14   ................
(i2c_get_raw_edid_by_fd        ) edid checksum = 0
(i2c_get_raw_edid_by_fd        ) Returning OK(0): success
(i2c_get_parsed_edid_by_fd     ) Returning (nil)
(i2c_close_bus                 ) Starting. fd=3 - /dev/i2c-8, callopts=CALLOPT_ERR_MSG
(i2c_close_bus                 ) Done.     fd=3, filename=(null), Returning: OK(0): success
(i2c_check_bus                 ) Done. flags=0x0cc1, bus info:
      Bus /dev/i2c-8 found:   true
      Bus /dev/i2c-8 probed:  true
      Bus accessible:          true
      Bus is eDP:              false
      Valid bus name checked:  true
      I2C bus has valid name:  true
      Address 0x37 present:    false
      Address 0x50 present:    false
```

matmo, 2021-01-09:
Sorry, duplicate of #169 posted a few hours earlier. I'm also on Arch fwiw

rockowitz, 2021-01-11:
@matmo, @suuuehgi I'm reopening this issue as it makes it easier for me to bounce between the details of it and #169.

rockowitz, 2021-01-11:
@matmo In your file of 4 **ddcutil detect** runs, there are EBUSY errors from i2c_set_addr().  The output with **--trace i2c** does not show these errors.  Were these commands run differently (e.g. from scripts)?  Can you produce a traced output with the EBUSY error? Does  option **--force-slave-address** avoid the error?

matmo, 2021-01-12:
Sorry, the errors are there as well, here are the full outputs: [ddcutil detect --trace i2c (broken)](https://github.com/rockowitz/ddcutil/files/5801405/detect_broken.txt), [ddcutil detect --trace i2c (detected)](https://github.com/rockowitz/ddcutil/files/5801406/detect_correctly.txt)

It seems this error is not related to the display (busno=8) but to the SMBus device (busno=9), **--force-slave-address** does not change anything. Here is my [ddcutil_interrogate](https://github.com/rockowitz/ddcutil/files/5801401/ddcutil_interrogate.txt).

closed 2021-02-13 by rockowitz