rockowitz/ddcutil #171: DDC communication failed with 2 monitors out of 3

Azenet, 2021-01-16:
Hello,

I have recently changed CPU and GPU and have only 1 monitor out of 3 work in ddcutil. All monitors are detected properly but only the monitor on bus 4 (VW225) responds properly to DDC commands. All monitors worked properly on my previous setup. Is this a driver issue?

Attached are outputs from [`ddcutil interrogate`](https://github.com/rockowitz/ddcutil/files/5823604/ddc-interrogate.txt) and [`ddcutil --bus=5 setvcp 10 0 --verbose --trace i2c --ddc`](https://github.com/rockowitz/ddcutil/files/5823605/ddc-setvcp.txt).

rockowitz, 2021-01-16:
What is your physical configuration?  It's not apparent how you have multiple monitors connected.   Whether this is a driver or hardware configuration issue I can't say.

i2cdetect shows that slave address x37 is not responsive on i2c-7.  Attempting to write to it results in errno EREMOTEIO. 

Slave address x37 is present on i2c-5, communication there fails with Linux errno EIO. 

If you can build from branch 1.0.0-dev, please run **ddcutil environment --verbose**.  The latest version contains much more extensive probing of /sys.

Azenet, 2021-01-16:
Thanks for the reply. My configuration is monitors on bus 4 and 5 through the same brand of DP to DVI-D cable, and the monitor on bus 7 is through an active dual DP to DVI-D adapter.

[Here](https://github.com/rockowitz/ddcutil/files/5824513/ddc-1.0.0-dev-environment.txt) is the output of ddcutil environment --verbose from branch 1.0.0-dev (639aa04).

rockowitz, 2021-01-16:
What is the video card?

>
> Thanks for the reply. My configuration is monitors on bus 4 and 5 
> through the same brand of DP to DVI-D cable,
i.e. a passive adapter.   Does it adjust voltage levels?

> and the monitor on bus 7 is through an active dual DP to DVI-D adapter.
I think you meant to say active DP to dual link DVI-D, e.g. 
https://www.amazon.com/StarTech-com-DisplayPort-DVI-Adapter-Dual-Link/dp/B00A493CNY

i2cdetect shows that slave address x37 is not present on /dev/i2c-7.  
Consistent with this, ddcutil DDC communication fails with Linux error 
ERMOTEIO.  Given that the monitor is known to support DDC, it appears 
that the adapter is not converting the I2C signals to/from the 
DisplayPort AUX channel.  If the adapter is known to work in other 
contexts perhaps it's an incompatibility between the adapter and your 
video card.

Azenet, 2021-01-16:
Video card is an ASUS RX 5700XT Strix-O8G-GAMING. I do not know whether voltages are adjusted in the passive adapter, only that they were working in my previous setup. And yes, I am using a [similar adapter](https://www.amazon.com/gp/product/B07TNJFKSR/) for bus 7. I will attempt to confirm in a known-good setup whether it all works and report back.

closed 2025-02-09 by rockowitz