rockowitz/ddcutil #173: DDC communication failed & Display not found for Dell Optiplex 7780

Txordi00, 2021-01-19:
Hi. I have a Dell Optiplex 7780 (an AIO computer). So far, I am being unable to change the brightness of its display. I think that my best chance is with this nice tool, but I don't know how to for the moment.

I have seen a similar issue [here](https://github.com/rockowitz/ddcutil/issues/14) but, so far, I don't know how to enable DDC/CI communication in the integrated monitor, which apparently was the solution for that issue.

I tried to follow [https://wiki.archlinux.org/index.php/backlight#External_monitors](https://wiki.archlinux.org/index.php/backlight#External_monitors). That is, I first load the kernel module `i2c_dev` by
`sudo modprobe i2c-dev`
Then the command `sudo ddcutil detect` gives
```
Invalid display
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           DEL
      Model:            OptiPlex 7780
      Serial number:    
      Manufacture year: 2017
      EDID version:     1.3
   DDC communication failed
```
And finally, when I try `sudo ddcutil capabilities` or `sudo ddcutil getvcp 10`, I simply get `Display not found`.

I attach here the output of the command `sudo ddcutil interrogate 2>&1`:
[ddcutil_interrogate.txt](https://github.com/rockowitz/ddcutil/files/5837051/ddcutil_interrogate.txt)

Thanks a lot for the help. Appreciated,
Jordi.

Txordi00, 2021-01-19:
I think that maybe the problem is that I can't find the way to access the OSD controls from Linux (Manjaro). I tried to run the Dell OSD app with Wine, but it doesn't work. Maybe I should install Windows...

rockowitz, 2021-01-20:
Slave address x37, used for DDC/CI  communication, on bus /dev/i2c-1 is not responsive.  The simplest way to confirm this is to install **i2cdetect**.  Command **i2cdetect 1** will probe /dev/i2c-7 and report the responsive slave addresses.

Presumably the Dell documentation should document a way to enable the physical OSD, as opposed to a Windows app that communicates with the monitor.

Txordi00, 2021-01-20:
Hi @rockowitz. Thanks a lot for the quick answer :)

I have run the command `i2cdetect 1` which probes /dev/i2c-1, but not /dev/i2c-7, right? Here is the output:
```
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1.
I will probe address range 0x08-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
I tried hard to find how to enable physical OSD or get an app for Linux, but I did found nothing so far. In the BIOS I can only enable or disable the OSD buttons, but not turn them to physical, neither control any display setting from there.

rockowitz, 2021-01-22:
My apologies for the confusion.  I was also thinking of another user when typed i2c-7.  My only suggestion, which I'm sure you've tried, is to look for a enable/disable DDC/CI setting in the BIOS. Perhaps what you describe as the "enable/disable the OSD buttons" setting in the BIOS is another way describing this setting.

Txordi00, 2021-01-22:
> My apologies for the confusion. I was also thinking of another user when typed i2c-7. My only suggestion, which I'm sure you've tried, is to look for a enable/disable DDC/CI setting in the BIOS. [...]

No problem. Thanks for answering again. Yeah, I checked that and it changes nothing... Perhaps this weekend I will try to install windows and the OSD app. Let's see if I can change it from there. I will let you know.

Txordi00, 2021-01-23:
Hi again. I installed Windows and the "OSD app". But the app only allows Windows to change the brightness of the monitor, and gives no control whatsoever (it does not even have any GUI). In Windows, I also used an utility called "Dell Comand | Configure" meant to modify system/BIOS settings of Dell computers and, even if there were tons of parameters, there was none related to DDC/CI. I finally tried another app called "Dell Display Manager", which requires a DDC/CI compliant monitor to work and, surprise, it didn't work. So I am assuming that this monitor cannot be used through DDC/CI :(

On another hand, and this is weirder, I used a Pop!_OS 20.10 LiveCD to modify partitions and, for my surprise, I could modify the screen brightness with it (with the GNOME brightness slider)! I installed it and when I first boot, the display brightness slider did nothing anymore XD. I applied the system updates and, bingo, it worked again. But still I am not able to use `ddcutil capabilities` and the `i2cdetect` command gives me the same output. It looks very weird to me...

Pop!_OS uses Kernel 5.8. I tried before with this kernel in Manjaro but it refused to work. And both use GNOME 3.38 and I tested both with Wayland and X11 (and so far, the change Wayland <-> X11 had no effect concerning this issue).

I left Manjaro installed just in case you want to try something more or do some experiment with it. 

Thanks for the help. Regards,
Jordi.

rockowitz, 2021-01-23:
From what you describe I suspect the display on your AIO is connected 
as if it is a laptop display. Laptop displays do not implement DDC/CI.  
Laptops use ACPI to control the screen brightness.  If so there should 
be entries in directory /sys/class/brightness.  (The exact entries 
vary.) Also look at directory /sys/class/drm.  An entry with "eDP" in 
the name indicates an "embedded Display Port" connection, which is the 
type of display interface used in current laptops.

The xrandr "--brightness" option adjusts the brightness level of the 
video output to the monitor, as opposed to the brightness of the 
monitor's backlight.  The xbacklight utility provides a simple way to 
adjust this option.

Txordi00, 2021-01-23:
Understood. Massive thanks for the help @rockowitz. I will try it on Monday, since it is my job's computer. I will let you know.

Txordi00, 2021-01-25:
I think that the issue is that Pop!_OS installs itself with the `dell_uart_backlight` Kernel module built and loaded while Manjaro does not even have it built. When I get time I will try to recompile the Kernel in Manjaro together with this specific module, load it at boot, and see if that resolves the issue. For now, I will keep using Pop.

BTW, I could change the brightness in Manjaro using X11 and `xrandr` at HDMI2. BUT the solution doesn't work with Wayland and still it looks worse than in Pop (less maximum brightness and a lot of color distorsion when I try to increase the brightness parameter above 1).

Thanks for your help,
Jordi.

closed 2021-02-22 by rockowitz