rockowitz/ddcutil #174: Can't controll second monitor when connected to docking station

boscard, 2021-01-27:
Hi,
First of all - great piece of software! You are doing awesome work! :)
I have setup with 2 computers and 2 DELL P2417H monitors. One computer is connected using HDMI ports and it works perfectly fine. I am able to control both monitors when input is set to HDMI.
Second computer is connected to docking station using USB-C and monitor are connected using DisplayPort. In this setup I'm able to control only one monitor but only when both are connected.

Logs in attached files:
* [ddcutil_detect.txt](https://github.com/rockowitz/ddcutil/files/5879337/ddcutil_detect.txt)
* [ddcutil_environment.txt](https://github.com/rockowitz/ddcutil/files/5879338/ddcutil_environment.txt)

rockowitz, 2021-01-27:
This is a long standing problem in the i915 driver.  See the [ddcutil FAQ](http://www.ddcutil.com/faq#docking), and issues #11 and #146.  It appears to be fixed in the 5.10 kernel.

The freedesktop.org bug report is [here](https://gitlab.freedesktop.org/drm/intel/-/issues/37)

However, some users have reported that the fixed driver fails to read the EDID properly every other time.  This appears to be a problem in the updated driver.  A workaround is in development.  See issue #xx

rockowitz, 2021-02-13:
Please build from branch 1.0.2-dev, execute the following commands, and submit the output: 

`ddcutil detect --verbose`
`ddcutil environment --very-verbose`

Thank you.

boscard, 2021-02-20:
Wow - it works for me like a charm!

```
$ sudo ./ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-6
   EDID synopsis:
      Mfg id:               AUO
      Model:                
      Product code:         8685
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2016
      EDID version:         1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:  /dev/i2c-8
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL P2417H
      Product code:         41179
      Serial number:        CW6Y774D4GJB
      Binary serial number: 877087298 (0x34474a42)
      Manufacture year:     2017
      EDID version:         1.4
   DDC communication failed

Display 1
   I2C bus:  /dev/i2c-9
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL P2417H
      Product code:         41179
      Serial number:        CW6Y774D4GJB
      Binary serial number: 877087298 (0x34474a42)
      Manufacture year:     2017
      EDID version:         1.4
   VCP version:         2.1

Display 2
   I2C bus:  /dev/i2c-10
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL P2417H
      Product code:         41179
      Serial number:        CW6Y785I0MLB
      Binary serial number: 810372162 (0x304d4c42)
      Manufacture year:     2018
      EDID version:         1.4
   VCP version:         2.1
```
[ddcutil_detect.txt](https://github.com/rockowitz/ddcutil/files/6013769/ddcutil_detect.txt)
[ddcutil_environment.txt](https://github.com/rockowitz/ddcutil/files/6013770/ddcutil_environment.txt)

boscard, 2021-02-20:
Hmm, seems that it works for some time and then it fails to detect displays on i2c-9 and i2c-10. It start to work again after powerinf off and on displays again.

rockowitz, 2021-02-20:
I've just posted some additional changes on branch 1.0.2-dev to distinguish the "phantom" /dev/i2c device from the ones representing the monitors connected through the dock.   See may latest message on #181 

When ddcutil detection fails intermittently for /dev/i2c-9 and /dev/i2c-10 (i.e. the dock connected displays), I assume the monitors still display video.  Is that the case? Could it be that you're seeing another instance of the "display detected every other time" problem?

Please send the usual output from **ddcutil detect --very-verbose** and **ddcutil environment --very-verbose**, from the latest 1.0.2-dev branch, when the detection fails.

boscard, 2021-03-04:
Everything seems to work fine to moment I execute **ddcutil environment --very-verbose**. After that detect is able to find only one incompatible display. Logs from detect and environment in attachments.
[ddcutil.detect.txt](https://github.com/rockowitz/ddcutil/files/6082329/ddcutil.detect.txt)
[ddcutil.environment.txt](https://github.com/rockowitz/ddcutil/files/6082330/ddcutil.environment.txt)

rockowitz, 2021-03-05:
@boscard  & @laur89  I suspect that there's some EDID buffering in DRM that is getting confused. What I see in the **environment** output is that the EDID requests fail, but the DDC requests succeed for the monitors in question.  I've also occasionally seen cases where the second block of a 256 byte EDID is returned as the first 128 bytes. 

The latest 1.0.2-dev branch contains several changes.  It treats reads that return errno=EIO as retryable instead of fatal, since that errno value is associated with the reads that fail.  I've also added trace tests that check for example, if the EDID block returned looks like the second 128 bytes of a 256 byte EDID. 

There are several things to try when the monitors become undetected.  I'm afraid I don't have a linear plan of attack.   I'll leave it to you if you find anything interesting.  

- Run **detect** with option **--f4--, which runs detect with numerous combinations of settings.  Are there variations for which detection works?
- Using **--trcfunc get_raw_edid_by_fd --trcfunc i2c_get_edid_bytes_directly** turns on tracing in the most relevant functions.  It reports retries, status codes from EDID reads, and whether a 128 byte EDID is wrong in a recognizable way.
- See if forcing the EDID read size by using **--edid-read-size 256** or **--edid-read-size 128** has an effect.
- Note that option **--very-verbose** with the **detect** command now reports the raw EDID, which may be helpful. 
-  
Thanks, 
Sanford

laur89, 2021-03-05:
If it's something as obscure I'd say maybe document the quirk, but leave it be.

`environment` is not a command one would be using daily anyway.

rockowitz, 2021-03-05:
@laur89, @boscard You may be right. It's time to get a release out the 
door.  It's not so much the **environment** command per se that concerns 
me, but that I'm somehow putting the driver into a failure state.  If 
nothing else, it would help make a bug report to the driver folks less 
vague.

Regarding output, I'm interested in reports that show something of 
interest after running the **environment** command, which I leave to 
your judgement.   Examples:
- Do the **--f4** tests reveal a mode that successfully reads the EDID?
- When tracing with **--trcfunc get_raw_edid_by_fd  --trcfunc 
i2c_get_edid_bytes_directly**:
    - Does reading now work? (which would indicate that retrying after 
EIO errors solves the problem)
    - Does using the **--edid-read-size** option  produce success?
    - Are there any trace messages regarding invalid EDID blocks? e.g. 
the block read looks like the second EDID block

closed 2025-02-09 by rockowitz