rockowitz/ddcutil #175: change video input from a samsung tv

albfan, 2021-02-03:
I'm trying to change input from a samsung tv but I get all the DDC communication failed, Display not found errors.

What I'm looking for is for what a chromecast does (when I cast something to it, it changes the source on tv) Not really sure if that is DDC command o different signal.

I just tried `sudo ddcutil --bus 0 setvcp 60 0x0f`

but there's no change. Any tip about what makes a display with several hdmi ports to change selected input source would be great.

```
sudo ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Invalid display
   I2C bus:             /dev/i2c-0
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  false
      /sys/bus/i2c/devices/i2c-0/name:   i915 gmbus dpc
   EDID synopsis:
      Mfg id:           SAM
      Model:            SAMSUNG
      Serial number:    
      Manufacture year: 2008
      EDID version:     1.3
      Product code:     1289
      Binary sn:        1 (0x00000001)
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 4c 2d 09 05 01 00 00 00   ........L-......
      +0010   30 12 01 03 80 10 09 78 0a ee 91 a3 54 4c 99 26   0......x....TL.&
      +0020   0f 50 54 bd ef 80 71 4f 81 00 81 40 81 80 95 00   .PT...qO...@....
      +0030   95 0f b3 00 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 a0 5a 00 00 00 1e 66 21 50 b0 51 00 1b 30   E..Z....f!P.Q..0
      +0050   40 70 36 00 a0 5a 00 00 00 1e 00 00 00 fd 00 18   @p6..Z..........
      +0060   4b 1a 51 17 00 0a 20 20 20 20 20 20 00 00 00 fc   K.Q...      ....
      +0070   00 53 41 4d 53 55 4e 47 0a 20 20 20 20 20 01 6f   .SAMSUNG.     .o
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

Invalid display
   I2C bus:             /dev/i2c-3
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      true 
      /sys/bus/i2c/devices/i2c-3/name:   AUX A/port A
   EDID synopsis:
      Mfg id:           BOE
      Model:            
      Serial number:    
      Manufacture year: 2016
      EDID version:     1.4
      Product code:     1746
      Binary sn:        0 (0x00000000)
      Extra descriptor: NV156FHM-N47
      Video input definition: 0x95 - Digital Input (DisplayPort)
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.328
      Red   x,y:        0.585, 0.363
      Green x,y:        0.350, 0.579
      Blue  x,y:        0.162, 0.143
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 09 e5 d2 06 00 00 00 00   ................
      +0010   01 1a 01 04 95 22 13 78 02 c9 a0 95 5d 59 94 29   .....".x....]Y.)
      +0020   24 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   $PT.............
      +0030   01 01 01 01 01 01 a8 39 80 36 71 38 17 40 30 20   .......9.6q8.@0 
      +0040   36 00 58 c2 10 00 00 1a fd 2d 80 0e 71 38 28 40   6.X......-..q8(@
      +0050   30 20 36 00 58 c2 10 00 00 1a 00 00 00 fe 00 42   0 6.X..........B
      +0060   4f 45 20 43 51 0a 20 20 20 20 20 20 00 00 00 fe   OE CQ.      ....
      +0070   00 4e 56 31 35 36 46 48 4d 2d 4e 34 37 0a 00 44   .NV156FHM-N47..D
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.
```

rockowitz, 2021-02-03:
Televisions use a protocol known as [Consumer Electronics Control](https://en.wikipedia.org/wiki/Consumer_Electronics_Control) (CEC) to switch inputs.  This is different from DDC/CI, and is carried over different pins on the HDMI connector.  Command line program **cec-ctl** in package v4l-utils can communicate with CEC devices.   Here's an example of a [USB-HDMI CEC adapter](https://www.amazon.com/Pulse-Eight-na-USB-HDMI-CEC-Adapter/dp/B005JU6LWM).

albfan, 2021-02-03:
So there's no way to contribute on that here? As I see model: SAMSUNG I suppose there was something to do here.

Yes it says HDMI-CEC (Anynet+)

I have a PC display which has an option to enable DDC, now I  got it.

Sorry for the noise.

closed 2021-02-03 by albfan