rockowitz/ddcutil #176: Release 1.0.0: Error opening /sys/bus/i2c/devices/i2c-0/device/device/device/class

OlivierBP, 2021-02-05:
Hi,

First of all, thank you for this amazing tool!

I have some issues with the new release 1.0.0.

My setup is:
- Monitor Dell P2415Q
- Docking Station Dell WD19TB

Everything worked with the docking station since the kernel 5.10.x as specified here https://gitlab.freedesktop.org/drm/intel/-/issues/37 . However, with the new release of ddcutil, I can't get or set the brightness and luminosity of my external display. I tried downgrading my kernel and ddcutil separately and it looks like the last release of ddcutil broke something for me.

## My env:

Version broken: 1.0.0
Version working: 0.9.9
Kernel: 5.10.13-arch1-1
OS: Arch

## Logs

Output of `/usr/bin/ddcutil -v detect` with version `1.0.0`:
```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  true
   Dynamic sleep adjustment enabled:            false

Error opening /sys/bus/i2c/devices/i2c-0/device/device/device/class: No such file or directory
Error opening /sys/bus/i2c/devices/i2c-1/device/device/device/class: No such file or directory
Invalid display
   I2C bus:  /dev/i2c-7
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      true
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-7/name:   AUX A/port A
   EDID synopsis:
      Mfg id:               SHP
      Model:
      Product code:         5293
      Serial number:
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2018
      EDID version:         1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:  /dev/i2c-9
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-9/name:   AUX C/port C
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL P2415Q
      Product code:         41150
      Serial number:        D8VXF9B710FB
      Binary serial number: 825247298 (0x31304642)
      Manufacture year:     2019
      EDID version:         1.4
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

Output of `/usr/bin/ddcutil -v detect` with version `0.9.9`:
```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Invalid display
   I2C bus:             /dev/i2c-7
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      true
      /sys/bus/i2c/devices/i2c-7/name:   AUX A/port A
   EDID synopsis:
      Mfg id:           SHP
      Model:
      Serial number:
      Manufacture year: 2018
      EDID version:     1.4
      Product code:     5293
      Binary sn:        0 (0x00000000)
      Extra descriptor: 0R99KLQ133D1
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.328
      Red   x,y:        0.640, 0.329
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.149, 0.060
      Extension blocks: 0
   EDID source:
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 4d 10 ad 14 00 00 00 00   ........M.......
      +0010   2a 1c 01 04 a5 1d 11 78 0e de 50 a3 54 4c 99 26   *......x..P.TL.&
      +0020   0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0
      +0040   35 00 26 a5 10 00 00 18 a4 a6 00 a0 f0 70 3e 80   5.&..........p>.
      +0050   30 20 35 00 26 a5 10 00 00 18 00 00 00 fe 00 30   0 5.&..........0
      +0060   52 39 39 4b 80 4c 51 31 33 33 44 31 00 00 00 00   R99K.LQ133D1....
      +0070   00 02 41 03 28 01 12 00 00 0b 01 0a 20 20 00 41   ..A.(.......  .A
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-9
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  false
      /sys/bus/i2c/devices/i2c-9/name:   AUX C/port C
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2415Q
      Serial number:    D8VXF9B710FB
      Manufacture year: 2019
      EDID version:     1.4
      Product code:     41150
      Binary sn:        825247298 (0x31304642)
      Extra descriptor:
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.659, 0.334
      Green x,y:        0.301, 0.639
      Blue  x,y:        0.149, 0.043
      Extension blocks: 1
   EDID source:
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 10 ac be a0 42 46 30 31   ............BF01
      +0010   2d 1d 01 04 a5 35 1e 78 3a e2 45 a8 55 4d a3 26   -....5.x:.E.UM.&
      +0020   0b 50 54 a5 4b 00 71 4f 81 80 a9 c0 a9 40 d1 c0   .PT.K.qO.....@..
      +0030   e1 00 d1 00 01 01 22 cc 00 50 f0 70 3e 80 18 10   ......"..P.p>...
      +0040   35 00 0f 28 21 00 00 1a 00 00 00 ff 00 44 38 56   5..(!........D8V
      +0050   58 46 39 42 37 31 30 46 42 0a 00 00 00 fc 00 44   XF9B710FB......D
      +0060   45 4c 4c 20 50 32 34 31 35 51 0a 20 00 00 00 fd   ELL P2415Q. ....
      +0070   00 1d 4c 1e 8c 36 00 0a 20 20 20 20 20 20 01 b5   ..L..6..      ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

Display 1
   I2C bus:             /dev/i2c-10
      I2C address 0x50 (EDID) responsive: true
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true
      /sys/bus/i2c/devices/i2c-10/name:   DPMST
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2415Q
      Serial number:    D8VXF9B710FB
      Manufacture year: 2019
      EDID version:     1.4
      Product code:     41150
      Binary sn:        825247298 (0x31304642)
      Extra descriptor:
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.659, 0.334
      Green x,y:        0.301, 0.639
      Blue  x,y:        0.149, 0.043
      Extension blocks: 1
   EDID source:
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c
      +0000   00 ff ff ff ff ff ff 00 10 ac be a0 42 46 30 31   ............BF01
      +0010   2d 1d 01 04 a5 35 1e 78 3a e2 45 a8 55 4d a3 26   -....5.x:.E.UM.&
      +0020   0b 50 54 a5 4b 00 71 4f 81 80 a9 c0 a9 40 d1 c0   .PT.K.qO.....@..
      +0030   e1 00 d1 00 01 01 22 cc 00 50 f0 70 3e 80 18 10   ......"..P.p>...
      +0040   35 00 0f 28 21 00 00 1a 00 00 00 ff 00 44 38 56   5..(!........D8V
      +0050   58 46 39 42 37 31 30 46 42 0a 00 00 00 fc 00 44   XF9B710FB......D
      +0060   45 4c 4c 20 50 32 34 31 35 51 0a 20 00 00 00 fd   ELL P2415Q. ....
      +0070   00 1d 4c 1e 8c 36 00 0a 20 20 20 20 20 20 01 b5   ..L..6..      ..
   VCP version:         2.1
   Controller mfg:      Mstar
   Firmware version:    1.4
   Monitor returns DDC Null Response for unsupported features: false
```
Here are the two `ddcutil interrogate`:
[working.txt](https://github.com/rockowitz/ddcutil/files/5933184/working.txt)
[broken.txt](https://github.com/rockowitz/ddcutil/files/5933183/broken.txt)

Let me know if I can test anything else to help

rockowitz, 2021-02-05:
First, thanks for thorough post. 

I think the problem results from changes to EDID detection that were made attempting to work around a problem in the driver's implementation of I2C for docking stations. Note that for 0.9.9, the  monitor appears at both /dev/i2c-9 and /dev/i2c-10.  DDC is responsive on the later. Under release 1.0.0, the monitor is only at /dev/i2c-9.  However, your "broken" interrogate output does show that while the  EDID can't be read at /dev/i2c-10, DDC works.
 
As reported in issues #169 and #170 by @suuuehgi and @matmo, ddcutil only detected the "working"  /dev/i2c device every other time.  Do you observe this on release 0.9.9 as well?

On release 1.0.0, try using option --edid-read-size=0.  If that solves the problem, try using --edid-read-size=128.

OlivierBP, 2021-02-05:
Thanks for the fast reply.

On release 0.9.9:
Indeed, I get ddcutil working every 2 times when using the dock. Works fine with a direct cable to the monitor. Seems to be the same as #169 and #170 .

On release 1.0.0:
- When using `--edid-read-size=0`, I still get the `Error opening /sys/bus/i2c/devices/i2c-0/device/device/device/class: No such file or directory` error, but it detects my external monitor twice (as in 0.9.9), and the brightness is changed. 
- Works also well with `--edid-read-size=128` (exact same behavior)
- In both cases, I can confirm that I don't need to run the command twice anymore

You solved my problem (thanks a lot), so I think it can be closed, unless you want me to provide more debug info.

rockowitz, 2021-02-05:
I'm going to keep the problem open until a fix is posted.  --edid-read-size-128 is intended as a workaround until the driver is fixed.  No normal user should require it.  --edid-read-size=0 tries both 128 and 256 byte reads.  That may have to become the default behaviour.

The "Error opening..." message is benign and just needs to be suppressed. /dev/i2c-0 and /dev/i2c-1 in your case are SMBus devices.

rockowitz, 2021-02-06:
@OlivierBP , @suuuehgi, @malmo

New branch 1.0.1-dev changes EDID reading to attempt 256 byte reads and then, if that fails 128 byte reads.  Micro-marginally slower, but it makes option ***--edid-read-size*** unnecessary.  Please test ddcutil with this branch.  

Also the diagnostic message "Error opening /sys/bus/i2c/devices/i2c-N/device/device/device/class"  is suppressed. It is benign.

suuuehgi, 2021-02-06:
> Please test ddcutil with this branch.

Works nicely! :relaxed:

OlivierBP, 2021-02-06:
Works nicely with branch `1.0.1-dev` for me too

closed 2021-02-13 by rockowitz