rockowitz/ddcutil #177: Second monitor connected over MST is not detected

siwonpawel, 2021-02-07:
Hi,

My graphic card is GeForce GTX 1050Ti and configuration is 2x Dell 2415U connected with DisplayPort using MST. Second monitor is not detected in ddcutil but shown and working in system.

[ddcutil_detect.txt](https://github.com/rockowitz/ddcutil/files/5939980/ddcutil_detect.txt)
[ddcutil_environment.txt](https://github.com/rockowitz/ddcutil/files/5939987/ddcutil_environment.txt)

rockowitz, 2021-02-13:
Please build from branch 1.0.2-dev, execute the following commands, and submit the output: 

`ddcutil detect --verbose`
`ddcutil environment --very-verbose`

Also, please clarify how MST is being used.  The 2415U does not support DisplayPort pass through, and there doesn't appear to be a docking station. Are you using DisplayPort MST hub such as [this one](https://www.amazon.com/displayport-mst-hub/s?k=displayport+mst+hub) from Monoprice?

siwonpawel, 2021-02-13:
I using these as shown on image
[image]
Here are logs:
[detect.txt](https://github.com/rockowitz/ddcutil/files/5976497/detect.txt)
[environment.txt](https://github.com/rockowitz/ddcutil/files/5976498/environment.txt)

rockowitz, 2021-02-13:
I must have looked in the wrong Dell manual before.  Now I see that the monitor supports DisplayPort pass through.

The problem, I'm afraid, is in the Nvidia driver.  There are 6 /dev/i2c devices. 0,1,2 are SMBus devices, while 3,4,5 are Nvidia devices.  Only /dev/i2c-5 supports either EDID (on slave address x50) or DDC (on slave address x37), so ddcutil sees only 1 monitor.  This is confirmed by the output of command **i2cdetect**, which the **environment** command executes through a command shell.

The **environment** output  ends at the point it is trying to examine a huge /var/log/kern.log.  Apparently it crashed.  Were there any error messages that went to the terminal?

siwonpawel, 2021-02-14:
No errors occurred, it just executed for long time so I cancelled it. Here are logs with no cancellation from my side
[detect.txt](https://github.com/rockowitz/ddcutil/files/5977588/detect.txt)
[environment.txt](https://github.com/rockowitz/ddcutil/files/5977589/environment.txt)

mdallaire, 2023-08-16:
Sorry to bump this old issue but I am in a very similar situation.
I have 3 Dell U2415 plugged via MST to a NVIDIA GeForce RTX 2060 and I can only adjust brightness on the first screen in the chain. I am able to adjust the brightness on the three of them when on Windows with the Dell Display Manager tools which makes me think it is a software issue.

[detect.txt](https://github.com/rockowitz/ddcutil/files/12362882/detect.txt)
[environment.txt](https://github.com/rockowitz/ddcutil/files/12362883/environment.txt)

rockowitz, 2023-08-23:
I'm afraid the answer is still the same: the Nvidia driver does not support DDC on MST devices.

closed 2025-02-09 by rockowitz