rockowitz/ddcutil #178: Error detecting i2c-dev module

devonallie, 2021-02-08:
When running 
`ddcutil detect`
The output is 
`Error opening file /lib/modules/5.10.0-rc6-00215-g7059c2c00a21-dirty/modules.builtin: No such file or directory
Error reading file /lib/modules/5.10.0-rc6-00215-g7059c2c00a21-dirty/modules.builtin: No such file or directory
Assuming module i2c-dev is not built in to kernel
ddcutil requires module i2c_dev.`

I compiled my own kernel and enabled i2c character devices however, ddcutil seemingly requires the i2c_dev module anyways. Is there another driver I need to compile into the kernel? Perhaps a firmware driver I am not aware of?

Any help in this regard would be much appreciated.

rockowitz, 2021-02-09:
The error message you're seeing reflects the fact that modules.builtin file listing modules compiled in the kernel was not found at the expected location.  That i2c-dev was not found in the kernel (because we couldn't look), and not in a module was incorrectly interpreted as i2c-dev not existing. 

This check iwas recently added to ddcutil initialization in order to quickly inform users of the absence of i2c-dev, instead of failing later more cryptically.  If the check can't determine if i2c-dev is installed it should not terminate ddcutil execution.  This bug will be fixed in release 1.0.2. 

In the meantime, you can copy the modules.builtin file created when compiling the kernel to the location specified in the error message.  Alternatively, if it is more convenient and you're building ddcutil from source, you can comment out the call to function validate_environment() by function master_initializer(), at approximately line 330 of file src/app_ddcutil/main.c.

rockowitz, 2021-02-13:
The initial checks for driver i2c-dev have been revised.  If module i2c-dev is not loaded and we can't tell if it is built into the kernel, a warning is issued but execution proceeds.  The checks in the **environment** command regarding i2c-dev have also been extended.  

Please build from branch 1.0.2-dev, execute the following commands, and submit the output: 

`ddcutil detect --verbose`
`ddcutil environment --very-verbose`

If ddcutil complains about not being able to find either the compiler configuration file or modules.builtin, where are they found on your system?

devonallie, 2021-02-14:
I have a forlder in my root directory with the kernel build files and managed to copy the modules.builtin file to 
`/lib/modules/5.10.0-rc6-00215-g7059c2c00a21-dirty/modules.builtin`. This eliminated the error I presented initially. 

I use the ddcutil-git package from the Arch Linux core repositories so I do not yet have access to the latest build. I would build from source but I am not sure how to and as such, cannot complete your desired testing. 

Another issue I am running into that is not related but perhaps you could assist is that as a normal user I do not have access to char devices. `ddcutil environment` reccomends I add a rule through udev but I do not know how to create a proper udev rule.

devonallie, 2021-02-14:
As an update to my third and unrelated point, I managed to resolve the issue by changing the permissions of the i2c character device with `chmod`. No udev modification required. I would however, reccomend providing the rule necessary to establish the i2c device as part of the i2c group. The response from `ddcutil environment` provides the first and third step for attaining ownership of the i2c device but not the second.

bricewge, 2021-06-04:
I have a similar issue here, on Guix:
``` shell
# ddcutil detect
Unable to read modules.builtin
Module i2c-dev is not loaded and ddcutil can't determine if it is built into the kernel
[...]
```
The path to `modules.builtin` is [hardcoded](https://github.com/rockowitz/ddcutil/blob/39ce66b42305b304816d635b42a5b38ce280e2dd/src/util/linux_util.c#L115) some distributions are already [tweaking it
](https://github.com/NixOS/nixpkgs/blob/nixos-21.05/pkgs/tools/misc/ddcutil/nixos-paths.diff).

A solution would be to use kmod, either the modinfo binary:
``` shell
$ for mod in thinkpad_acpi i2c_dev; do modinfo $mod | grep -E '^(name|filename):'; done
filename:       /run/booted-system/kernel/lib/modules/5.12.8/kernel/drivers/platform/x86/thinkpad_acpi.ko.gz
name:           thinkpad_acpi
name:           i2c_dev
filename:       (builtin)
```
Or libkmod's [`kmod_module_get_path`](https://github.com/kmod-project/kmod/blob/b6ecfc916a17eab8f93be5b09f4e4f845aabd3d1/libkmod/libkmod-module.c#L733-L742) directly.

rockowitz, 2021-06-05:
Thank you for the detailed post.  It will take me a bit of time to fully think through the considerations, but I do want to acknowledge your message. 

kmod_module_get_path() depends on struct kmod_ctx, which is returned by kmod_new().  Per the inline doc for kmod_new(), if the dirname argument is null, the library path defaults to /lib/modules/`uname -r`, which is not correct for guix.  Apparently, config option --with-rootdir allows for changing the default. Presumably guix builds libkmod using this option. The -b option to modinfo, etc. allows for specification of the modules directory root.  I've not traced it through, but I assume this ultimately becomes the dirname argument to kmod_new().   The bottom line seems to be that I just have to assume that libkmod on guix is built defaulting to the proper directory path, which is reasonable.  Otherwise using libkmod would require that ddcutil know a priori the correct directory, which would just relocate the problem.

That your modinfo example does not require the -b option suggests the above is correct. Once I'm convinced of it I'll recode the module checks to use libkmod, which has the added benefit of simplifying the.

bricewge, 2021-06-06:
My comment what just a suggestion; I was wondering how this issue could be fixed without adding a hardcoded patch as it's done in NixOS. Stubbing upon libkmod I taught it may interest you that a library implementing a robust way to know the modules path exist. 

> That your modinfo example does not require the -b option suggests the above is correct.

Kind of, we are actually [patching](https://github.com/guix-mirror/guix/blob/f7331f06dbabd855ac536ff0ec367bb5e25ea959/gnu/packages/patches/kmod-module-directory.patch) kmod to use an environment variable

> Once I'm convinced of it I'll recode the module checks to use libkmod, which has the added benefit of simplifying the code.

Fair enough and an additional library is needed.

rockowitz, 2021-06-07:
I've upload changes using libkmod to check if i2c-dev is builtin or is a module file. 

Note that the check for i2c-dev at the start of ddcutil execution was purely advisory as a helpful diagnostic instead of ddcutil failing less clearly later on.

It's not clear to me from your messages whether you're building ddcutil because you're using it, or simply because you're porting packages to NixOS.  If you're up for it, I'd appreciate it if you try to build and use ddcui, the graphical user interface.  (Use branch 0.2.0-dev) It's much less mature than the ddcutil command, so I would not be surprised if there are issues to address.

bricewge, 2021-06-07:
Thank you, that was quick !

I was using `ddcutil` to debug `ddcci-driver-linux`. `ddcutil` is already [packaged](https://git.savannah.gnu.org/cgit/guix.git/tree/gnu/packages/hardware.scm#n61) in Guix. With your last modification and no patch, it is working well:
``` sh
$ /gnu/store/djxl34qkhqw4k9d3i4nb6g278jw6lsqz-ddcutil-1.1.1-dev/bin/ddcutil detect
(module_status_using_libkmod) Starting. module_alias=i2c-dev
(module_status_using_libkmod) Done.     module_alias=i2c-dev, returning 1
Display 1
   I2C bus:  /dev/i2c-10
   EDID synopsis:
      Mfg id:               BNQ
      Model:                BenQ GW2765
      Product code:         30934
      Serial number:        [REDACTED]
      Binary serial number: [REDACTED]
      Manufacture year:     2017,  Week: 18
   VCP version:         2.1

```

As for `ddcui` I've [packaged](https://paste.debian.net/1200348/) it, that was a straight forward process. The `glib` dependency is missing for the documentation, but CMake flag it clearly as missing. So no issues at all! As for the software itself, the Features view is the best part, discovering  and setting value through is faster than using `ddcutil`.

closed 2025-02-09 by rockowitz