rockowitz/ddcutil #180: Regression: 1.0.1 can't detect both of my monitors

carlocastoldi, 2021-02-09:
I updated ddcutil to `1.0.1` on my Manjaro desktop and now ddcutil cannot detect my two monitors.

`uname -a`: `Linux carlo-manjaro 5.10.13-2-MANJARO #1 SMP PREEMPT Sat Feb 6 10:28:21 UTC 2021 x86_64 GNU/Linux` (changing to LTS 5.4.19 didn't work either)

<details align="center">
  <summary><code>ddcutil detect</code> in <code>0.9.9</code></summary>
  <pre><code>
Display 1
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           SAM
      Model:            LC24RG50
      Serial number:    H4ZM801857
      Manufacture year: 2019
      EDID version:     1.3
   VCP version:         2.1
Display 2
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            S24B150
      Serial number:    H4MD301250
      Manufacture year: 2013
      EDID version:     1.3
   VCP version:         2.0
  </code></pre>
</details>

<details align="center">
  <summary><code>ddcutil detect</code> in <code>1.0.1</code></summary>
  <pre><code>
No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.
  </code></pre>
</details>

<details align="center">
  <summary><code>ddcutil environment</code> in <code>0.9.9</code></summary>
  <pre><code>
The following tests probe the runtime environment using multiple overlapping methods.

*** Basic System Information ***

ddcutil version: 0.9.9

/proc/version:
   Linux version 5.10.13-2-MANJARO (builduser@development) (gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.36) #1 SMP PREEMPT Sat Feb 6 10:28:21 UTC 2021

Architecture:     x86_64
Distributor id:   ManjaroLinux
Release:          20.2.1
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/boot/vmlinuz-5.10-x86_64 root=UUID=5af0480f-7819-407a-97ab-a895cdf30141 rw udev.log_priority=3 bootsplash.bootfile=bootsplash-themes/manjaro-man/bootsplash

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:01:00.0 (boot_vga flag is set)
   Device class:        x030000    VGA compatible controller
   Vendor:              x1002      Advanced Micro Devices, Inc. [AMD/ATI]
   Device:              x6939      Tonga PRO [Radeon R9 285/380]
   Subvendor/Subdevice: 1682/9380  XFX Pine Group Inc.
   Driver name:         amdgpu
   Driver version:      Unable to determine
   I2C device:          i2c-3      name: AMDGPU DM i2c hw bus 3
   I2C device:          i2c-1      name: AMDGPU DM i2c hw bus 1
   I2C device:          i2c-2      name: AMDGPU DM i2c hw bus 2
[... 284 lines cut ...]
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3 4 5 6 7 8 9 10
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              false
   dev_i2c_common_group_name:     root
   all_dev_i2c_has_group_i2c:     false
   any_dev_i2c_has_group_i2c:     false
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       false
   cur_uname:                     carlo
   cur_uid:                       1000
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
  </code></pre>
</details>

rockowitz, 2021-02-09:
On 1.0.1, please run the following commands and submit the output: 

$ ddcutil environment --very-verbose
$ ddcutil detect --verbose --trcfunc i2c_get_raw_edid_by_fd --trcfunc i2c_get_edid_bytes_directly
$ ddcutil detect --verbose --f4

(**--f4** is a "utility" option currently used to trigger test code) 

Thank you.

carlocastoldi, 2021-02-09:
> On 1.0.1, please run the following commands and submit the output:
>
> $ ddcutil environment --very-verbose
> $ ddcutil detect --verbose --trcfunc i2c_get_raw_edid_by_fd --trcfunc [...]

[ddcutil_detect_f4.log](https://github.com/rockowitz/ddcutil/files/5953888/ddcutil_detect_f4.log)
[ddcutil_detect.log](https://github.com/rockowitz/ddcutil/files/5953890/ddcutil_detect.log)
[ddcutil_environment.log](https://github.com/rockowitz/ddcutil/files/5953891/ddcutil_environment.log)

Thank you a lot for the quick reply!
And sorry if I kept you waiting :\

rockowitz, 2021-02-10:
I'm surprised by the trace output, though I have hu a hunch.                                                                                   
What happens if you call detect with option ***--edid-read-size 128*** or ***--edid-read-size 256***?

carlocastoldi, 2021-02-10:
>  What happens if you call detect with option _**--edid-read-size 128**_
```
Display 1
   I2C bus:  /dev/i2c-1
   EDID synopsis:
      Mfg id:               SAM
      Model:                LC24RG50
      Product code:         3984
      Serial number:        H4ZM801857
      Binary serial number: 1129788471 (0x43573437)
      Manufacture year:     2019
      EDID version:         1.3
   VCP version:         2.1

Display 2
   I2C bus:  /dev/i2c-3
   EDID synopsis:
      Mfg id:               SAM
      Model:                S24B150
      Product code:         2435
      Serial number:        H4MD301250
      Binary serial number: 808596788 (0x30323534)
      Manufacture year:     2013
      EDID version:         1.3
   VCP version:         2.0
```

> or _**--edid-read-size 256**_?
```
No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.
```

Something positive, it seems?

carlocastoldi, 2021-02-10:
Oh fuck me.
I am afraid I wrongly read the version my distro just published. It's `1.0.0-1` (as you can see from the first `ddcutil environment`)
I am terribly sorry, as I have actually updated to `1.0.1` the problem no longer exists :)

I hope I didn't make you waste too much time

closed 2021-02-11 by rockowitz