rockowitz/ddcutil #184: No displays found. [AOC]

20NE, 2021-02-18:
monitor manufacturer: AOC

$ ddcutil detect
```
No displays found.
```

$ ddcutil environment

```
*** Basic System Information ***

ddcutil version: 1.0.1

/proc/version:
   Linux version 5.4.97-1-lts (linux-lts@archlinux) (gcc version 10.2.0 (GCC)) #1 SMP Wed, 10 Feb 2021 09:32:28 +0000

Architecture:     x86_64
Distributor id:   (null)
Release:          (null)
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/vmlinuz-linux-lts root=UUID=59411b00-910a-42c9-ad92-2b11d9c13c70 rw resume=/dev/sda12 loglevel=3 drm_kms_helper.edid_firmware=edid/edid

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:00:02.0 (boot_vga flag is set)
   Device class:        x030000    VGA compatible controller
   Vendor:              x8086      Intel Corporation
   Device:              x0152      Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller
   Subvendor/Subdevice: 1458/d000  Gigabyte Technology Co., Ltd
   Driver name:         i915
   Driver version:      Unable to determine
   I2C device:          i2c-3      name: i915 gmbus dpc
   I2C device:          i2c-1      name: i915 gmbus vga
   I2C device:          i2c-4      name: i915 gmbus dpb
   I2C device:          i2c-2      name: i915 gmbus panel
   I2C device:          i2c-0      name: i915 gmbus ssc
   I2C device:          i2c-5      name: i915 gmbus dpd

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Current user: xxx (1000)

Checking /dev/i2c-* devices...

Devices /dev/i2c-* must exist and the logged on user must have read/write permission for those devices (or at least those devices associated 
with monitors).

Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c

Alternatively, this can be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89, 0 Feb 15 11:08 /dev/i2c-0
   crw-rw---- 1 root i2c 89, 1 Feb 15 11:08 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 2 Feb 15 11:08 /dev/i2c-2
   crw-rw---- 1 root i2c 89, 3 Feb 15 11:08 /dev/i2c-3
   crw-rw---- 1 root i2c 89, 4 Feb 15 11:08 /dev/i2c-4
   crw-rw---- 1 root i2c 89, 5 Feb 15 11:08 /dev/i2c-5
   crw-rw---- 1 root i2c 89, 6 Feb 15 11:08 /dev/i2c-6

Current user (xxx) has RW access to all /dev/i2c-* devices.

*** Primary Check 3: Check that module i2c_dev is loaded ***

Checking for module i2c_dev...
   Module i2c-dev is NOT built into kernel
   Loadable i2c-dev module found
   Module i2c_dev is loaded

*** Primary Check 4: Driver specific checks ***

Performing driver specific checks...
No driver specific checks apply.

*** Additional probes ***

Scanning /proc/modules for driver environment...
   Found other loaded module:      i2c_i801
   Found other loaded module:      i2c_dev
   Found video driver module:      i915
   Found other loaded module:      i2c_algo_bit
   Found other loaded module:      drm_kms_helper
   Loaded drm module depends on:   i915,drm_kms_helper,

Testing if modules are loaded using /sys...
   Module amdgpu           is NOT loaded
   Module fbdev            is NOT loaded
   Module fglrx            is NOT loaded
   Module fturbo           is NOT loaded
   Module i915             is loaded
   Module mgag200          is NOT loaded
   Module nvidia           is NOT loaded
   Module nouveau          is NOT loaded
   Module radeon           is NOT loaded
   Module vboxvideo        is NOT loaded
   Module vc4              is NOT loaded
   Module drm              is loaded
   Module i2c_algo_bit     is loaded
   Module i2c_dev          is loaded
   Module i2c_piix4        is NOT loaded
   Module ddcci            is NOT loaded

Examining /sys/bus/i2c/devices...
   /sys/bus/i2c/devices/i2c-0/name:   i915 gmbus ssc
   /sys/bus/i2c/devices/i2c-1/name:   i915 gmbus vga
   /sys/bus/i2c/devices/i2c-2/name:   i915 gmbus panel
   /sys/bus/i2c/devices/i2c-3/name:   i915 gmbus dpc
   /sys/bus/i2c/devices/i2c-4/name:   i915 gmbus dpb
   /sys/bus/i2c/devices/i2c-5/name:   i915 gmbus dpd
   /sys/bus/i2c/devices/i2c-6/name:   SMBus I801 adapter at f040

Env_Accumulator:
   architecture:                  x86_64
   distributor_id                 
   Drivers detected:              i915
   /dev/i2c device numbers:       0 1 2 3 4 5 6
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3 4 5 6
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     xxx
   cur_uid:                       1000
   cur_user_in_group_i2c:         true
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
```

rockowitz, 2021-02-18:
Please execute the following commands and submit the output as attachments or gist references. The output is far too voluminous to include in the message proper. 
$ ddcutil detect --verbose
$ ddcutil environment --very-verbose

Also, please describe your configuration.

rockowitz, 2021-02-18:
An additional question: 

Were displays detected with an earlier version?  0.9.9? 1.0.0?

20NE, 2021-02-20:
I tried on version before v1.0.1 with error:
`DDC communication failed`
may because of i915 [bug](https://bugs.freedesktop.org/show_bug.cgi?id=100954)
It said that v1.0.1 addresses this i915  bug.
[ddcutil_detect.txt](https://github.com/rockowitz/ddcutil/files/6014042/ddcutil_detect.txt)
[ddcutil_environment.txt](https://github.com/rockowitz/ddcutil/files/6014043/ddcutil_environment.txt)

rockowitz, 2021-02-20:
The checks of devices /dev/i2c-0 .. /dev/i2c-5 show that none are responsive on slave address x50 (EDID) or x37 (DDC) requests.

Consistently the output of command i2cdetect shows no monitors that are responsive on slave address x50 and x37.  In fact, no I2C slave addresses are responsive on any I2C bus.

Consistently, xrandr reports only a only single VGA connected display.

Consistently, the DRM layer reports only the single VGA connected display.

udev reports that /dev/i2c-1 is associated with the VGA connector, but that only indicates what /dev/i2c device would be used on that connector.

Consistently, /sys finds the EDID for the VGA connected display, but no evidence of an associated /dev/i2c device. 

Bottom line:  There's no I2C communication between the host and the display.  Given that you have a VGA connection to an inexpensive monitor this is not surprising.  This is not a ddcutil problem.

closed 2021-02-21 by 20NE