rockowitz/ddcutil #185: Samsung monitor only responding to DDC on one input?

pfps, 2021-02-23:
I have a Samsung monitor (a U28H750UQNXZA) with three inputs - two HDMI and one DP.  
I've tried detecting the monitor from a desktop using a DP connection and from two laptops using two different USB hubs that have HDMI output.   The only connection that works is one of the HDMI inputs, independent of which laptop or which USB hub is involved.

Is it possible that this monitor only supports DDC/CI on only one of its inputs?  Is there any way to tell for sure?

rockowitz, 2021-02-23:
I myself have a U32H750, which is the same monitor, only bigger.

I'm a little baffled by your description of USB hubs with HDMI output.  
What exactly are they?  What are the outputs from your laptops?

All the monitor inputs do work.  However, getting the monitor to switch 
inputs, at least from the OSD, can be annoying, IIRC. You'll probably 
find it useful to know that the values used by the monitor for feature 
x60, which switches inputs, are not those defined in the MCCS spec.  
I've not been able to reverse engineer whatever DDC/CI command sequences 
are used to manage Picture-by-Picture, which was simply not envisaged 
when the MCCS spec was defined.

You've probably seen my description of the monitor in therogues gallery 
of monitors <https://www.ddcutil.com/monitor_notes/#samsung_u32h750>on 
the ddcutil web site.  As I say there, it's a DDC/UI disaster area.

Sanford

pfps, 2021-02-23:
I have a Lenovo C940 (personal) and Dell XPS 13 (9310, work).   Neither has an HDMI output, but both have USB-C outputs (with Thunderbolt, but that's not relevant here, I think).   I have two USB-C hubs (or docking stations - both terms are in common use for these kinds of devices, I think).  They have a variety of outputs - USB-A, ethernet, HDMI, etc.  So you can use them to connect to a monitor, which I do.

So the monitor connection is laptop via USB-C to hub/dock via HDMI to one the two Samsung monitor's HDMI inputs.  When a laptop (either one) is connected to the HDMI 1 port of the monitor (through either hub/dock) the monitor is detected correctly and I can switch inputs using using ddcutil.  However, when a laptop is connected to the HDMI 2 port the monitor is seen and can be used, but ddcutil (version 0.9.9) detect says "DDC communication failed" after showing correct information about the monitor.

So I guess I'll have to use ssh to the laptop connected to HDMI-1 if I want to switch inputs from the other laptop.  That is unless someone knows how to turn on DDC/CI for the other inputs.  (I've seen descriptions of monitors that only listen for DDC/CI commands on the active input, so I guess this is a bit better.)

And, yes, the descriptions of some of the capabilities are problematic.   ddcutil reports that feature 60 is input source and that 11 is for HDMI-1, 12 for HDMI-2, and 0f for DP, but really 5 is HDMI-1, 6 is HDMI-2, and 15 is DP.   Only the last aligns with what is reported.  And, also yes, the reason to want to use ddcutil for this monitor is the **^$*$#**  joystick control, which requires multiple actions to switch inputs.