rockowitz/ddcutil #195: IIyama ProLite XUB3493WQSU PBP / PIP no options

totalgaara, 2021-03-26:
Hi,

First i want to thank you for that awesome,project, it could be my solution for my virtualisation setup. I'm running Manjaro on host and a VM with Windows 8.1 with gpu passtrough, i've got two gpu : 

Main : R9 290
VM : GTX 745

For now i've been using software like Moonlight and GeForce experience to remote control my VM, but it's really not perfect, sometime it feels "slow". i then tried the PBP mode on my monitor and it's so much better, the only issue : it's absolutely not adaptive, if i shutdown my vm, it doesn't go like "ok there is one display, use full monitor", i'm stuck in dual monitor mode. 

I would like to make a script using dd-utils to enable PBP and disable it when the VM is shutdown but i can't see any PBP options...

Here is some command i wrote for you (PS For now all command are used as root as i'm testing): 

```bash
 sudo ddcutil detect                                                                                                     
Display 1
   I2C bus:  /dev/i2c-4
   EDID synopsis:
      Mfg id:               IVM
      Model:                PL3493WQ
      Product code:         30227
      Serial number:        1171403701438
      Binary serial number: 16843009 (0x01010101)
      Manufacture year:     2020
      EDID version:         1.4
   VCP version:         2.2
```

```bash
sudo ddcutil getvcp all  
VCP code 0x02 (New control value             ): One or more new control values have been saved (0x02)
VCP code 0x0b (Color temperature increment   ): 100 degree(s) Kelvin
VCP code 0x0c (Color temperature request     ): 3000 + 35 * (feature 0B color temp increment) degree(s) Kelvin
VCP code 0x10 (Brightness                    ): current value =   100, max value =   100
VCP code 0x12 (Contrast                      ): current value =    50, max value =   100
VCP code 0x14 (Select color preset           ): User 1 (0x0b), Tolerance: Unspecified (0x00)
VCP code 0x16 (Video gain: Red               ): current value =   100, max value =   100
VCP code 0x18 (Video gain: Green             ): current value =   100, max value =   100
VCP code 0x1a (Video gain: Blue              ): current value =   100, max value =   100
VCP code 0x1e (Auto setup                    ): Auto setup not active (sl=0x00)
VCP code 0x20 (Horizontal Position (Phase)   ): current value =    50, max value =   100
VCP code 0x30 (Vertical Position (Phase)     ): current value =    50, max value =   100
VCP code 0x52 (Active control                ): Value: 0x00
VCP code 0x60 (Input Source                  ): DisplayPort-1 (sl=0x0f)
VCP code 0x62 (Audio speaker volume          ): Volume level: 25 (00x19)
VCP code 0x6c (Video black level: Red        ): current value =    50, max value =   100
VCP code 0x6e (Video black level: Green      ): current value =    50, max value =   100
VCP code 0x70 (Video black level: Blue       ): current value =    50, max value =   100
VCP code 0x86 (Display Scaling               ): Linear expansion (compression) on h and v axes (sl=0x08)
VCP code 0x8d (Audio mute/Screen blank       ): Unmute the audio (sl=0x02), Invalid value (sh=0x00)
VCP code 0xa5 (Change the selected window    ): Full display image area selected except active windows (sl=0x00)
VCP code 0xaa (Screen Orientation            ): 0 degrees (sl=0x01)
VCP code 0xac (Horizontal frequency          ): 23364 hz
VCP code 0xae (Vertical frequency            ): 60.03 hz
VCP code 0xb6 (Display technology type       ): LCD (active matrix) (sl=0x03)
VCP code 0xc0 (Display usage time            ): Usage time (hours) = 1058 (0x000422) mh=0xff, ml=0xff, sh=0x04, sl=0x22
VCP code 0xc6 (Application enable key        ): 0x006f
VCP code 0xc8 (Display controller type       ): Mfg: Mstar (sl=0x05), controller number: mh=0xff, ml=0x16, sh=0x00
VCP code 0xc9 (Display firmware level        ): 0.0
VCP code 0xca (OSD/Button Control            ): OSD enabled, button events enabled (sl=0x02), Host control of power unsupported (sh=0x00)
VCP code 0xcc (OSD Language                  ): English (sl=0x02)
VCP code 0xd6 (Power mode                    ): DPM: On,  DPMS: Off (sl=0x01)
VCP code 0xdc (Display Mode                  ): Standard/Default mode (sl=0x00)
VCP code 0xdf (VCP Version                   ): 2.2
```

```bash
sudo ddcutil environment                                                                                                                                                      ✔ 
The following tests probe the runtime environment using multiple overlapping methods.

[... 130 lines cut ...]
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              false
   dev_i2c_common_group_name:     root
   all_dev_i2c_has_group_i2c:     false
   any_dev_i2c_has_group_i2c:     false
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       false
   cur_uname:                     root
   cur_uid:                       0
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.

```

Do you know what could be the issues ? i tried to figure it out by myself but i'm not really sure what i've to look at :/

Thank you

rockowitz, 2021-03-27:
What I believe you're saying is that you'd like to control PBP using ddcutil.   The question is what are the command sequences?  Most likely it is some combination of manufacturer specific features and non-standard extensions to feature X60 (Input Source). You could try to reverse engineer the sequences by changing settings in the OSD and observing what has changed in ddcutil.  This has not proven to be productive.  See issue #163.  The only real solution, I believe, is to sniff the I2C bus while the Windows application provided by the manufacturer makes changes.  See issue #192 for my description of inexpensive hardware to do this.

totalgaara, 2021-03-27:
Hi, thanks for your reply :)

I did some tests on Windows with SoftCCMS, and here is the results : 
- 2 screens detected since both gpu are connected into the monitor, on linux i can only see one, even if the virtual machine is running

When i try to go to PBP mode here is what happend

> Opcode 0x60=00 200**E2F0F** become Opcode 0x60=00 200**E210F**
and another value seems to be changing but it seems to be constructor related:

> Opcode 0xA5=00 FFFF**0000**  become Opcode 0xA5=00 FFFF**0200**

I tried, leaving the PBP mode, to activate it again using this code but.. nothing happends

I maybe paranoic using this protocol can this could messup my screen? I'm scary at every value i'm trying to change :/

rockowitz, 2021-03-29:
I appreciate that English is not your native language, but I'm having difficulty understanding what you've written.   When you write "i try to go to PBP mode here is what happened", I assume you mean you set PBP mode using the OSD, then observed the values using MCCS.  Where are you getting what you're calling the "opcode" values? The value for feature xA5 (window selection) sort of makes sense, where FFFF is the maximum value, and 0200 is the current value.   However, this would make more sense as FFFF0002, since the byte ordering in DDC communication is MH,ML,SH,SL (max value high order byte, max value low order byte, current value high order byte, current value low order byte). I have no idea how to make sense of the feature x60 values, where you're highlighting 5 hex digits.

How are you trying to set the VCP feature values?  Using MCCS or ddcutil?

I can't guarantee that playing with VCP feature values won't permanently "messup" your screen, But I've never heard of this happening.  At worst, you may have to restore factor settings using ddcutil or the OSD.

totalgaara, 2021-03-29:
Hi, I'm sorry for my English, Normally my writing is not so bad, after reading me again, your're right that it's difficult to understand what i mean.. 

What i did was : 

-> Take a capture of the current state of my screen, with PBP mode disabled first
-> Take another capture after switching to PBP mode using the OSD menu

Then i switrched back to "PBP OFF" mode and try to sent the values of 

200E210F on "0x60" (display input) then change the value of 0xA5 to "FFFF0200" with MCCS but that didn't do anything

rockowitz, 2021-03-30:
How are you sending the feature values to the monitor?  My copy of softMCCS does not have the ability to send specific values to the monitor.  Or are you using **ddcutil setvcp***?  The results you're seeing are the get feature value responses.  After the "opcode", the 00 byte indicates that the feature is settable, as opposed to momentary.  The remaining 4 bytes are known as MH, ML, SH, SL.  usually the high and lower order bytes of the maximum value, followed by the high and low order bytes of the current value.  Some features use the bytes in feature specific ways.  When setting a feature value, you send only the SH and SL bytes for the new value, i.e. 2 bytes.  You can't change the maximum value. If using **ddcutil setvcp**, since you're sending a 2 byte hex value, specify it in hex notation, e.g. **ddcutil setvcp xa5 x0200.

Since you're seeing variation in the MH,ML bytes for feature x60, the monitor is likely using that feature in a special way.

totalgaara, 2021-04-04:
Sorry for the delayed reply, i was very busy. :) There is some slider on softmmcs to adjust and that's change the value, but i don't know if it does anything.

I tried your example command "ddcutil setvcp xa5 x0200" with ddcutil and it kind of work. The monitor was set on PIP mode but actually the screen wasn't split ?. __ I feel like we are getting closer to a result ;)

closed 2025-02-10 by rockowitz