rockowitz/ddcutil #196: How to wake up monitorig?

dev-zero, 2021-03-29:
I have an Iiyama PL2792UH which I use on 2 computers. The monitor enters sleep rather quickly if it doesn't find an active input (input source switching set to manual), at which point I get a "Display not found" from `ddcutil` when trying to switch the input (when using `--model`, resp. a "DDC communication failed" when using `--bus`), despite that `ddcutil detect` still shows the monitor (although with a `DDC communication failed`) and wayland/sway still has it configured and usable.
When I wakeup the screen via one of the button I can set the input with `ddcutil`.
Is there any advice on how to wakeup the screen from the attached but not active computer?

rockowitz, 2021-03-30:
The fact that ddcutil always detects the monitor reflects the fact that's it's always possible to read the EDID.  In fact, it should be possible to read the EDID even with the monitor turned off - that circuit can be powered from the video cable. 

I'm confused as to your configuration.  If the monitor is sleeping, how do you see the output of **ddcutil detect**?   

If I get it right, you have 2 inputs on the monitor.  To make the description less abstract, let's say computer 1 is on the monitor's HDMI connector, and computer 2 is on the monitor's DVI connector. At some point, perhaps you've turned the monitor off and then on again, it detects an input from computer 1 on the HDMI connector, but not computer 2 on the DVI connector.  Yet despite detecting only 1 active input, it does not accept DDC requests on that input.  So you can't use **ddcutil setvcp x60** to switch the active input.  This would be very strange.

dev-zero, 2021-03-30:
It's slightly different: Both computers are on, active input is set to HDMI from Computer 1. I turn off Computer 1. Monitor goes into sleep mode. Computer 2 (connected via DP) still has the monitor configured, can obviously read its EDID, but sending any commands fails.

KeyofBlueS, 2021-03-30:
Just my 2 cents, as a workaround, You could  try to run this script on the notebook:

```
#!/bin/bash

# set the display bus
# example
BUS='1'

while true; do
	
	# if display isn't responding to ddcutil getvcp 10 (brightness),
	# force off and then on dpms, this could wake up the display
	if ! ddcutil --bus="${BUS}" getvcp 10; then
		xset dpms force off
		sleep 2
		xset dpms force on
	fi
	# check every five seconds
	sleep 5
done

exit 0

```

dev-zero, 2021-03-30:
@KeyofBlueS thanks a lot for the hint. I am trying with the corresponding command for Sway (wayland): `swaymsg 'output "Iiyama North America PL2792UH 1166300301390" dpms on'` but nothing happens.

For completeness, here's the output of `ddcutil detect` when the external monitor is in sleep mode:

```
$ ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-6
   EDID synopsis:
      Mfg id:               SHP
      Model:                
      Product code:         5152
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2015
      EDID version:         1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:  /dev/i2c-7
   EDID synopsis:
      Mfg id:               IVM
      Model:                PL2792UH
      Product code:         26190
      Serial number:        1166300301390
      Binary serial number: 1390 (0x0000056e)
      Manufacture year:     2020
      EDID version:         1.4
   DDC communication failed
```

dev-zero, 2021-04-01:
@KeyofBlueS yes, I tried that, to no avail. With `dpms off` I get a reconfigure from Sway on the screen configs, but the monitor does not wake up, also not after re-enabling it afterwards. Yes, it is probably not a direct ddcutil issue, but given the low-level nature of tool I thought you here would know best at figuring out where the problem lies.