rockowitz/ddcutil #198: I think color points are parsed incorrectly

danielzfranklin, 2021-04-05:
Thanks for this great tool, it's been very helpful for me as I try to understand EDID well enough to write code to generate pretend EDIDs.

I've spent the last few hours trying to understand EDID's 10-bit decimals. My understanding is that we can convert base-10 to the binary representation by doing (Rust code):

```rust
fn bin_frac(f: f32) -> [u8; 2] {
    let n = (f * 1024_f32).round() as u16;
    let msb = (n >> 2) as u8; // most significant 8 bits
    let lsb = (n & 0b00_000_00011) as u8; // least significant 2 bits
    [msb, lsb]
}
```

When I test this on the examples in section 3.7 of the [EDID 1.4 spec](https://glenwing.github.io/docs/VESA-EEDID-A2.pdf) I get the correct values. However, when I try to confirm this on the output of `ddcutil` it doesn't match the raw bytes of the edid.

Here's my laptop's EDID and `ddcutil`'s parsed colors

```
      White x,y:        0.313, 0.329
      Red   x,y:        0.590, 0.370
      Green x,y:        0.350, 0.555
      Blue  x,y:        0.155, 0.110
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 30 e4 e5 05 00 00 00 00   ........0.......
      +0010   00 1c 01 04 a5 22 13 78 02 38 d5 97 5e 59 8e 27   .....".x.8..^Y.'
      +0020   1c 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 24 36 80 a0 70 38 1f 40 30 20   ......$6..p8.@0 
      +0040   35 00 58 c2 10 00 00 19 00 00 00 00 00 00 00 00   5.X.............
      +0050   00 00 00 00 00 00 00 00 00 00 00 00 00 fe 00 4c   ...............L
      +0060   47 20 44 69 73 70 6c 61 79 0a 20 20 00 00 00 fe   G Display.  ....
      +0070   00 4c 50 31 35 36 57 46 43 2d 53 50 44 31 00 3e   .LP156WFC-SPD1.>
```

Based on my code the red x of `0.590` should have been parsed from a most significant 8 bits of 151 and a least significant 8 bits of 0. The most significant 8 bits should be the EDID's 27th byte, but that has a value of `97`.

rockowitz, 2021-04-05:
You've confused decimal and hexadecimal values.

For your EDID: 
  byte 19h, has value 38h = 00111000b
  rx low bits are bits 0,1, i.e. 00

  byte 1Bh has value 97h = 10010111b = 151d 

 (151*4 + 0) / 1024 = .5898

Your first sentence can be restated more completely as  "... the red x of .590  should have been parsed from a most significant 8 bits of **151d** = **97h**, and a least significant **2** bits of **0d**.   The most significant 8 bits should be the EDIDs 27th (i.e. 19h) byte, which has a value of **97h** = **151d**.

danielzfranklin, 2021-04-05:
Thanks for the detailed correction

closed 2021-04-05 by danielzfranklin