rockowitz/ddcutil #201: Nvidia 465.24 broke my ddcutil

paulhargreaves, 2021-04-16:
Arch user. Upgraded and along with a new kernel, new versions of nvidia-dkms/utils appeared.
These appear to stop ddcutil finding my monitor, and ddcutil investigate shows missing 0x37/0x50.

Reverted kernel (no difference), but on reverting the dkms ddcutil is working again.

Attached interrogate's from both configs - 51113 is 460.67 (working) and 51114 is nvidia 465.24.02 (broken).

[ddcutil-interrogate-51113.txt](https://github.com/rockowitz/ddcutil/files/6324593/ddcutil-interrogate-51113.txt)
[ddcutil-interrogate-51114.txt](https://github.com/rockowitz/ddcutil/files/6324594/ddcutil-interrogate-51114.txt)

No other differences, literally a quick downgrade/reboot.

rockowitz, 2021-04-16:
interrogate does not find i2cdetect.  Please install package i2c-tools and run i2cdetect.  Does it find slave addresses x50 and x37?

paulhargreaves, 2021-04-17:
Installed. i2cdetect -l shows:
```
$ sudo i2cdetect -l
i2c-3	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-1	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-4	i2c       	NVIDIA i2c adapter 4 at 48:00.0 	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-5	i2c       	NVIDIA i2c adapter 5 at 48:00.0 	I2C adapter
$ sudo i2cdetect 4
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-4.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
$ sudo i2cdetect 5
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-5.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- 4a 4b -- -- -- -- 
50: 50 -- -- -- 54 -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --          
```               

It does look like it's seeing 37 and 50, if I'm reading it right? My first time running i2cdetect...

There are 2 adapters in the host, a 1030 (attached to the monitor via HDMI) and a 2070 (attached to the same monitor via DP) - i2cdetect is showing the 1030.

Full interrogate with i2cdetect installed:  
[withi2cdetect.txt](https://github.com/rockowitz/ddcutil/files/6328765/withi2cdetect.txt)

I haven't downgraded the drivers to re-run i2cdetect in the working state. If it's needed, happy to.

paulhargreaves, 2021-04-18:
FYI - downgraded drivers and re-ran, i2cdetect shows exactly the same results, nothing on i2c-4, same 6 entries on i2c-5.

paulhargreaves, 2021-04-18:
And found someone else complaining on the Nvidia forum about it: [https://forums.developer.nvidia.com/t/465-24-02-regression-i2c-gddccontrol-doesnt-work-after-suspend-resume/175307](https://forums.developer.nvidia.com/t/465-24-02-regression-i2c-gddccontrol-doesnt-work-after-suspend-resume/175307)

rockowitz, 2021-04-18:
It pretty clearly is a Nvidia driver problem.  A couple things I note.  First, every attempt to read the EDID returns a repeating 4 byte pattern, typically: 
~~~
                 +0          +4          +8          +c            0   4   8   c   
         +0000   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0010   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0020   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0030   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0040   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0050   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0060   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
         +0070   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
~~~

Attempting to read feature x10 (brightness) returns a repeating pattern as well: 
` 6e 88 02 00 6e 88 02 00 88 02 00`

instead of something like: 
`6e 88 02 00 10 00 00 64 00 57 97`

It would be interesting to see if **get-edid||parse-edid** (package read-edid) can find the EDID.  Also **xrandr --verbose** might show the EDID.  (Though success here with failure everywhere else would mean that X found some other way such as the video BIOS to read the EDID.)

I gave up on using the Nvidia driver on my primary workstation some time ago.  Much too painful.  Often the driver would stop working with a new kernel, and fail to recompile because of the latest kernel changes.  Plus the driver developers are not very responsive to I2C problems.  The amdgpu driver is part of the kernel, so AMD cards simply do not have these problems, and bug reporting is part of the normal kernel/freedesktop.org bug tracking.

paulhargreaves, 2021-04-19:
I was using nvidia because I had vga=off in boot to help with video card passthough (2 years). I've just retested and don't need that, so switched to nouveau. So, at least for me, I'm no longer needing a fix, but I suspect other people will start with current versions of drivers and be stuck, thinking it just doesn't work, rather than knowing it could work if they pick an earlier driver.

paulhargreaves, 2021-04-19:
I'll briefly switch back to nvidia, 465 driver to run the tests you asked for.

```
$ sudo get-edid | parse-edid
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
No EDID on bus 0
No EDID on bus 2
No EDID on bus 3
No EDID on bus 4
2 potential busses found: 1 5
Will scan through until the first EDID is found.
Pass a bus number as an option to this program to go only for that one.
Bus 1 doesn't really have an EDID...
256-byte EDID successfully retrieved from i2c bus 5
Looks like i2c was successful. Have a good day.
Checksum Correct

Section "Monitor"
	Identifier "S2719DGF"
	ModelName "S2719DGF"
	VendorName "DEL"
	# Monitor Manufactured week 3 of 2020
	# EDID version 1.3
	# Digital Display
	DisplaySize 600 340
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-230
	VertRefresh 40-144
	# Maximum pixel clock is 600MHz
	#Not giving standard mode: 1280x800, 60Hz
	#Not giving standard mode: 1680x1050, 60Hz
	#Not giving standard mode: 1920x1200, 60Hz
	#Not giving standard mode: 1152x864, 75Hz
	#Not giving standard mode: 1600x1200, 60Hz
	#Not giving standard mode: 1280x1024, 60Hz
	#Not giving standard mode: 1920x1080, 60Hz

	#Extension block found. Parsing...
#WARNING: I may have missed a mode (CEA mode 90)
	Modeline 	"Mode 16" +hsync -vsync 
	Modeline 	"Mode 0" +hsync -vsync 
	Modeline 	"Mode 1" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 2" 74.250 1920 2008 2052 2200 1080 1082 1087 1125 +hsync +vsync interlace
	Modeline 	"Mode 3" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 5" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
	Modeline 	"Mode 6" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 7" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 8" 25.200 640 656 752 800 480 490 492 525 -hsync -vsync
	Modeline 	"Mode 9" 27.027 1440 1478 1602 1716 480 484 487 525 -hsync -vsync interlace
	Modeline 	"Mode 10" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 11" 27.000 720 732 796 864 576 581 586 625 -hsync -vsync
	Modeline 	"Mode 12" 27.000 1440 1464 1590 1728 576 578 581 625 -hsync -vsync interlace
	Modeline 	"Mode 13" 74.250 1280 1720 1760 1980 720 725 730 750 +hsync +vsync
	Modeline 	"Mode 14" 74.250 1920 2448 2492 2640 1080 1082 1089 1125 +hsync +vsync interlace
	Modeline 	"Mode 15" 148.500 1920 2448 2492 2640 1080 1084 1089 1125 +hsync +vsync
	Modeline 	"Mode 17" +hsync +vsync interlace
	Modeline 	"Mode 18" +hsync +vsync 
	Modeline 	"Mode 19" -hsync -vsync 
	Option "PreferredMode" "Mode 16"
EndSection

```

```
$ xrandr --verbose
Screen 0: minimum 8 x 8, current 2560 x 1440, maximum 32767 x 32767
DVI-D-0 disconnected (normal left inverted right x axis y axis)
[... 156 lines cut ...]

```
$ sudo ddcutil environment | grep nvidia
   Driver name:         nvidia
   Found video driver module:      nvidia
   Found other loaded module:      i2c_nvidia_gpu
   Loaded drm module depends on:   nvidia_drm,drm_kms_helper,
   Module nvidia           is loaded
   Drivers detected:              nvidia, vfio-pci
$ sudo ddcutil environment | grep nouv
   Module nouveau          is NOT loaded
$  sudo ddcutil detect
No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.
$ 

$ sudo dmesg | grep 465.24.02
[    2.747928] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  465.24.02  Thu Apr  8 17:55:47 UTC 2021
[    2.938849] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  465.24.02  Thu Apr  8 17:50:51 UTC 2021
$ 

```

Switching back to nouveau, it's a trivial effort for me to switch between drivers, so if there are other things you're interested in, happy to run them.

rockowitz, 2021-04-19:
Now this gets weird.  read-edid (and also xrandr) read the EDID successfully. read-edid reads the EDID 1 byte at a time.  I've 
tried this, and there are cases for which it does not work.  The **--f4** option triggers tests of alternative ways to read the EDID.
Please send the output of the following command with the 465 driver: **ddcutil detect --verbose --f4**

Background:  Support for I2C on monitors connected to a docking station was added in kernel 5.10 for DRM video drivers (amdgpu, i915, etc.) The implementation has bugs, and the **--f4** option tests strategies for reading the EDID looking for a way to work around the bugs.

paulhargreaves, 2021-04-19:
```
$ sudo dmesg | grep 465.24.02
[    2.684750] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  465.24.02  Thu Apr  8 17:55:47 UTC 2021
[    2.806856] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  465.24.02  Thu Apr  8 17:50:51 UTC 2021
$ sudo ddcutil detect --verbose --f4
ddcutil version:            1.1.0
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  true
   Dynamic sleep adjustment enabled:            false

Configuration file:         (none)

===========> IO STRATEGY 1:
   i2c_io_strategy:          FILEIO
   EDID read uses I2C layer: Directly
   EDID read bytewise:       false
   write before read:        false
   EDID read size:           128
   Valid displays:           0
   Elapsed time:             0.011 seconds

No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.

===========> IO STRATEGY 2:
   i2c_io_strategy:          FILEIO
   EDID read uses I2C layer: Directly
   EDID read bytewise:       false
   write before read:        false
   EDID read size:           256
   Valid displays:           0
   Elapsed time:             0.011 seconds

No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.

===========> IO STRATEGY 3:
   i2c_io_strategy:          FILEIO
   EDID read uses I2C layer: Directly
   EDID read bytewise:       false
   write before read:        false
   EDID read size:           dynamic
   Valid displays:           0
   Elapsed time:             0.022 seconds

No displays found.
Is DDC/CI enabled in the monitor's on screen display?
Run "ddcutil environment" to check for system configuration problems.

===========> IO STRATEGY 4:
   i2c_io_strategy:          FILEIO
   EDID read uses I2C layer: Directly
   EDID read bytewise:       false
   write before read:        true
   EDID read size:           128
   Valid displays:           0
   Elapsed time:             0.011 seconds

No displays found.
[... 724 lines cut ...]
   13 FILEIO    I2C Layer   DNA         false    128       0        0.638
   14 FILEIO    I2C Layer   DNA         false    256       0        0.685
   15 FILEIO    I2C Layer   DNA         false    dynamic   0        0.646
   16 FILEIO    I2C Layer   DNA         true     128       0        0.640
   17 FILEIO    I2C Layer   DNA         true     256       0        0.686
   18 FILEIO    I2C Layer   DNA         true     dynamic   0        0.646
   19 IOCTL     Directly    false       false    128       0        0.012
   20 IOCTL     Directly    false       false    256       0        0.012
   21 IOCTL     Directly    false       false    dynamic   0        0.023
   22 IOCTL     Directly    false       true     128       0        0.011
   23 IOCTL     Directly    false       true     256       0        0.011
   24 IOCTL     Directly    false       true     dynamic   0        0.022
   25 IOCTL     Directly    true        false    128       0        0.635
   26 IOCTL     Directly    true        false    256       0        0.683
   27 IOCTL     Directly    true        false    dynamic   0        0.645
   28 IOCTL     Directly    true        true     128       0        0.637
   29 IOCTL     Directly    true        true     256       0        0.686
   30 IOCTL     Directly    true        true     dynamic   0        0.646
   31 IOCTL     I2C Layer   DNA         false    128       0        0.639
   32 IOCTL     I2C Layer   DNA         false    256       0        0.686
   33 IOCTL     I2C Layer   DNA         false    dynamic   0        0.647
   34 IOCTL     I2C Layer   DNA         true     128       0        0.639
   35 IOCTL     I2C Layer   DNA         true     256       0        0.685
   36 IOCTL     I2C Layer   DNA         true     dynamic   0        0.647
```

JakubFojtik, 2021-04-19:
Hi, I have the same problem on a similar setup. Including output for HDMI and also DisplayPort just for comparison, although DDC previously worked only on HDMI.

ddcutil detect --verbose --f4:
[HDMI](https://github.com/rockowitz/ddcutil/files/6337686/interrogate_hdmi.txt)
[DP](https://github.com/rockowitz/ddcutil/files/6337687/interrogate_dp.txt)

paulhargreaves, 2021-04-19:
For completeness, I reverted the drivers back to 460, and re-ran the f4. For brevity here is the last part of it:

```
Display 36
   I2C bus:  /dev/i2c-5
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-5/name:   NVIDIA i2c adapter 5 at 48:00.0
   EDID synopsis:
      Mfg id:               DEL
      Model:                S2719DGF
      Product code:         53477
      Serial number:        DCZ0CY2
      Binary serial number: 809060437 (0x30394855)
      Manufacture year:     2020,  Week: 3
   VCP version:         2.1
   Controller mfg:      RealTek
   Firmware version:    1.0
   Monitor returns DDC Null Response for unsupported features: false

   SUMMARY

      I2C IO    EDID        EDID Read   Write    EDID Read Valid    Seconds
      Strategy  Method      Bytewise    b4 Read  Size      Displays         
      =======   ========    =========   =======  ========= ======== =======
    1 FILEIO    Directly    false       false    128       1        0.205
    2 FILEIO    Directly    false       false    256       1        0.236
    3 FILEIO    Directly    false       false    dynamic   1        0.216
    4 FILEIO    Directly    false       true     128       1        0.208
    5 FILEIO    Directly    false       true     256       1        0.236
    6 FILEIO    Directly    false       true     dynamic   1        0.215
    7 FILEIO    Directly    true        false    128       1        0.247
    8 FILEIO    Directly    true        false    256       1        0.316
    9 FILEIO    Directly    true        false    dynamic   1        0.255
   10 FILEIO    Directly    true        true     128       1        0.247
   11 FILEIO    Directly    true        true     256       1        0.317
   12 FILEIO    Directly    true        true     dynamic   1        0.254
   13 FILEIO    I2C Layer   DNA         false    128       1        0.246
   14 FILEIO    I2C Layer   DNA         false    256       1        0.319
   15 FILEIO    I2C Layer   DNA         false    dynamic   1        0.256
   16 FILEIO    I2C Layer   DNA         true     128       1        0.249
   17 FILEIO    I2C Layer   DNA         true     256       1        0.316
   18 FILEIO    I2C Layer   DNA         true     dynamic   1        0.255
   19 IOCTL     Directly    false       false    128       1        0.208
   20 IOCTL     Directly    false       false    256       1        0.235
   21 IOCTL     Directly    false       false    dynamic   1        0.215
   22 IOCTL     Directly    false       true     128       1        0.208
   23 IOCTL     Directly    false       true     256       1        0.235
   24 IOCTL     Directly    false       true     dynamic   1        0.214
   25 IOCTL     Directly    true        false    128       1        0.248
   26 IOCTL     Directly    true        false    256       1        0.315
   27 IOCTL     Directly    true        false    dynamic   1        0.254
   28 IOCTL     Directly    true        true     128       1        0.246
   29 IOCTL     Directly    true        true     256       1        0.317
   30 IOCTL     Directly    true        true     dynamic   1        0.255
   31 IOCTL     I2C Layer   DNA         false    128       1        0.248
   32 IOCTL     I2C Layer   DNA         false    256       1        0.317
   33 IOCTL     I2C Layer   DNA         false    dynamic   1        0.257
   34 IOCTL     I2C Layer   DNA         true     128       1        0.247
   35 IOCTL     I2C Layer   DNA         true     256       1        0.320
   36 IOCTL     I2C Layer   DNA         true     dynamic   1        0.255
```

This probably doesn't help much other than show that the f4 functionality works on my host if the drivers aren't broken ;-)

rockowitz, 2021-04-19:
@paulhargreaves @JakubFojtik If either of you can easily uninstall ddcutil 1.1.0 and replace it with release 0.9.9, please do so and let me know if the problem with latest Nvidia driver still exists.  That version predates all the recent changes to DDC communication to handle docking station connected displays.  Thanks.

rockowitz, 2021-04-21:
@paulhargreaves  @JakubFojtik  The new 1.1.1-dev branch contains changes to the **environment** command to test additional variations of reading the EDID, seeking a work-around for the 465 driver problem.  If you can, please build from that branch and submit the output of **ddcutil environment --verbose**.  Note that this branch requires that packages libi2c0 and libi2c-dev be installed.

Please submit output as attachments of some sort rather than inline. That makes it much easier to follow an issue, and easier to work with the output.  Thank you.

paulhargreaves, 2021-04-21:
Sorry for inlining, understood.
I've got i2c-tools installed, does that satisfy the requirement? Arch doesn't have libi2c0 nor libi2c-dev, I've had a brief look at Debian and I _think_ it does, but can't be certain.

Compiled cleanly without error anyway, so will reboot with 465 driver installed and provide verbose output.

paulhargreaves, 2021-04-21:
[ddcutilverbose.txt](https://github.com/rockowitz/ddcutil/files/6352629/ddcutilverbose.txt)

rockowitz, 2021-04-21:
Thanks.  Looks like there's a workaround for reading the EDID.  I have a hunch as to what might work for DDC communication.

rockowitz, 2021-04-22:
Unfortunately, while I can work around the bug in the I2C read() implementation when reading the EDID, I don't see a way to do so when reading DDC responses.  The Nvidia 465.24 driver is simply unusable for DDC communication.

paulhargreaves, 2021-05-02:
FYI - recently released 465.27 doesn't fix the problem.

paulhargreaves, 2021-05-20:
FYI - recently released 465.31 is also the same.

luisbocanegra, 2021-06-14:
Have you guys tried using the nvidia module settings as per the [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia/) from the documentation? Usign the module settings makes ddcutil detect my monitor again and it seems to be working fine on the latest nvidia driver from Arch.
```
❯ sudo ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-4
   EDID synopsis:
      Mfg id:               CMN
      Model:                
      Product code:         5389
      Serial number:        
      Binary serial number: 0 (0x00000000)
      Manufacture year:     2018,  Week: 42
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Display 1
   I2C bus:  /dev/i2c-8
   EDID synopsis:
      Mfg id:               DEL
      Model:                DELL P2419H
      Product code:         53466
      Serial number:        2K40943
      Binary serial number: 811094082 (0x30585042)
      Manufacture year:     2020,  Week: 23
   VCP version:         2.1
```
```
❯ modinfo nvidia | grep version 
version:        465.31
```
```
❯ cat /etc/modprobe.d/nvidia.conf
options nvidia "NVreg_DynamicPowerManagement=0x02"
options nvidia "NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100"
```
Attached both interrogates with and without kernel configuration just in case it may be helpful.
[ddcutil-1.1.0-detect-verbose-f4-WORKING.txt](https://github.com/rockowitz/ddcutil/files/6648621/ddcutil-1.1.0-detect-verbose-f4.txt)
[ddcutil-1.1.0-detect-verbose-f4-NOT-WORKING.txt](https://github.com/rockowitz/ddcutil/files/6648683/ddcutil-1.1.0-detect-verbose-f4-NOT-WORKING.txt)

paulhargreaves, 2021-06-15:
Indeed, this works. Thanks for pointing it out!

For my config I only needed the options nvidia NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100 entry, no changes to xorg nor 90-nvidia_i2c.conf.

`$ modinfo nvidia | grep version
version:        465.31
$ ddcutil detect
Display 1
   I2C bus:  /dev/i2c-5
   EDID synopsis:
      Mfg id:               DEL
      Model:                S2719DGF
      Product code:         53477
      Serial number:        DCZ0CY2
      Binary serial number: 809060437 (0x30394855)
      Manufacture year:     2020,  Week: 3
   VCP version:         2.1
# cat /etc/modprobe.d/nvidia-drm.conf 
options nvidia NVreg_RegistryDwords=RMUseSwI2c=0x01;RMI2cSpeed=100`

The other fixes on that page (xorg) aren't needed on my system.

closed 2021-06-15 by paulhargreaves