rockowitz/ddcutil #203: [AOC 2778X] Invalid display : DDC communication failed

BaptisteBdn, 2021-04-26:
Hi,

I have an AOC 2778X connected through HDMI from an Intel Graphic 530 (iGPU) and I have the following error : 
```
ddcutil version:            1.1.0
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  true
   Dynamic sleep adjustment enabled:            false

Configuration file:         (none)
Invalid display
   I2C bus:  /dev/i2c-0
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-0/name:   i915 gmbus dpc
   EDID synopsis:
      Mfg id:               AOC
      Model:                2778X
      Product code:         10104
      Serial number:        E33G3BA000665
      Binary serial number: 665 (0x00000299)
      Manufacture year:     2016,  Week: 12
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

The DDC/CI option in the monitor is enabled.
It seems to be working with another display (no errors at least).
Output of `sudo ddcutil interrogate` : [log.txt](https://github.com/rockowitz/ddcutil/files/6380501/log.txt)
I have a GPU passthrough that's why I am not using the NVIDIA card.

Do you have any idea what it could be ?

Thanks a lot

rockowitz, 2021-04-27:
That ddcutil works with your other monitor indicates it's a problem in the monitor's DDC implementation.   This is confirmed by lines 523-530 in the interrogate log - what should be a get feature value response is just 0 bytes.  You could try increasing the amount of time ddcutil sleeps before trying to read the monitor's response.  Try option **--sleep-multiplier 4.0**.  If that works check if a lower sleep multiplier value also works.  (There's no point in trying a multiplier value greater than 4.)

BaptisteBdn, 2021-04-27:
I still have the same error when I'm trying with --sleep-multiplier 4.0 : [log.txt](https://github.com/rockowitz/ddcutil/files/6382560/log.txt)

BaptisteBdn, 2021-04-27:
I tried to used ClickMonitorDDC_7_2 on my Windows virtual machine and it turns out I can see and use 3 inputs : DP, DVI, VGA.
However the linux host uses the HDMI. Is there anything I can do to fix that ?

BaptisteBdn, 2021-04-27:
Swapping the cable between the host and the VM worked.
I can now use `sudo ddcutil -d 1 setvcp 60 17` to switch to HDMI input.

closed 2021-04-29 by BaptisteBdn