rockowitz/ddcutil #204: ddcutil can't work with HP ZHAN 66 AMD! Communication failed!

C1oud555, 2021-04-28:
Please help me !!! 
It's a laptap with AMD cpui and integrated gpu. An external moniter connected to the laptap through the HDMI port. But it can't work anyway.
[detect.txt](https://github.com/rockowitz/ddcutil/files/6392679/detect.txt)
[environment.txt](https://github.com/rockowitz/ddcutil/files/6392685/environment.txt)
[interrogate.txt](https://github.com/rockowitz/ddcutil/files/6392681/interrogate.txt)

rockowitz, 2021-04-29:
Lines 594/601 of the interrogate output, and the comparable lines of 
environment  show that the host is receiving a corrupt DDC response 
packet to the request for the value of feature x10 (brightness).  Many 
of the response bytes are duplicated. Most likely a poor DDC 
implementation in the monitor, conceivably the HDMI cable.

Things to try:
- Does ddcutil work with a different monitor?
- Use a different HDMI cable
- Use a different connection type, if possible - in this case 
DisplayPort instead of HDMI
- Use option "--sleep-multiplier"  to give the monitor more time to 
assemble its responses, e.g. "ddcutil detect --verbose 
--sleep-multiplier 4.0"

Good luck.

C1oud555, 2021-04-29:
I tried the 2 and 4 suggestions since I don't have an extra monitor or DP cable. Both of them don't work.
Could it be the graphics card problem? Because the monitor can work with Deepin Linux distribution under AMD 3400G which is similar to my laptop's AMD 4500U

rockowitz, 2021-04-30:
The only other suggestion I have for you is to increase the  number of retries for all ddc operations from the default 10 to the maximum 15 using option **--maxtries 15,15,15**, but frankly if the default number of retries doesn't work increasing the value rarely has any effect.

Other than that, I'm afraid that I don't have any further suggestions for you.  DDC is a finicky protocol, and sometimes a particular video chipset/driver cannot handle a particular monitor with a marginal DDC implementation.

closed 2021-05-09 by C1oud555