rockowitz/ddcutil #205: KDE Power management system crashes on screen sleep

OlavStornes, 2021-05-03:
# Summary
When idle for a few minutes (when "Dim screen" should be triggered), a crash notification is shown and the screen stays on. This also happens if i have locked the desktop 

When not logged in, the screen dims as expected.

I was expecting that this was a [KDE-issue](https://bugs.kde.org/show_bug.cgi?id=436340), but i was guided here

# Steps to reproduce
1. Log into pc
2. Wait for the screen to dim
3. Receive a notification that kde power settings has crashed, and an entry in journald

# Observed result
Screen stays on for an indefinite time

# Expected result
Screen should dim

# Software/os-versions
EndeavourOS: 5.11.16-zen1-1-zen
KDE Plasma Version: 5.21.4 
KDE Frameworks Version: 5.81.0
Qt Version: 5.15.2
GPU: Nvidia geforce RTX 2070 w. driver version 465.27

Is there something i could attempt to further debug this?

```
03.05.2021 15:42	systemd-coredump	Process 35439 (org_kde_powerde) of user 1000 dumped core.

Stack trace of thread 35439:
#0  0x00007f1099a76ef5 raise (libc.so.6 + 0x3cef5)
#1  0x00007f1099a76f80 __restore_rt (libc.so.6 + 0x3cf80)
#2  0x00007f1099a76ef5 raise (libc.so.6 + 0x3cef5)
#3  0x00007f1099a60862 abort (libc.so.6 + 0x26862)
#4  0x00007f1099a60747 __assert_fail_base.cold (libc.so.6 + 0x26747)
#5  0x00007f1099a6f646 __assert_fail (libc.so.6 + 0x35646)
#6  0x00007f1092046c57 n/a (libddcutil.so.4 + 0x1bc57)
#7  0x00007f109203b530 ddca_open_display2 (libddcutil.so.4 + 0x10530)
#8  0x00007f1093402855 n/a (powerdevilupowerbackend.so + 0x17855)
#9  0x00007f1093403622 n/a (powerdevilupowerbackend.so + 0x18622)
#10 0x00007f109a0d1b76 n/a (libQt5Core.so.5 + 0x2edb76)
#11 0x00007f109a0d5c0b _ZN6QTimer7timeoutENS_14QPrivateSignalE (libQt5Core.so.5 + 0x2f1c0b)
#12 0x00007f109a0c725f _ZN7QObject5eventEP6QEvent (libQt5Core.so.5 + 0x2e325f)
#13 0x00007f109a09a81a _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent (libQt5Core.so.5 + 0x2b681a)
#14 0x00007f109a0f26a5 _ZN14QTimerInfoList14activateTimersEv (libQt5Core.so.5 + 0x30e6a5)
#15 0x00007f109a0f2f52 n/a (libQt5Core.so.5 + 0x30ef52)
#16 0x00007f10980ac02c g_main_context_dispatch (libglib-2.0.so.0 + 0x5402c)
#17 0x00007f10980ffb59 n/a (libglib-2.0.so.0 + 0xa7b59)
#18 0x00007f10980a9781 g_main_context_iteration (libglib-2.0.so.0 + 0x51781)
#19 0x00007f109a0f3321 _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE (libQt5Core.so.5 + 0x30f321)
#20 0x00007f109a09917c _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.so.5 + 0x2b517c)
#21 0x00007f109a0a1634 _ZN16QCoreApplication4execEv (libQt5Core.so.5 + 0x2bd634)
#22 0x000056237d1021eb n/a (org_kde_powerdevil + 0x61eb)
#23 0x00007f1099a61b25 __libc_start_main (libc.so.6 + 0x27b25)
#24 0x000056237d10225e _start (org_kde_powerdevil + 0x625e)

Stack trace of thread 35447:
#0  0x00007f1099b2e37f __poll (libc.so.6 + 0xf437f)
#1  0x00007f10980ffae8 n/a (libglib-2.0.so.0 + 0xa7ae8)
#2  0x00007f10980ab593 g_main_loop_run (libglib-2.0.so.0 + 0x53593)
#3  0x00007f10932ad558 n/a (libgio-2.0.so.0 + 0x102558)
#4  0x00007f10980da0c1 n/a (libglib-2.0.so.0 + 0x820c1)
#5  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#6  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

Stack trace of thread 35444:
#0  0x00007f1099b2e37f __poll (libc.so.6 + 0xf437f)
#1  0x00007f109937763b n/a (libxcb.so.1 + 0xc63b)
#2  0x00007f109937937b xcb_wait_for_event (libxcb.so.1 + 0xe37b)
#3  0x00007f109454a161 n/a (libQt5XcbQpa.so.5 + 0x5e161)
#4  0x00007f1099eb208f n/a (libQt5Core.so.5 + 0xce08f)
#5  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#6  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

Stack trace of thread 35461:
#0  0x00007f1099b30bf3 pselect (libc.so.6 + 0xf6bf3)
#1  0x00007f10958fd524 n/a (libusbmuxd-2.0.so.6 + 0x2524)
#2  0x00007f10958fe8a9 n/a (libusbmuxd-2.0.so.6 + 0x38a9)
#3  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#4  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

Stack trace of thread 35446:
#0  0x00007f1099b2e37f __poll (libc.so.6 + 0xf437f)
#1  0x00007f10980ffae8 n/a (libglib-2.0.so.0 + 0xa7ae8)
#2  0x00007f10980a9781 g_main_context_iteration (libglib-2.0.so.0 + 0x51781)
#3  0x00007f10980a97d2 n/a (libglib-2.0.so.0 + 0x517d2)
#4  0x00007f10980da0c1 n/a (libglib-2.0.so.0 + 0x820c1)
#5  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#6  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

Stack trace of thread 35448:
#0  0x00007f1099b3a60d recvmsg (libc.so.6 + 0x10060d)
#1  0x00007f1097ff4b57 n/a (libudev.so.1 + 0x10b57)
#2  0x00007f1097ff55c3 udev_monitor_receive_device (libudev.so.1 + 0x115c3)
#3  0x00007f109204a232 n/a (libddcutil.so.4 + 0x1f232)
#4  0x00007f10980da0c1 n/a (libglib-2.0.so.0 + 0x820c1)
#5  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#6  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

Stack trace of thread 35445:
#0  0x00007f1099b2e37f __poll (libc.so.6 + 0xf437f)
#1  0x00007f10980ffae8 n/a (libglib-2.0.so.0 + 0xa7ae8)
#2  0x00007f10980a9781 g_main_context_iteration (libglib-2.0.so.0 + 0x51781)
#3  0x00007f109a0f3321 _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE (libQt5Core.so.5 + 0x30f321)
#4  0x00007f109a09917c _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.so.5 + 0x2b517c)
#5  0x00007f1099eb0ea2 _ZN7QThread4execEv (libQt5Core.so.5 + 0xccea2)
#6  0x00007f109a352098 n/a (libQt5DBus.so.5 + 0x17098)
#7  0x00007f1099eb208f n/a (libQt5Core.so.5 + 0xce08f)
#8  0x00007f1098b1f299 start_thread (libpthread.so.0 + 0x9299)
#9  0x00007f1099b39053 __clone (libc.so.6 + 0xff053)

```

rockowitz, 2021-05-05:
I suspect there are 2 things going on here.  First, there are problems with the Nvidia 465.27 driver.  (See issues #201, #206.) Does the failure occur when using a 460 series driver?

That said, the assert failures indicate that the ddca_open_display2() call is encountering a state regarded as logically impossible.  It should handle these cases more gracefully.  

I believe the PowerDevil folks are using their own private copy of libddcutil, since ddcutil is present only in some distributions.  I will look into how it can be replaced by the the latest libddcutil version.

d-vogel, 2021-05-05:
Hey,
I'm the person who implemented ddcutil support in powerdevil. I did it years ago and that was my first contribution. Since then the feature has not really been tested and did not evolve. The maintainer of Powerdevil was enthusiastic for the feature, tested it, but it seems the display brightness infrastructure from powerdevil requires a massive overhaul, which was above my skill level. As a result the feature stayed a bit of a "hidden" feature, disabled by default and no distribution enabled it by default.

So there's a good chance the ddcutil API version used in powerdevil is now mostly deprecated.
We weren't using any special version of ddcutil. The one used by the system was selected with some CMakelist magic.

Thinking back about it, a better option would have been to develop a plasmoid (panel applet) for ddcutil. That would have allowed to have independent brightness adjustment for internal screen and external monitors on laptops (which is not supported by powerdevil). I keep that on my list of todos for the time when my professional life will clear up.

rockowitz, 2021-05-05:
Hi Dorian, Thanks for jumping in. If I understand you correctly,it's up to the user to install libddcutil, which may or may not be included in the distribution they use.  PowerDevil links to libddcutil using the dlopen() family of calls (or equivalent).  If if finds libddcutil it uses it, if not it continues to function without it.  Since no distribution enabled the feature, any user has had to build their own copy of PowerDevil, and also download one of my builds libddcutil builds (from Launchpad, Fedora COPR, or the openSUSE build service), or to build it themselves from source.  @OlavStornes, Does this describe your situation?

The ddcutil implementation has actually been quite stable for some time now.  The major version number was bumped because  fields were added to structs.  Functions have been added, largely based on the needs exposed by ddcui, but these have been upwardly compatible.  So all that has been necessary is to recompile existing callers. This may, however, need to change in order to support display attachment/detachment.

Regards, Sanford

OlavStornes, 2021-05-05:
Hi, and thank you both for answering. 

> Does the failure occur when using a 460 series driver?

I'll try to rollback this week. For now, i haven't attempted this unfortunately

> Does this describe your situation?

As far as i can see it doesn't seem to describe my situation, although i could be unknowingly wrong about it.  
I cant seem to find `libddcutil`. However, i have ddcutil (i assume they are, for all intents and purposes the same?)

```bash
pacman -Q | grep '\(ddcutil\|powerdevil\)' -> 
    ddcutil 1.1.0-1
    powerdevil 5.21.5-1
```

rockowitz, 2021-05-05:
On Debian, Ubuntu, Fedora etc. by convention there are 3 packages: ddcutil, libddcutilN, and libddcutil-dev.  All come from the same source "package", also named ddcutil.  Since pacman builds from source, there's only ddcutil. 

ddcutil 1.1.0 introduced configuration files. (Arch is very good at picking up the latest version.) Create file $HOME/.config/ddcutil/ddcutilrc and include the following lines: 

~~~
[libddcutil]
options : --trcfile api_displays --trcfile ddc_packet_io  --trcfile ddc_display_lock --trcfile ddc_watch_displays
~~~
Then of course recreate the problem while capturing the trace output and submit  the output as an attachment. 

If you are comfortable building from the current development branch on github, I can provide an option string that is less insanely voluminous while yielding more useful information. 

Re PowerDevil, did you do something to enable use of the libddcutil option, or did you build it as obtained from the repository? If the later, Dorian's statement that no distribution enables the feature by default is incorrect.

OlavStornes, 2021-05-06:
Gotcha.

> Re PowerDevil, did you do something to enable use of the libddcutil option, or did you build it as obtained from the repository? If the later, Dorian's statement that no distribution enables the feature by default is incorrect.

I was installing KDE plasma via the online installer of EndeavourOS, so i can't vouch if i explicitly chose to enable the use ddcutil at that point (when installing via endeavourOS i could choose which DE i wanted, so i chose KDE plasma with all related packages). However, the arch packages [declares it as a dependency for powerdevil](https://archlinux.org/packages/extra/x86_64/powerdevil/).

> Then of course recreate the problem while capturing the trace output and submit the output as an attachment.

I tried to write a file with that snippet, but it didn't seem to give me a trace of some sorts.

-   Wrote it to `~/.config/ddcutil/ddcutilrc`
-   Reset my computer to ensure it got loaded properly
-   Looked for a trace file
    -   I don't have **XDG config path** or **XDG data home** set, so i assume the default folder would be at `~/.local/share/ddcutil` or something similar. That should've done the trick, or did i miss something?

Another interesting find while i was debugging: This error happens specifically on screen dimming, but not on shutting the screen off. I didn't realize this before i disabled "dim screen" as an option, since the timeout was assumedly reset whenever powerdevil crashed

This log message came immediately before the crash: [tmplogfiles.txt](https://github.com/rockowitz/ddcutil/files/6435967/tmplogfiles.txt)

This line peaked my interest:

    org.kde.powerdevil: [DDCutilBrightness::setBrightness] failed, trying to detect()

So, to clarify; this is my plasma-related power settings where things work as expected.

[image]

rockowitz, 2021-05-08:
I've fixed a bug that caused the tracing related options for libddcutil in the ini file to ignored.  The fix is in the 1.1.1-dev branch. This branch enables more precise tracing of relevant functions, so the configuration file should look like:
~~~
[libddcutil]
options : --trcfunc ddca_open_display2 --trcfunc ddca_close_display --trcfunc ddc_open_display --trcfunc lock_distinct_display --trcfunc unlock_distinct_display --trcfunc ddc_close_display --trcfunc dummy_display_change_handler --trcfunc watch_displays_using_udev
~~~

The XDG file location spec can be found [here](https://specifications.freedesktop.org/basedir-spec/latest/ar01s03.html).  The bottom line is that unless one or more of the relevant environment variables are set, configuration files for ddcutil are placed in directory $HOME/.config/ddcutil, data files in $HOME/.local/share/ddcutil, and cache files in $HOME/.cache/ddcutil. 

Turning the screen off uses the X11 (or Weston) DPMS api.  Controlling the brightness, on the other hand, requires libddcutil.

When I boot into Arch (as with Ubuntu and Fedora), my power management screen is as shown below, without the option to dim the screen.  Interestingly, Help in all cases documents the screen you show.
[image]

Bottom line is that I need the trace output to pinpoint which assert is failing, and it will very helpful to know if the problem was triggered by the Nvidia 465 driver.

OlavStornes, 2021-05-08:
After a few hours of debugging, i've come to this conclusion:

-   Installed ddcutil from source on the mentioned branch:
-   Added the configuration file as mentioned

```bash
ddcutil --version
---------------------
Applying ddcutil options from /home/olavstornes/.config/ddcutil/ddcutilrc: --trcfunc ddca_open_display2 --trcfunc ddca_close_display --trcfunc ddc_open_display --trcfunc lock_distinct_display --trcfunc unlock_distinct_display --trcfunc ddc_close_display --trcfunc dummy_display_change_handler --trcfunc watch_displays_using_udev
ddcutil 1.1.1
Built with support for USB connected displays.
Built without function failure simulation.
Built with libdrm services.
    
Copyright (C) 2015-2021 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

However, i could not find anything at either location that you mentioned either.

    ls -lahgo ~/.local/share/ddcutil

    total 8,0K
    drwxr-xr-x  2 4,0K mai    8 16:14 .
    drwxr-xr-x 79 4,0K mai    8 19:34 ..

> When I boot into Arch (as with Ubuntu and Fedora), my power management screen is as shown below, without the option to dim the screen. Interestingly, Help in all cases documents the screen you show.

Thats strange to see. When i think of it, i'm not sure if i've ever experimented with the brightness slider either. (for claritys sake, this happens on a desktop over a DPI-connection. Not sure if that is of any relevance).

I do unfortunately need to revert back to the original version of ddcutil, as i need this computer for a day-to-day basis.

rockowitz, 2021-05-09:
Let me try to go through your observations in order.  

Re what I take to be the output of command "ddcutil --version", it's odd that it's reporting ddcutil options applied.  It looks like you've defined the options in the **[global]** or **[ddcutil]** section of the ini file, as opposed to the **[libddcutil]** section.  If it's in the  **global]** section that's ok, as the options would apply to any of ddcutil, libddcutil, or ddcui. That there's no trace output from **ddcutil --version** is expected.  If the command were, for example, **ddcutil detect** there should be output for the traced functions.

That $HOME/.local/share/ddcutil and $HOME/.cache/ddcutil are non-existent or empty is not surprising.  The former currently holds only user supplied feature definitions, the latter currently is used for cached capabilities strings if that is enabled.   Trace output is written to the terminal, though perhaps KDE is redirecting it.

The only meaning of "DPI-connection" that seems at all relevant to me is "Display Parallel Interface", as described [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md), and [here](https://hackaday.com/tag/display-parallel-interface/).  Is that really what you're doing?  Or did you mean to type "DVI"?  If the latter, whether the monitor connection is DVI, HDMI, or DisplayPort should not matter in this situation. 

I appreciate that you need the computer back in its usual state.  Without knowing the logic path that leads to the failure and whether the problem is triggered by the the Nvidia 465 driver, there's not much more I can do.   Thank you for your effort in reporting the problem.

OlavStornes, 2021-05-09:
After sleeping on it, I've noticed i missed to explain a few things and wrote wrong on some other. Debugging while low on energy is apparently not a good idea. For that i should apologize

I tried first to use `[libddcutil]` as the section, but since no logs got written on (my assumed) paths, i guessed wrongly that it was a misconfiguration, and rewrote it to (what i think was) global after some consulting of the documentation.

And yes, it was of course not "DPI-connection", whatever that is. What i really meant to say was DisplayPort.

I'm happy to retry debugging again, but i may need a bit more info beforehand:
You mentioned that the output should be written to terminal, but should that happen when i only run `ddcutil`? I would assume that terminal output happens whenever a process is attached to a specific terminal that executed a specific command, else i would think it would be in journalctl.

If that's the case, is there a command i could run? I tried something akin to `ddcutil interrogate --timestamp --trcfile api_displays --trcfile ddc_packet_io  --trcfile ddc_display_lock --trcfile ddc_watch_displays` which gave me a lot of information, but i'm not sure if that is what you need for debugging purposes.

OlavStornes, 2021-05-10:
Thank you for your time explaining. It really is a solid learning experience! I can give it another go when i know where the terminal output goes (or if there is a file that it logs to)

[Output of environment-command](https://github.com/rockowitz/ddcutil/files/6454650/ddcutil-envs.txt)

rockowitz, 2021-05-11:
The current 1.1.1 branch provisionally implements a facility for writing trace output to a file.  Right now it's activated by the **--s1** option, which takes a file name as an argument, e.g. "--s1 /tmp/libddcutil.log".  The file is emptied (this may change) each time libddcutil is initialized, i.e. when it's loaded. The option is also valid on the **ddcutil** command.   Note that any error messages re opening the trace file go to the terminal and these messages will be lost when **libddcutil** is called from PowerDevil (KDE).  I suggest you test your **--s1** option on the **ddcutil** command line, so you can see any such error messages, e.g. `ddcutil detect --trcfunc ddc_open_display`.

The contents of your **ddcutilrc** file should now be: 
~~~
[libddcutil]
options : --s1 mylog.tmp --tid --ts --trcfunc ddca_open_display2 --trcfunc ddca_close_display --trcfunc ddc_open_display --trcfunc lock_distinct_display --trcfunc unlock_distinct_display --trcfunc ddc_close_display --trcfunc dummy_display_change_handler --trcfunc watch_displays_using_udev
~~~

(The **--tid** and **--ts** options prefix the trace messages with a thread id and timestamp.)

Note that the assert message itself is written as usual to stderr, not to the trace log.

rockowitz, 2021-05-13:
@OlavStornes  The current 1.1.1-dev branch has changes when tracing to a file that you should be aware of:
- The trace file is not erased each time **libddcutil** is loaded, so data will not be lost if the library is automatically restarted after an assert() failure. 
- Improved assert() tracing.

OlavStornes, 2021-05-13:
Huh, what a coincidence. I was literally just pulling from source, and noticed that the git repo was freshly updated.

So, a new try is afoot. I've upgraded, which can be confirmed by:

    ddcutil --version

    ddcutil 1.1.1
    Built with support for USB connected displays.
    Built without function failure simulation.
    Built with libdrm services.
    
    Copyright (C) 2015-2021 Sanford Rockowitz
    License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.

After that i wrote the aforementioned config back to `~/.config/ddcutil/ddcutilrc`

However, nothing seemed to happen when i specified `libddcutil` as a section. Not even a log file. However, when i edited the log file to this:

    [global]
    options : --s1 ~/ddcutil.log --tid --ts --trcfunc ddca_open_display2 --trcfunc ddca_close_display --trcfunc ddc_open_display --trcfunc lock_distinct_display --trcfunc unlock_distinct_display --trcfunc ddc_close_display --trcfunc dummy_display_change_handler --trcfunc watch_displays_using_udev

I got a sign that the options where accepted this time, so i went with it.
After this i tried to run `ddcutil detect`, which gave me this response:

    Applying ddcutil options from ~/.config/ddcutil/ddcutilrc: --s1 ~/ddcutil.log --tid --ts --trcfunc ddca_open_display2 --trcfunc ddca_close_display --trcfunc ddc_open_display --trcfunc lock_distinct_display --trcfunc unlock_distinct_display --trcfunc ddc_close_display --trcfunc dummy_display_change_handler --trcfunc watch_displays_using_udev
    Traced function not found: ddca_open_display2

To make it run as expected, i needed to remove tracing of both `ddca_close_display` and `ddca_open_display2`.

So, i ran `ddcutil detect` afterwards, [and this is how the log looks](https://github.com/rockowitz/ddcutil/files/6474283/ddcutil.log)

Tried to enable screen dimming to provoke a crash, but didn't log anything else than a kde-related crash as mentioned in the original post. (however, i think it's weird that the stack trace of the original crash mentioned `ddca_open_display2`. Could that be related?)

rockowitz, 2021-05-13:
@OlavStornes  **ddca_open_display2()** and **ddca_close_display()** are API functions exposed by **libddcutil**.  They are not in the command line executable **ddcutil**.  So you do want to include them in the options string in the **[libddcutil]** section of the configuration file.  If they're included in either the **[gl**obal] or **[ddcutil]** section they'll be they'll be passed to ddcutil when it starts and rejected.

Assuming that your local version of **libddcutil.so** (which is installed somewhere under /usr/local) will be found before the system copy, it's just a matter of waiting for Plasma PowerDevil to load it.  At that point you should start seeing contents in the trace file - until then it will be empty.  When you get the failure in Plasma, note the time.  The call tree in the trace file at that time will hopefully show what happened.

I'm afraid I'm being vague about ensuring that your version of **libddcutil.so" is loaded.  This can vary from system to system.  It may be configured automatically for you when you **make install**, or you may have to call  **ldconfig -n &lt;name of directory where libddcutil was installed>**.  If you're relying on environment variable $LD_LIBRARY_PATH to find your copy, you need to specify it in the special configuration file read by KDE at startup.  See https://userbase.kde.org/Session_Environment_Variables .

OlavStornes, 2021-05-13:
Apologies, but this is starting to fall into confusing territory for me. Earlier i thought that putting it under global would be applied to everything, including libddcutil.
However as mentioned on the previous post. I tried to write the config as you said, with no sign of a log file being written, indicating that it didn't run at all with those settings.

When i compiled from source, i did use `make install` and `ldconfig -n /usr/local/lib` to be sure. (where that directory is where libddcutil is located for me)

    ll /usr/local/lib

    total 2,4M
    drwxr-xr-x 3 root root 4,0K mai   13 19:44 cmake
    lrwxrwxrwx 1 root root   19 mai   13 19:44 libddcutil.so -> libddcutil.so.4.0.0
    lrwxrwxrwx 1 root root   19 mai   13 19:44 libddcutil.so.4 -> libddcutil.so.4.0.0
    -rwxr-xr-x 1 root root 2,4M mai   13 19:44 libddcutil.so.4.0.0
    drwxr-xr-x 2 root root 4,0K mai   13 19:44 pkgconfig

I dont have `LD_LIBRARY_PATH`, so i assume that is not relevant for me?

Tried to provoke a crash this time as well, but no log files this time either, unfortunately.

rockowitz, 2021-05-14:
Re the config file sections, you're correct that the [global] section applies to both ddcutil and libddcutil.  The problem with applying it to ddcutil is that parsing will fail with a "Traced function not found" message for ddca_open_display2.   So you want it only in the [libddcutil] section. 

Since you've used "ldconfig -n" environment variable $LD_LIBRARY_PATH doesn't matter.  It's another way to add a directory to the library path, used mainly for testing.

At this point when you run PowerDevil, the test version of libddcutil should be loaded. Here's how to verify it. 

First, get the process number for PowerDevil: 
~~~
$ ps -ux | grep powerdevil
~~~
The first number on the "org_kde_powerdevil" line is the process number.  If there's no such line, kde has not yet started PowerDevil.

Then, use the following command to see the actual library used by powerdevil
~~~
$ grep libddcutil /ps/PROCESS_NUMBER/maps
~~~
If nothing is found, libddcutil has not yet been loaded.

When there's a crash, the libddcutil log file should be non-empty.

OlavStornes, 2021-05-18:
ps -ux | grep powerdevil
    olavsto+   92250  0.1  1.1 1039204 181884 ?      Sl   18:53   0:01 /usr/bin/systemsettings5 powerdevilprofilesconfig
    olavsto+   92920  0.0  0.2 558704 46244 ?        Sl   18:56   0:00 /usr/lib/org_kde_powerdevil
    olavsto+   94753  0.0  0.0   9276  2648 ?        S    19:03   0:00 grep --color=auto powerdevil

In this case, i tried to run:

    grep libddcutil /ps/92920/maps

However, when just attempting to run libddcutil, i get notified that the command is not found.

    [~]$ libddcutil
    bash: libddcutil: command not found

rockowitz, 2021-05-18:
The output of the grep command indicates that libddcutil has not been loaded by  powerdevil.  

Re "command not found", recall that libddcutil is a shared library.  It is not an executable command.

closed 2021-08-17 by rockowitz