rockowitz/ddcutil #206: DDCutil Only detecting one out of three monitors

cpebble, 2021-05-03:
I've previously used this to control brightness on my screens, where it correctly detected two monitors. A few months go by and i returned to a pet project using ddclib, and suddenly it only detects one out of three(i added one in the meantime). The two monitors are the same, the third is a new dell monitor.

I ran `ddcutil interrogate` Here: [pastebin](https://pastebin.com/4xengUfL).

The monitors all have ddc/ci enabled, and are connected via two hdmi ports, and a single displayport cable, without daisy chaining.
Any advice on how to fix this?

rockowitz, 2021-05-04:
Looks like you've hit the bug in the Nvidia 465 driver that is discussed in issue #201.   The problem is reported for both 465.24 and 465.27.  I suggest you back out to the 460 driver.

toshi2k2, 2021-08-05:
Reporting the same problem for 470.57 driver version as well. Only one monitor being detected out of two.

gjvc, 2022-03-10:
This seems to be also present using  470.103.01 too.  Two out of three monitors detected.

rockowitz, 2022-03-10:
I would also mention that the original post by @cpebble reports simultaneous use of what appears to be another DDC program ddclib (I am unfamiliar with it).   In this case his problem may be due to conflict between **dddcutil** and the other program.  These conflicts can often be resolved using **ddcutil** option ***--force-slave-address***. Release 1.2.2 does a better job of diagnosing the conflicts, reporting monitors as Busy instead of Invalid.  For details, see the [Release Notes](http://www.ddcutil.com/release_notes/) for **ddcutil** 1.2.2.