rockowitz/ddcutil #207: DDC Communication Failed - Dell Monitor

jeff-stacey, 2021-05-17:
I'm having trouble getting ddcutil to control my Dell P2212H monitor from my Lenovo T440s laptop. I've tried connecting it first over VGA and then a mini-DP to DVI adapter cable but neither works. In both cases `ddcutil detect` appears to be able to read the monitor's model information but says that DDC communication failed. DDC/CI is enabled in the monitor's onscreen menu. I would appreciate any help with getting this to work. The output from `ddcutil interrogate` [is here](https://github.com/rockowitz/ddcutil/files/6497095/interrogate.txt).

rockowitz, 2021-05-18:
I don't have an answer for you, but I do have some thoughts. 

Looking at lines 575-582 of the interrogate report, I see that the monitor is responding to a request for the value of features x10 (brightness) with gibberish.  It doesn't even look like a corrupted reply packet.  Interestingly, it's the same value over time.

My first thought was a poor DDC implementation on the monitor.  I have a similar Dell P2411h, and it's the monitor I plug in when I want to test how ddcutil handles failures.  It often returns corrupted packets.  I've tested other P2411h's and they show the same behavior.  But what comes back looks like a corrupted reply packet (e.g. a double byte), and is different each time. Since the same sequence comes back each time, that's probably not the case here, but ti's worth trying option **--sleep-multiplier 2.0** to see if slowing things down resolves the problem.

Have you tried using a monitor with an actual DisplayPort input?  The DP-DVI conversion might be the problem.  If a monitor with a real DP input works it may be that you you need an active rather than passive converter.  (For a detailed discussion, see [DDC Over DisplayPort](http://www.ddcutil.com/displayport/#ddc-over-displayport.)

And of course, does the P2411h work successfully on other compaters?

jeff-stacey, 2021-05-19:
Correct me if I'm wrong but I don't think the device being read from in lines 575-582 is the monitor in question. The "Extra Descriptor" field matches the model number for the laptop's built-in display. The Dell P2212H's information is in the EDID for  `/dev/i2c-3`. That device is returning a hard error (`ENXIO(-6)`) from the read you mentioned. 

I noticed that later in the log xrandr says that the P2212H is connected over HDMI - maybe this is because I'm using a passive DP/DVI adapter.

I don't have any other monitors or computers to try right now but I will try to get my hands on one soon.

rockowitz, 2021-05-19:
You're absolutely right.  The I2C device in question is /dev/i2c-3.  And ENXIO is a hard error.

It appears that you're using a passive adapter. (Active adapters are labelled as such.) You could try an active adapter. It will cost in the $10-$20 range.

I don't attach any significance to the fact that the monitor is reported as HDMI instead of DVI.  For our purposes they're the same, just with different pins.  Both use TDMS encoding for video, and both use the SCL/SCA lines for I2C communication.