rockowitz/ddcutil #208: Pi4 + Vizio D32f-G4 TV = Invalid Display, DDC Communication failed

Geektoolkit, 2021-05-31:
Hi, I got excited when I saw news that the pi4 could now support this, and am hoping to integrate it into my Dynaframe project to allow for dynamic brightness changes when showing pictures (digital picture frame)

I have this hooked over HDMI to a 2019 Vizio D23F-G4 tv (1080p) and was hoping it'd work, but I'm getting odd results. I looked through as many support issues as I could find, and I'm not clear on if my TV just doesn't support DDC, if I'm doing something wrong, etc.  I updated the pi to 5.10, enabled the KMS driver, etc. from this thread; 
https://github.com/raspberrypi/linux/issues/3152#

Below are verbose outputs in hopes that it helps:

pi@leftframe:~ $ ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false

Failure getting EDID for /dev/i2c-1: status code=DDCRC_INVALID_EDID(-3010): unable to parse EDID
Open failed for /dev/usb/hiddev0: errno=EACCES(13): Permission denied
  USB bus 1, device 3, vid:pid: 046d:c52b - Logitech:USB Receiver
Invalid display
   I2C bus:             /dev/i2c-12
      I2C address 0x30 (EDID block#)  present: false
      I2C address 0x37 (DDC)          present: false
      I2C address 0x50 (EDID)         present: true 
      /sys/bus/i2c/devices/i2c-12/name: Broadcom STB : 
   EDID synopsis:
      Mfg id:           VIZ
      Model:            D32f-G4
      Serial number:    Unspecified
      Manufacture year: 2019
      EDID version:     1.3
      Product code:     0
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.261, 0.271
      Red   x,y:        0.644, 0.322
      Green x,y:        0.263, 0.614
      Blue  x,y:        0.150, 0.061
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 59 3a 00 00 01 01 01 01   ........Y:......
      +0010   00 1d 01 03 80 45 27 78 0a e5 ad a4 52 43 9d 26   .....E'x....RC.&
      +0020   0f 42 45 a5 ce 00 d1 c0 01 01 01 01 01 01 01 01   .BE.............
      +0030   01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 3f 43 21 00 00 1a 7f 21 56 aa 51 00 1e 30   E.?C!....!V.Q..0
      +0050   46 8f 33 00 3f 43 21 00 00 1e 00 00 00 fd 00 32   F.3.?C!........2
      +0060   4c 1e 50 10 00 0a 20 20 20 20 20 20 00 00 00 fc   L.P...      ....
      +0070   00 44 33 32 66 2d 47 34 0a 20 20 20 20 20 01 11   .D32f-G4.     ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

For: ddcutil environment --verbose >> out.txt I'm attaching
[out.txt](https://github.com/rockowitz/ddcutil/files/6568189/out.txt)
 the file (it's a bit large)
 
 I'll keep searching through issues to see if this is a dupe, but if not maybe I found something new :)

rockowitz, 2021-06-02:
I have not tested ddcutil with a Pi 4, but your problem is pretty clearly that the TV does not support DDC.  It is possible to read the EDID (slave address x50) on the I2C bus, but DDC (slave address x37) is not supported.  This is verified by the output of i2cdetect. To confirm, I'd suggest connecting a monitor known to support DDC to the Pi, and/or connecting the TV to a desktop system with almost any recent video card.

CEC is a different on-the-wire protocol.  It is carried on pin 13 of the HDMI connector, while I2C is on pins 15/16.  (See [Wikipedia](https://en.wikipedia.org/wiki/HDMI)).  I am not familiar with CEC, but it may well address what you're trying to do.  The standard user space interface is provided by [libcec](https://github.com/Pulse-Eight/libcec).

Geektoolkit, 2021-06-02:
Thankyou so much! I had a feeling that's what was going on, but just wanted to confirm. I'll try the pi4 on some other monitors and see if that works. I'm looking to dim them for my Dynaframe project (digital art frames) so if they don't work I'll have to find some that do so I can test things out. I really appreciate the fast turnaround and that you've developed this software!

Geektoolkit, 2021-06-02:
Closing issue, will reopen a new one if I find an issue on the pi4 with a valid monitor.

closed 2021-06-02 by Geektoolkit