rockowitz/ddcutil #209: Quick way to get EDID's

papuSpartan, 2021-06-19:
Is there a quicker way to get the EDID's of all connected monitors other than 
```
ddcutil --async detect --v
```
and then having to pull it from the output?

rockowitz, 2021-06-19:
If by "connected" you mean DDC capable, no. The only way to determine if 
a display with an EDID supports DDC is to try, and that incurs the 
delays that are part of the DDC protocol.

If you just want to get the EDIDs of all monitors, whether or not they 
support DDC, xrandr will do that quickly.

papuSpartan, 2021-06-21:
@rockowitz Yea no i just wanted to know how to get the EDID's faster. Suprised i didn't think of xrandr doing this. That should be just what i need considering monitor selection using EDID seems to be faster.

Thanks

closed 2021-06-21 by papuSpartan

rockowitz, 2021-06-21:
When you write "monitor selection using EDID seems to be faster", if 
you're referring to monitor selection in ddcutil that's not the case.  
Monitor probing is the same in (almost) all situations.  The one 
exception is the "--bus" option.  If you know a priori the I2C bus your 
desired monitor is on, ddcutil does not examine any other /dev/i2c device.