rockowitz/ddcutil #212: Obtaining Display Data via DDC

Xanth0k1d, 2021-07-14:
Long story short, I got a broken laptop with a working edp display, so I bought a driver board for it (RTD2556).
Problem is only some GPU/HDMI port works i.e. Backlight is on but nothing shows if connected to a GTX760 unless I manually change the refresh rate to 59Hz, raspberry pi zero wouldn't even detect the display. (So far it seems to be working with integrated intel graphics on a few laptops)

I believe that the trick here is to feed the computer the correct data via DD/CCI, however, I'm not really sure how much information do I need to feed to the computer( Will feeding the resolution and refresh rate be enough?)

Notably, there's an option to switch DD/CCI off on the driverboard, changing it made no difference though.

I know that this might be off the topic, but I suppose this repo is the only place where someone knows about the protocal could shed some lights on this matter. (My eventual plan is to use an arduino to isolate and feed the display data)

Any help would be appreciated (Do close this if considered irrelevant)

rockowitz, 2021-07-16:
Presumably you've removed the panel from the display and connected it to the RTD2556.  I assume that it's a recent enough laptop that the display is eDP, not the older LVDS.  Embedded display port does not support DDC/CI.  So the question is what does it mean that there is an option to switch DDC/CI on or off on the driverboard?  Does it mean that the board passes DDC/CI I2C communication on to the display (meaningless for eDP), or does it process the DDC/CI commands itself in some way, possibly altering the video output where appropriate (e.g. red gain)?

As for resolution and refresh rate, DDC/CI has no role, other than possibly reporting current values.  The display's EDID block specifies supported resolutions and refresh rates.  (See the wikipedia page.) The video driver uses this information to create a proper video signal for the display.  So the question becomes whether the driver board returns the actual EDID from the eDP panel?

You can use **xrandr --verbose** to see the raw EDID and interpreted values.

I hope this background is helpful.

Xanth0k1d, 2021-07-18:
Yep, that's how I got to this repo actually.
The display is a 30 pin edp, according to the seller of the driver board, there's this universal firmware that supports all 1920x1080 30-pin edp displayd regardless of their size. 
I got saw the enable/disable DDC/CI option while I was using the menu buttons on the driver board to browse through settings to troubleshoot my problems.

I will try with xrandr, it should return the raw EDID of all displayes connected I guess?

Thanks for the reply!

Xanth0k1d, 2021-07-18:
Yep, it's the EDID
The default one is 60hz, but only after I used vnc to change it to 59.96 before the display started to show.
Really appreciate the xrandr trick. I'm now looking for a hardware solution for this.

closed 2021-08-11 by rockowitz