rockowitz/ddcutil #215: Contributing to docs

Thomi7, 2021-08-03:
I wanna change [this page](https://www.ddcutil.com/command_detect/) in the docs. Where do I find the sources for the docs so I can set up a PR?

rockowitz, 2021-08-03:
The document source is maintained offline, and the generated html pages 
are uploaded.  Please send the changes you would like to see.  I will 
review and incorporate them as appropriate.  If the changes are 
extensive, submit them as an attachment. Thank you.

Thomi7, 2021-08-03:
In command_detect:

Currently at the end the page says:
Option: --verbose
Show additional information, including the EDID.

When I run (version 1.1.0) `ddcutil detect --verbose` I don't get the EDID information. Only if it's there. For querying that you need to use --very-verbose. You could just change the --verbose to --very-verbose and mention that there's also a verbose mode with less options.

rockowitz, 2021-08-11:
Thank you for pointing out the discrepancy between the documentation for command **detect** and its current behaviour.  The current development branch, 1.2.0- dev, has been modified to be consistent with the behaviour.  The online documentation will be updated once version 1.2.0 is released.

***--very-verbose*** is an undocumented option whose output can vary over time as needed for remote problem diagnosis.  Help output in 1.2.0-dev has been modified to not report it.

fabian-thomas, 2022-08-19:
Closing this because in the newest version the diff between verbose and very-verbose is zero. So the documentation is fine.

closed 2022-08-19 by fabian-thomas