rockowitz/ddcutil #218: no displays found on Arch, but works OOTB on Slackware and Void

aliasless, 2021-08-17:
Hi, I have used ddcutil on both Void and Slackware using proprietary nvidia drivers. Other than having to load i2c-dev, ddcutil would detect my monitor OOTB. However, on Arch, with no difference in steps I took to set it up, ddcutil is unable to detect my monitor. I am also certain that no configuration has changed on the monitor itself, as it was working prior to switching to Arch, and it started working again after switching the OS back.

[ddcutil interrogation output](https://github.com/rockowitz/ddcutil/files/6996735/interrogation.txt)

Any guidance would be appreciated. Thanks.

rockowitz, 2021-08-17:
Are you using the same version of the Nvidia driver on Arch as on Void and Slackware?  There have been problems with recent Nvidia drivers, at least on Arch. See Issue #201.

You do not appear to be using the special Nvidia drivers settings.  See [Special Nvidia Driver Settings](https://www.ddcutil.com/nvidia)

aliasless, 2021-08-17:
Installing the special Nvidia settings has fixed the issue. I do not know what version I was using on Void, but on Slackware I was probably using 460.67, and on Arch I am currently using 470.63. It sounds like it's likely that I hadn't needed to install the special settings before because of using an older version. Thanks.

closed 2021-08-20 by rockowitz