rockowitz/ddcutil #220: Release 1.2.0-rc2

rockowitz, 2021-08-31:
Release candidate 1.2.0-rc2 is now available as branch 1.2.0-rc2.

Changes since 1.2.0-rc1: 

- (libddcutil) Ensure that /sys/class/drm/cardN devices exist before watching for displays to be connected or disconnected.  Without this check, the loop watching for display hotplug changes would repeatedly emit "No drm class cards found in /sys/class/drm" messages.
- Implement retrieval and free of single DDCA_Display_Info structs.  API functions affected: ddca_get_display_info(), ddca_free_display_info(), ddca_get_display_refs()
- Fix glib assert failure in ddca_close_all_displays()
- Turn off various debug messages remaining in the code.

For a full description of what has changed since release 1.1.0, see file CHANGELOG.md and also the preliminary release notes on www.ddcutil.com.

kloczek, 2021-09-13:
Please make not `1.2.0-rc2` but `1.2.rc2` (with dot instead dash).
That kind of versioning wioll make ddcutil packaking friendly bacause `1.2.rc2` is smaller as bersion string than `1.2.0` and that would allow regular package `1.2.rc2` and upgrade from `1.2.rc2` to final `1.2.0`
Similar versioning pattern is now used in Gnome https://discourse.gnome.org/t/new-gnome-versioning-scheme/4235

rockowitz, 2021-09-16:
Please explain the situation from which your request arises. Does the current scheme pose a problem to you as a developer, a packager, or an end-user? Release candidates are intended for installation from source by individual users, not for inclusion in distributions.

"1.2.rc2" is smaller as a version string than "1.2.0" only if special rules are applied.   "0" sorts before "r", not after.  For proper sorting, you need to split the version string into its components, convert each to an integer, and perform an integer sort.  As a string, "1.11.0" would sort before "1.2.0", not after.  The change you suggest would require that for proper sorting the 3rd component must first be checked if it is an integer, with different rules for comparing integers to integers, non-integers to non-integers, and integers to non-integers.

Specifying a beta, release candidate, etc. using a suffix is a commonly used scheme.  The gnome scheme you link to explicitly says that is intended for the gnome "meta-package", not for individual packages.  Can you cite other situations in which the gnome scheme is used?

Note also that this change would force a non-upwardly-compatible change to the libddcutil API.

In summary, the change you request has significant costs, both to ddcutil, to programs that use it, and to code that handles version version identifiers.    To make it requires an especially compelling use case.

kloczek, 2021-09-16:
`1.2.0-rc2` contains dash which is character forbiden to use by any packabing software (rpm,  dpkg, Solaris IPS etc.)
In other words `1.2.0-rc2` cannot be used to try build package when new tag is added to the git repo and build will fail (because wrong version string).
Digits are at the end of the alphabet so versions strings like `1.2.a1`, `1.2.aphna`, `1.2.b1`, `1.2.beta2` `1.2.rc1`, `1.2.rc2` and at the end `1.2.0` are ascending. Buy this it will be no problems with upgrade between those versions ad at the same time version string contains clear indication that some versions are not prod.
Everything is exaplained in gnome versioning scheme which is packaging friendly.

rockowitz, 2021-09-24:
I have returned from vacation and am attending to non-urgent **ddcutil** issues.

There are many versioning schemes. [Semantic Versioning](https://semver.org/) is a common one, and **ddcutil** adheres to it.  In fact, the new gnome versioning scheme you cite has this to say: 

> Q: What about semantic versioning/semver?
> A: Individual components of the GNOME software development platform—like GLib or GTK—typically follow [semver](https://semver.org) rules. 

Changing to the gnome scheme, where the patch field may not be an integer, would entail a non-upwardly compatible change to the **ddcutil** API (function ***ddca_ddcutil_version()***).   Such a change requires a compelling actual use case.

As I've noted, I have no expectation that release candidates will be packaged, let alone distributed.  If it is happening, I need to understand how and why.  I maintain the debian package, and only formal releases, not release candidates, are distributed. Likewise, my published builds on OBS, COPR, and launchpad are based on formal releases, not release candidates.

The change you request will not be made in release 1.2.0.  If you still feel it is important, please open a separate issue as an enhancement request.  Please explain why the change is important in practice, not simply from general principles. That a significant Linux distribution requires the change because it has reason to distribute release candidates would probably make the strongest argument.  If you are using automated software to build ddcutil automatically, please describe.

kloczek, 2021-09-24:
Please .. I'm not talking about "many versioning schemes".
I've mention about very specyfic context with versioning sheme which is _packaging friendly_
If you are not interested to interract in some civilised interraction with packagers like me that is fine but please do not spread that across "many versioning schemes" because (again) I've mention very specyfic context.

What is produced on layer with projests source trees are ultimately consumend by packaging layer. Above that is layer with OS distributions.

All about what I'm asking is just adjust a bit versioning schemes to be packaging friendly. Only this .. and nothing more.
This is not rocket science.

closed 2021-11-24 by rockowitz