rockowitz/ddcutil #221: Samsung C24FG70 only detected on DP and not HDMI

cybercatgurrl, 2021-09-03:
I've attached the interrogation. If I'm right the HDMI is i2c-6 in the hdmi log. It's i2c-9 in the displayport log. There's some sort of response on i2c-6 but i'm guessing it's not what it's expecting. I recently switched from Windows and I used to be able to control the brightness there so I know it supports it. Let me know if there's anything else I can do to help get this fixed.
[ddcutil_hdmi.log](https://github.com/rockowitz/ddcutil/files/7103879/ddcutil_hdmi.log)
[ddcutil_displayport.log](https://github.com/rockowitz/ddcutil/files/7103880/ddcutil_displayport.log)

rockowitz, 2021-09-03:
You do not appear to be using the [special settings](https://www.ddcutil.com/nvidia) often needed by the proprietary Nvidia driver. 

Also, there have been problems with recent Nvidia drivers.  (See #201)  If the special settings dont' work, try dropping back to  an earlier dirver (e.g. 460 series).

cybercatgurrl, 2021-09-04:
Thanks, I’ll give it a go later. The output from ddcutil environment made it seem like I didn’t need to use based on what it said so I didn’t bother.

spacecowgirl314, 2021-09-04:
I gave both the settings in xorg.conf and modprobe.d folder a try and it's not showing up in /proc/driver/nvidia/params. Not sure what else to do.

closed 2025-02-11 by rockowitz