rockowitz/ddcutil #224: DDC communication failed on Radeon RX 5500 XT

sierratango88, 2021-09-17:
Hello,

Thank you for this great program.

I have a problem with getting `ddcutil` to work with my Radeon RX 5500 XT on my LG W2453 (HDMI) and HP w22 (DisplayPort) monitors. It will always have a DDC communication failure when trying to run `ddcutil detect` and so the monitors cannot be controlled.

Previously, I used a GeForce GTX 1050 Ti with these same monitors and `ddcutil` could detect them and control them quite well.

Other monitor control programs on Windows using DDC/CI also cannot detect the Radeon RX 5500 XT.

Is there any way to get this card to work with `ddcutil` or must there be some changes required to the firmware and/or driver?

Thanks again.

[ddcutil.log](https://github.com/rockowitz/ddcutil/files/7188198/ddcutil.log)

rockowitz, 2021-09-20:
The Radeon 5xxx cards do seem to have a lot of I2C problems.  It is interesting that you report the problem with Windows programs as well (which programs, what OS and hardware environment?).  All I can suggest at this point is looking in the logs and dmesg output for amdgpu errors.  I am currently on vacation and will give this problem more thought when I return and have more resources available.

sierratango88, 2021-09-21:
The programs I previously tried on Windows 10 were [ControlMyMonitor](https://www.nirsoft.net/utils/control_my_monitor.html), [Monitorian](https://github.com/emoacht/Monitorian) and ClickMonitorDDC (now abandoned). This was using the same Radeon RX 5500 XT and hardware indicated in [ddcutil.log](https://github.com/rockowitz/ddcutil/files/7188198/ddcutil.log).

Please enjoy your vacation. This issue is low priority and can wait.

Thanks.