rockowitz/ddcutil #225: Please move ~/.cacheddutil folder to ~/.cached/ddcutil

gadefox, 2021-09-18:
Thanks.. Or maybe `/` is missing between `cache` and `ddutil` in the source code??

rockowitz, 2021-09-19:
I am on vacation and do not have a test configuration that uses caching.  Can you describe in more detail what you're seeing? Is $XDG_CACHE_HOME explicitly set in your environment? Output of a script that demonstrates the problem would be helpful. 

I assume that when you write "ddutil" it's just a typo for "ddcutil".  The string "ddutil" does not exist in the source code.

gadefox, 2021-09-19:
Yes, it's a typo.. And I suggest you enjoy the vacation and we'll solve this when you're back - this is really a minor issue. Anyway, I'm using `ddcutil 1.1.1-dev` version and `$XDG_CACHE_HOME` is set to `~/.cache` (`/home/fox/.cache`). I don't use any script, but the `ddcui` app instead. Please note the folder is created sometimes, so I can't replicate this. I really believe, you have a typo in the code, because the diff between `~/.cacheddcutil` and `~/.cache/ddcutil` s one letter only.

rockowitz, 2021-09-22:
The name of the cache directory was not properly constructed when environment variable $XDG_CACHE_HOME was set but not terminated with "/".  This has been fixed with the latest updates to branch 1.2.0-dev.  Similar changes apply to all the $XDG_xxx_HOME directory names.

To aid in remote diagnosis, command **ddcutil environment --verbose** now reports the directories of the XDG desktop spec.  It also reports the contents of the configuration file ddcutilrc, cache file capabilities, and the user defined feature files (.mccs).  

Please check that your bug is fixed and that the **environment** output looks reasonable. Thank you.

closed 2021-10-04 by gadefox