rockowitz/ddcutil #226: can only switch input to DP1 not DP2

philipp1992, 2021-09-24:
Hi Ggys,

i have a samsung c49rg94ssu monitor with two DisplayPorts.
When using ddcutil i can successfully switch to DP1 with setvcp 60 9 and also to HDMI. However i can not switch to DP2 neither from the current source (DP1) nor from the DP2 source. I have tried all values from 0 to 30.

Switching from DP2 to DP1 works from both sources.

Any idea how i can further debug this issue? I have updated the monitor to the latest firmware and also did a factory reset.

kind regards
Philipp

rockowitz, 2021-09-24:
Bear in mind that on some monitors you can switch the display from the active input to the input on which ddcutil is executed, on others you cannot.  For example, suppose you have a Windows system on DP-1 and a Linux system on DP-2, and the DP-1 is currently active. Then a **ddcutil setvcp x60** command issued from DP-2 might or might not be recognized.  It depends on the monitor.

It's interesting that value x09 represents DP-1, which differs from the value in the MCCS spec (x0f).  Non-standard values for feature x60 seems to be common on Samsung monitors.  The values for my U32H750 are: 

Value | Connector
--------|--------------
x05   | HDMI-1
x06   | HDMI-2
X0f   | DisplayPort

If your Linux system is connected to DP-2 and that input is active, **ddcutil getvcp  60** should show you the value for DP-2.

Finally, picture-by-picture was never imagined when the MCCS spec was defined, so manufacturers are "extending" the implementation of feature x60 to accommodate.

pwhelan, 2023-08-20:
I also have the same problem with my C49RG9. I am able to switch to DisplayPort-1 from DisplayPort-2 but not to DisplayPort-2 from DisplayPort-1. I have actually brute forced every single two byte value in an attempt to do so.