rockowitz/ddcutil #233: ddcutil C api: set i2c_force_slave_addr_flag

FedeDP, 2021-11-06:
Hi!

As per user bug report: https://github.com/FedeDP/Clightd/issues/82, i was wondering whether there's a way to set i2c_force_slave_addr_flag to true; a simple workaround is to declare it as extern, and then set its value.  
But then i need to compile my software as Position Independed Executable (well nothing against that), and it feels like a subtle workaround.
Is there any better solution?

Thank you!

rockowitz, 2021-11-08:
I've added functions ddca_enable_force_slave_address() and ddca_is_force_slave_address_enabled() to the API.

FedeDP, 2021-11-08:
Hi! Thank you very much!

closed 2021-11-08 by FedeDP

rockowitz, 2021-11-09:
A couple things I should have noted: 

The ddcutil configuration file allows for passing ddcutil options like **--force-slave-address** to libddcutil as well.  See [Conifguration File](http://www.ddcutil.com/config_file/).

The **--force-slave-address** option allows ddcutil to seize control from another application using the I2C bus.  As such, it could clobber the other application, which is why it's an option.

FedeDP, 2021-11-09:
Wow that's great! I looked for something like that, but user defined feature captured my attention; later i realized udf was not what i was looking for, and stopped looking around.
Well, that said, i think i will tell the Clightd user who reported the bug, to use a configuration file; it seems a much better and suitable solution all in all.
Thank you very much!

rockowitz, 2021-11-09:
Given that **--force-slave-address** can be specified in the configuration file, do you still think there's value in having the API functions?

FedeDP, 2021-11-10:
I think you can easily drop that now.
Sorry for the confusion :(

rockowitz, 2021-11-10:
Not a problem. :)  I should have suggested the config file first.  If your code were actually making use the functions I'd be fine with leaving them in, but I prefer to not add functions until they're clearly necessary.

FedeDP, 2021-11-10:
I agree :D
Thanks again, appreciated that!