rockowitz/ddcutil #234: COMPILATION ERROR

tathastu871, 2021-11-16:
https://github.com/termux/termux-packages/issues/7936#issuecomment-970182425

rockowitz, 2021-11-17:
The latest 1.2.1-dev branch addresses some of the issues in your thread termux/termux-packages/issues/7936

1) File i2c_util.c: The warning re the comparison test is indeed a bug and has been fixed. However, it would only have been hit if an i2c device where named "/dev/i2c-". 

2) File data_structures.c: Formatting changed. 

3) Files simple_ini_file.c, persistent_capabilities.c: Added "#include <strings.h>" to provide definition for index()

4) -Wcompound-token-split-by-macro] warnings.  This warning occurs only in recent releases of clang.  The macros work.  I suggest you turn off this warning.

5) ld.lld error undefined symbol: __assert_fail. This is a Linux specific function used, in Linux, by the C header file macros assert() etc. Trying to use assert() within macro TRACED_ASSERT() etc. does not work.  If you can't find an equivalent function in termux, I suggest you use abort() to unconditionally kill the application.

6) ld.lld error: undefined symbol: index.  I expect this will have been resolved by the inclusion of header file <strings.h>

7) ld.lld: error: undefined symbol wordexp. You will need to provide a replacement for this function or revise the parsing code that uses it.   The thread indicates such a replacement exists.

I suggest you execute configure with the following options: 

- --enable-usb=no  Monitors that use USB for MCCS communication are rare. Note that if --enable-udev=no is specified, it implies --enable-usb=no. 
- --enable-envcmds=no. This removes the **environment** and **usbenvironment** commands, which are used for remote problem diagnosis and promiscuously poke around.  This implies --enable-drm=no and --enable-x11=no, which further reduces the code that may have to be modified.

Lastly, and most critically, do your video drivers expose /dev/i2c devices for external monitors?  Slave address x37 (DDC) must be active, not just slave address x50 (EDID).

tathastu871, 2021-11-17:
still getting 
ld.lld error: 
on dev branch

tathastu871, 2021-11-20:
```
  CC       error_info.lo
  CC       libmain/api_error_info_internal.lo
ld.lld: error: undefined symbol: TRACED_ASSERT
ld.lld: error: undefined symbol: TRACED_ASSERT_IFF
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
```

rockowitz, 2021-11-21:
Please submit the entire build output as an attachment.

tathastu871, 2021-11-21:
```
make  all-recursive
make[1]: Entering directory '/data/data/com.termux/files/home/ddcutil'
Making all in src
make[2]: Entering directory '/data/data/com.termux/files/home/ddcutil/src'
Making all in util
make[3]: Entering directory '/data/data/com.termux/files/home/ddcutil/src/util'
  CC       data_structures.lo
  CC       ddcutil_config_file.lo
  CC       debug_util.lo
  CC       edid.lo
  CC       file_util.lo
  CC       file_util_base.lo
  CC       glib_util.lo
  CC       error_info.lo
  CC       glib_string_util.lo
  CC       i2c_util.lo
  CC       linux_util.lo
  CC       multi_level_map.lo
  CC       output_sink.lo
  CC       report_util.lo
  CC       simple_ini_file.lo
  CC       string_util.lo
  CC       subprocess_util.lo
  CC       sysfs_util.lo
  CC       timestamp.lo
  CC       utilrpt.lo
  CC       xdg_util.lo
  CC       sysfs_i2c_util.lo
  CC       device_id_util.lo
  CC       x11_util.lo
  CCLD     libutilaux.la
  CCLD     libutil.la
make[3]: Leaving directory '/data/data/com.termux/files/home/ddcutil/src/util'
Making all in usb_util
make[3]: Entering directory '/data/data/com.termux/files/home/ddcutil/src/usb_util'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/data/data/com.termux/files/home/ddcutil/src/usb_util'
Making all in base
make[3]: Entering directory '/data/data/com.termux/files/home/ddcutil/src/base'
  CC       base_init.lo
  CC       core.lo
  CC       ddc_errno.lo
  CC       build_info.lo
  CC       core_per_thread_settings.lo
  CC       ddc_packets.lo
  CC       dynamic_features.lo
  CC       dynamic_sleep.lo
  CC       displays.lo
  CC       execution_stats.lo
  CC       feature_lists.lo
  CC       feature_metadata.lo
  CC       feature_sets.lo
core.c:834:24: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
                       ap);
                       ^~
core.c:835:47: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
     printf("          &ap=%p, ap=%p\n", &ap, ap);
                                  ~~          ^~
core.c:961:62: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
         printf("(%s) &args=%p, args=%p\n", __func__, &args, args);
                                     ~~                      ^~~~
  CC       last_io_event.lo
core.c:1000:62: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
         printf("(%s) &args=%p, args=%p\n", __func__, &args, args);
                                     ~~                      ^~~~
core.c:1037:62: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
         printf("(%s) &args=%p, args=%p\n", __func__, &args, args);
                                     ~~                      ^~~~
core.c:1078:62: warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
[... 709 lines cut ...]
2 warnings generated.
  CCLD     libddcutil.la
  CCLD     ddcutil
ld.lld: error: undefined symbol: TRACED_ASSERT
>>> referenced by ddc_displays.c
>>>               ddc_displays.o:(ddc_initial_checks_by_dh) in archive ./.libs/libcommon.a
>>> referenced by ddc_displays.c
>>>               ddc_displays.o:(ddc_initial_checks_by_dh) in archive ./.libs/libcommon.a
>>> referenced by ddc_displays.c
>>>               ddc_displays.o:(ddc_initial_checks_by_dh) in archive ./.libs/libcommon.a
>>> referenced 45 more times

ld.lld: error: undefined symbol: TRACED_ASSERT_IFF
>>> referenced by ddc_packet_io.c
>>>               ddc_packet_io.o:(ddc_open_display) in archive ./.libs/libcommon.a
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [Makefile:881: ddcutil] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/data/data/com.termux/files/home/ddcutil/src'
make[2]: *** [Makefile:978: all-recursive] Error 1
make[2]: Leaving directory '/data/data/com.termux/files/home/ddcutil/src'
make[1]: *** [Makefile:549: all-recursive] Error 1
make[1]: Leaving directory '/data/data/com.termux/files/home/ddcutil'
make: *** [Makefile:456: all] Error 2
```

rockowitz, 2021-11-22:
First, I requested that you "Please submit the entire build output as an attachment."  Is that so difficult? There's a field at the bottom the comment box with the text "Attach files by dragging & dropping, selecting or pasting them."

Second, you need to look at what the build output is saying.  Long before your initial report of simply
~~~
CC       error_info.lo
  CC       libmain/api_error_info_internal.lo
ld.lld: error: undefined symbol: TRACED_ASSERT
ld.lld: error: undefined symbol: TRACED_ASSERT_IFF
clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
~~~

There are numerous warnings like: 
~~~
ddc_packet_io.c:298:16: warning: implicit declaration of function 'TRACED_ASSERT' is invalid in C99 [-Wimplicit-function-declaration]
               TRACED_ASSERT(rc < 0);
               ^
~~~

This should have suggested that the problem is in file base/core.h that defines TRACED_ASSERT() and TRACED_ASSERT_IFF().  Recall that you were not seeing this error in earlier builds. Have you modified this file?  That would be the first place I would look.

Re the other warnings: 
- warning: format specifies type 'void *' but the argument has type 'va_list' (aka '__builtin_va_list') [-Wformat]
  The source code has been modified to not cause this warning.
- warning: too many arguments in call to 'get_sysfs_drm_displays'
   GPtrArray * prev_displays = get_sysfs_drm_displays(wdd->drm_card_numbers, false);
  The source code has been modified to not cause this warning.
-  [-Wcompound-token-split-by-macro]
  This warning exists only in recent clang versions. Apparently your environment enables it. I have tried to 
  turn it off, but since I can't trigger the warning on my system the change may or may not work on yours.

tathastu871, 2021-11-22:
Build log:
[log_1.2.1-dev.txt](https://github.com/rockowitz/ddcutil/files/7577793/log_1.2.1-dev.txt)
Previously on this branch added abort() solved __assert_fail related problem but started TRACED_ASSERT_IFF related errors. Rebuilded on latest and build log attached

rockowitz, 2021-11-23:
If you haven't solved the __assert_fail problem please submit your copy of core.h.

tathastu871, 2021-11-24:
> If you haven't solved the __assert_fail problem please submit your copy of core.h.

[Uploading core.h.txt…]()

rockowitz, 2021-11-24:
Well duh.  Apparently your changes to use abort() instead of __assert_fail() were incorrect, causing the TRACED_ASSERT() and TRACED_ASSERT_IFF() macros to fail.  That's where you need to look.

tathastu871, 2021-11-24:
> Well duh. Apparently your changes to use abort() instead of __assert_fail() were incorrect, causing the TRACED_ASSERT() and TRACED_ASSERT_IFF() macros to fail. That's where you need to look.

But above core.h was original corrections were already reducted

rockowitz, 2021-11-25:
I suggest you stop trying to port ddcutil to Termux.  Doing this kind of port requires a knowledgeable C programmer, and as I've seen you state more than once, you're not a programmer. You're in over your head.

Furthermore, I remain sceptical that even if you were to successfully build ddcutil it would do anything useful.  There needs to be a physical I2C communication channel as part of the connection between the Android device and the monitor.  Just because a video signal is transmitted does not guarantee that the I2C signals are as well.  The physical GPU in the Android device has to implement I2C communication, and the video driver needs to expose that communication as /dev/i2c devices. I've seen no evidence that any of these requirements are satisfied.

closed 2021-11-25 by tathastu871