rockowitz/ddcutil #235: Release 1.2.1-rc1

rockowitz, 2021-11-24:
Release candidate 1.2.1-rc1 is now available as branch 1.2.1-rc1.

Significant changes since release 1.2.0: 

- Option ***--settings***.  Until now, details of the current configuration and option settings have been shown at the start of any command when ***--verbose*** is specified. This information has become increasingly large distracting.  Instead, this information is now shown only when option ***--settings*** is specified.
- Option ***--syslog***.  This option sends trace and debug messages to the system log in addition to the trace location.
- Option ***--wall-timestamp*** (or ***--wts***) prefaces trace and debug messages with the current wall time.
- Numerous memory leaks have been plugged.  Thanks to @FedeDP Federico Di Pierro (developer of [Clightd](https://github.com/FedeDP/Clightdl)) for his extensive help in identifying and fixing these leaks.
- API functions **ddca_open_display()** and **ddca_open_display2()** now always return **DDCRC_ALREADY_OPEN** if the display is already open in the current thread.  Previously an assert failure would occur under certain circumstances, in particular when called from KDE PowerDevil. Thanks to @shaumux Shaumyadeep Chaudhuri for his patient help in diagnosing this situation when **ddca_open_display2()** (or **ddca_open_display()**) was called without a prior call to **ddca_close_display()**. 

For a fuller description of the changes since release 1.2.0, see file CHANGELOG.md and also the preliminary [release notes](https://www.ddcutil.com/release_notes) on www.ddcutil.com.

FedeDP, 2021-11-30:
Hi @rockowitz !
Didn't want to bother you with a new issue; moreover i don't even know if this is an issue!
On my [Clightd](https://github.com/FedeDP/Clightd) project i am seeing a weird behavior with ddcutil:
* Clightd catches SIGTER/SIGINT to gracefully quit
* I noticed that the behavior was broken since i implemented ddc support (and another feature that requires `libusb`), 
because both libusb and ddcutil start their own threads, and the signal is then caught by a casual thread in my process
* I fixed libusb issue by blocking those signals before everything else is run, using an attribute constructor with 101 priority: https://github.com/FedeDP/Clightd/blob/pipewire_support/src/modules/signal.c#L14

The issue:
* DDCUtil `_ddca_init` starts before my function anyway...i set` debug=true `in _ddca_init and can confirm that it runs before anything else

Weird things:
* Per documentation, `__attribute__ ((constructor))` functions run **after** `__attribute__ ((constructor ($PRIO)))` functions; but in this case it is running before everything else
* I tried to set an higher PRIO to `_ddca_init` (ie: making it start later), but it does still start as early as possible
* I tried to add `__asm__ (".section .init \n call block_signals \n .section .text\n");` to my `block_signals` function, ie: putting it in the ".init" section that should be started earlier than attribute constructor ones, and once more `_ddca_init` is starting before that.

I don't know what else i can try; one more thing: Clightd uses my own [libmodule](https://github.com/FedeDP/libmodule) that makes extensive usage of constructor attribute, and it works just fine.  

Sorry, it was quite a long, hopefully detailed, description; feel free to contact me privately if you wish :)

rockowitz, 2021-12-01:
@FedeDP  Sigh, just when I thought 1.2.1 was ready for release. :(

Here are my initial thoughts.

What if I surfaced a variant of function _ddca_terminate() (file api_base.c) in the API?  Would your signal handler be able to call it?  I haven't thought through all the bookkeeping involved, but it should be doable without requiring that all callers use it.  Almost any API functions subsequently called would have to fail with status code DDCRC_INVALID_OPERATION or some such.

FedeDP, 2021-12-01:
Hi! While it seems like a working solution, i am more keen to understand why is that even happening in the first place; i mean, this is not a bug in ddcutil imo, and it shouldn't block your release process :) 

I was looking for a discussion, ie: if you've got any idea, may be ddcutil does something in the makefile/configure script?
Just wondering...

Note that while **there is** a weird behavior, it is not worrisome: it just means that Clightd is not gracefully quitting and its resources are not fully released; but it is quitting anyway therefore there is no hurry to fix this.

FedeDP, 2021-12-01:
I was able to reproduce the issue with libmodule too; i have a simple example:
```
// build with gcc test.c -o test -lddcutil
#include <stdio.h>

void __attribute__ ((constructor (101))) top() {
    printf("test\n");
}
    
int main(int argc, char** argv) {
    return 0;
}
```
Obviously ddcuti `_ddca_init()` is built with `debug = true`; you can see that libddcutil init is called before the program ctor; it seems like a bug or at least undocumented behavior, but it's so weird; it's like constructor from shared objects run before constructor in main program, in any case.  
I will dig into it further until i find at least some documentation for the behavior.
By the way, this is not a bug report, thus feel free to ignore me :)

closed 2025-02-11 by rockowitz