rockowitz/ddcutil #236: Input switchung seems to works but switching back after a few seconds

b4sst4rdSK, 2021-11-26:
i try to switch input to dp2 via 

`ddcutil --model PL3466WQ setvcp 60 10 && ddcutil scs`

it seems the display switches input but after a few seconds the input returns  to dp1

the capabilities show me: 

`  Feature: 60 (Input Source)
      Values:
         11: HDMI-1
         12: HDMI-2
         0f: DisplayPort-1
         10: DisplayPort-2 `

[output.txt](https://github.com/rockowitz/ddcutil/files/7607636/output.txt)

Please be kind, this is my 1st issue. ;-) Can you help me?

rockowitz, 2021-11-26:
There have been other reports of this behaviour.  See for example #32.

I do have a couple suggestions.  First, if your monitor's OSD allows you to choose between picking a specific monitor and something like "Scan Sources" or "Autodetect", try turning off the autodetection as that may be overriding the DDC request to change input. 

Second, see issue #136.  In that case an Iiyama monitor required the "ddcutil scs" command for changes to stick.  (Specifying the ***--noverify*** option, as mentioned in that issue,  is no longer required.  Verification is no longer performed for feature x60.)

Though it does not appear to be at issue in your case, it's good practice to verify the actual feature values by manually setting the input in the OSD and reading the value using **ddcutil getvcp x60** or a Windows counterpart.  Capabilities strings lie, and monitors deviate from the MCCS specification.

No need to request kindness.  Your report includes the **ddcutil interrogate** output, and submits it as an attachment so that it doesn't clutter the issue thread and I don't have to copy/paste it into an editor to work with it.  The problem is clearly described.  And please, report back what worked and didn't work for you.  That's all I ask.

Sanford

b4sst4rdSK, 2021-11-28:
the monitor input scan is set to manuell. 

Both ways not working for me. I try to switch input on windows OS and on Linux. Nothing helps. 

Maybe iiyama Suppoert can help. I try to open up a ticket, we will see if this might be helpful. 

Thank you for support. 

Sascha

adam-kulagowski, 2022-02-07:
@b4sst4rdSK did had any success getting this working? I'm considering this ( GB3466WQSU ) monitor as well and inability of input switch via DDC is kinda deal breaker for me :(

b4sst4rdSK, 2022-02-07:
@adam-kulagowski at this moment, there is no solution. Ilyama customer Support promised me a firmware update which supports the suggested functions. I dont know how long this coding still bei in progess.

adam-kulagowski, 2022-02-07:
Thanks for the update.

b4sst4rdSK, 2022-02-20:
after a custom firmware update, now the monitor supports ddc cmd´s and i can switch input per shell script. The firmware update and a special flashing device were send by iiyama.

closed 2022-02-20 by b4sst4rdSK