rockowitz/ddcutil #238: Invalid display, HP ZR2440w, Double byte in packet

woky, 2021-12-05:
I'd like to report issue with monitor HP ZR2440w, plugged in HDMI port of Asus Laptop (UX310). The test environment is very close to https://github.com/rockowitz/ddcutil/issues/46 except the different monitor model and that I don't see `Unexpected source address` errors, only `Double byte in packet` ones (pasted at the bottom). I'm running Arch Linux.

What's perplexing is that it has worked for me in the past, I used to control brightness without problem with the exact same hardware setup. I updated BIOS firmware on this laptop at some point but I don't remember if it was just before I started observing this issue. I also don't remember when it's started happening to pinpoint some kernel upgrade. I can only say it's been failing for circa a year. :-(

ddcutil detect --verbose --ddc --force-slave-address >> [ddcutil-detect.txt](https://github.com/rockowitz/ddcutil/files/7655503/ddcutil-detect.txt)

<hr/>

```
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e 6e
```

rockowitz, 2021-12-05:
Please submit the output of **sudo ddcutil interrogate** as an attachment. 

Please also submit the output of **sudo ddcutil detect --f4** as an attachment.

NoAccount1, 2024-12-28:
I have the same display, also plugged in HDMI

```
Spec :
 - Motherboard: X470 GAMING PLUS MAX (MS-7B79)
 - CPU : AMD Ryzen 5 1600
 - GPU : NVIDIA GeForce GTX 970 [Discrete]
```

I get the same error with `ddcutil detect --verbose --ddc --force-slave-address`

Here are the other commands :

`# ddcutil interrogate` > [ddcutil_interrogate.log](https://github.com/user-attachments/files/18266454/ddcutil_interrogate.log)
`# ddcutil detect --verbose --ddc --force-slave-address` > [ddcutil_detect_--verbose.log](https://github.com/user-attachments/files/18266457/ddcutil_detect_--verbose.log)

And I get an error with `sudo ddcutil detect --f4` :
```log
===========> IO STRATEGY 1:
   i2c_io_strategy:          IOCTL
   EDID read uses I2C layer: I2C Layer
   EDID read bytewise:       false
   write before read:        false
   EDID read size:           128
ddcutil: i2c_bus_core.c:614: i2c_check_bus: Assertion `sys_drm_connectors' failed.
[1]    30039 IOT instruction (core dumped)  ddcutil detect --f4
```